zigbeefordomoticz/Domoticz-Zigbee #1543: [Model Certification] - ZLL 1-10V

Krakinou, 2023-02-19:
1. Please do provide device information such as Brand and Model name.
Contrôleur de gradation 1-10V ZLL 1-10V - No Brand that I can decypher
It seems to work well for more than a year now, integrated without notable issue. Idem for domoticz integration.

2. Provide the URL where you have purchase the device
https://fr.aliexpress.com/item/1005003619684238.html?spm=a2g0o.order_list.order_list_main.14.39675e5b0TWQcy&gatewayAdapt=glo2fra

3. Finaly copy-paste all information provided in the log
````
{"PluginInfos":{"HardwareID":8,"HomeFolder":"/opt/yunohost/domoticz/plugins/Domoticz-Zigbee/","StartupFolder":"/opt/yunohost/domoticz/","UserDataFolder":"/opt/yunohost/domoticz/","WebRoot":"","Database":"/opt/yunohost/domoticz/domoticz.db","Language":"fr","Version":"6.3","Author":"pipiche38","Name":"Zigbee for Domoticz","Address":"0.0.0.0","Port":"9999","SerialPort":"/dev/ttyUSB20","Username":"","Password":"","Key":"Zigate","Mode1":"V2","Mode2":"USB","Mode3":"False","Mode4":"9440","Mode5":"http://myuser:mypassword@127.0.0.1:8080","Mode6":"2","DomoticzVersion":"2022.2","DomoticzHash":"eea9db734","DomoticzBuildTime":"2022-11-05 13:05:35","PluginBranch":"stable6","PluginVersion":"6.3.010","TimeStamp":1676805391,"available":"6.3.010","available-firmMajor":"05","available-firmMinor":"0321","FirmwareVersion":"0321","FirmwareUpdate":false,"PluginUpdate":false,"LogLevel":4,"Zigpy":false,"CoordinatorModel":"ZiGate+ (V2)","CoordinatorFirmwareVersion":"0321","DisplayFirmwareVersion":"Zigate - 0321","CoordinatorIEEE":"00158d0006a3fcac","NetworkSize":"Total: 10 | Routers: 4 | End Devices: 6","NetworkDevices":{"1037":{"LUMI":["lumi.sensor_magnet.aq2"],"LiXee":["ZLinky_TIC-historique-mono"]},"1286":{"SONOFF":["S26R2ZB"]},"100b":{"eWeLight":["ZB-DL01"]},"0000":{"TUYATEC-r9hgssol":["RH3001"],"eWeLink":["TH01","DS01"]}}},"Analytics":1,"Device":{"Version":"3","ZDeviceName":"Plafonnier","Status":"inDB","SQN":"55","Ep":{"01":{"1000":{},"0000":{"4000":"2.0.1","0004":"eWeLight","0005":"ZB-DL01","0000":"3","0001":"3","0002":"2","0003":"0","0006":"20211229","0007":"1"},"0003":{},"0004":{},"0005":{},"0006":{"4001":"0","4002":"0","0000":"00"},"0008":{"0000":"c9"},"ClusterType":{"640":"LvlControl"}}},"Heartbeat":"51555","RIA":"3","LQI":67,"Battery":{},"Model":"ZB-DL01","ForceAckCommands":[],"MacCapa":"8e","IEEE":"a4c1386ac187bde2","Type":"Switch/LvlControl","ProfileID":"0104","ZDeviceID":"0101","App Version":"3","Attributes List":{"Request":{"01":{"1000":1650230377.0905747,"0000":1650230377.087977,"0003":1650230377.088391,"0004":1650230377.0887992,"0005":1650230377.0891962,"0006":1650230377.089614,"0008":1650230377.0900137}}},"DeviceType":"FFD","HW Version":"0","Last Cmds":[],"LogicalType":"Router","Manufacturer":"100b","Manufacturer Name":"eWeLight","NbEp":"1","PowerSource":"Main","ReadAttributes":{"Ep":{"01":{"0006":{"TimeStamp":1676805178,"iSQN":{"4001":27,"4002":27,"0000":612},"Attributes":{"4001":"00","4002":"00","0000":"00"},"ZigateRequest":{}},"0008":{"TimeStamp":1676805183,"iSQN":{"0000":613},"Attributes":{"0000":"00"},"ZigateRequest":{}},"0000":{"TimeStamp":1663108167,"iSQN":{"4000":31,"0004":29,"0005":29,"0000":10,"0001":29,"0002":29,"0003":30,"0006":30,"0007":30,"000a":30,"000f":30,"0010":31,"0015":31,"f000":31},"Attributes":{"4000":"00","0004":"00","0005":"00","0000":"00","0001":"00","0002":"00","0003":"00","0006":"00","0007":"00","000a":"86","000f":"86","0010":"86","0015":"86","f000":"86"},"ZigateRequest":{}}}}},"ReceiveOnIdle":"On","Stack Version":"2","Stamp":{"time":1676805183.4751952,"Time":"2023-02-19 12:13:03","MsgType":"8102","LastSeen":1676805183,"LastPing":1663108167},"ZCL Version":"3","Health":"Live","RoutingTable":[],"bitfield":"4001","Neighbours":[{"SQN":0,"Status":"","TimeStamp":1667599507,"Time":1667599507,"Devices":[{"0000":{"_relationshp":"Parent","_devicetype":"Coordinator","_depth":0,"_lnkqty":61,"_rxonwhenidl":"Rx-On","_IEEE":"00158d0006a3fcac","_permitjnt":"??"}},{"adf9":{"_relationshp":"Sibling","_devicetype":"Router","_depth":1,"_lnkqty":115,"_rxonwhenidl":"Rx-On","_IEEE":"00158d0005d29a50","_permitjnt":"??"}}]},{"SQN":0,"Status":"","TimeStamp":1667599640,"Time":1667599640,"Devices":[{"0000":{"_relationshp":"Parent","_devicetype":"Coordinator","_depth":0,"_lnkqty":59,"_rxonwhenidl":"Rx-On","_IEEE":"00158d0006a3fcac","_permitjnt":"??"}},{"adf9":{"_relationshp":"Sibling","_devicetype":"Router","_depth":1,"_lnkqty":115,"_rxonwhenidl":"Rx-On","_IEEE":"00158d0005d29a50","_permitjnt":"??"}}]},{"SQN":0,"Status":"","TimeStamp":1671054239,"Time":1671054239,"Devices":[{"0000":{"_relationshp":"Parent","_devicetype":"Coordinator","_depth":0,"_lnkqty":24,"_rxonwhenidl":"Rx-On","_IEEE":"00158d0006a3fcac","_permitjnt":"??"}},{"adf9":{"_relationshp":"Sibling","_devicetype":"Router","_depth":1,"_lnkqty":115,"_rxonwhenidl":"Rx-On","_IEEE":"00158d0005d29a50","_permitjnt":"??"}},{"08ae":{"_relationshp":"Child","_devicetype":"End Device","_depth":2,"_lnkqty":85,"_rxonwhenidl":"Rx-Off","_IEEE":"00124b002892f854","_permitjnt":"??"}}]}],"NeighbourTableSize":3,"Param":{},"ConfigSource":"8043","WriteAttributes":{"Ep":{"01":{"1000":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0000":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0003":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0004":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0005":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0006":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0008":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}}}}},"ReadConfigureReporting":{},"BindingTable":{"Devices":[{"a4c1386ac187bde2":{"sourceEp":"01","Cluster":"0006","targetIEEE":"00158d0006a3fcac","targetNickName":"Zigate","targetEp":"01"}},{"a4c1386ac187bde2":{"sourceEp":"01","Cluster":"0000","targetIEEE":"00158d0006a3fcac","targetNickName":"Zigate","targetEp":"01"}},{"a4c1386ac187bde2":{"sourceEp":"01","Cluster":"0003","targetIEEE":"00158d0006a3fcac","targetNickName":"Zigate","targetEp":"01"}},{"a4c1386ac187bde2":{"sourceEp":"01","Cluster":"0004","targetIEEE":"00158d0006a3fcac","targetNickName":"Zigate","targetEp":"01"}},{"a4c1386ac187bde2":{"sourceEp":"01","Cluster":"0005","targetIEEE":"00158d0006a3fcac","targetNickName":"Zigate","targetEp":"01"}},{"a4c1386ac187bde2":{"sourceEp":"01","Cluster":"0008","targetIEEE":"00158d0006a3fcac","targetNickName":"Zigate","targetEp":"01"}},{"a4c1386ac187bde2":{"sourceEp":"01","Cluster":"1000","targetIEEE":"00158d0006a3fcac","targetNickName":"Zigate","targetEp":"01"}}],"SQN":4,"TimeStamp":1650127710.1752007,"BindingTableTableSize":7,"Status":"Success"},"SWBUILD_1":"20211229","Max Rx":"0042","Bind":{"01":{"1000":{"Target":"0000","Stamp":1657578851,"Phase":"requested","Status":"","i_sqn":4896},"0000":{"Target":"0000","Stamp":1657578851,"Phase":"requested","Status":"","i_sqn":4890},"0003":{"Target":"0000","Stamp":1657578851,"Phase":"requested","Status":"","i_sqn":4891},"0004":{"Target":"0000","Stamp":1657578851,"Phase":"requested","Status":"","i_sqn":4892},"0005":{"Target":"0000","Stamp":1657578851,"Phase":"requested","Status":"","i_sqn":4893},"0006":{"Target":"0000","Stamp":1657578851,"Phase":"requested","Status":"","i_sqn":4894},"0008":{"Target":"0000","Stamp":1657578851,"Phase":"requested","Status":"","i_sqn":4895}}},"macapa":"8e","Capability":["Full-Function Device","Main Powered","Receiver during Idle","Standard security","NwkAddr should be allocated"],"CertifiedDevice":false,"SWBUILD_3":"2.0.1","descriptor_capability":"00","Max Buffer Size":"42","_rawNodeDescriptor":"100b004200422a00008e424001","Log_UnknowDeviceFlag":1674931025.8948665,"Max Tx":"0042","ConfigureReporting":{},"server_mask":"2a00","Epv2":{"01":{"ClusterIn":{"1000":{},"0000":{},"0003":{},"0004":{},"0005":{},"0006":{},"0008":{}},"ClusterOut":{},"ProfileID":"0104","ZDeviceID":"0101"}},"ConsistencyCheck":"ok","RollingLQI":[68,68,67,66,68,67,67,67,66,66,67],"CheckParam":false}}
````

pipiche38, 2023-02-19:
Is it working and behaving correctly?

Krakinou, 2023-02-19:
as far as I can tell, yes.

-------- Message d'origine --------

pipiche38, 2023-02-19:
Config file has been created on the release candidate branch for the next release.
If you want to check you need to switch 

```
git pull
git checkout config-release-6.3.011
git pull
```

restart the plugin

When it will be released or if you want to switch back to stable6 (default) branch you just have to do

`git checkout stable6`

and restart the plugin

closed 2023-02-23 by pipiche38