zigbeefordomoticz/Domoticz-Zigbee #1544: [Model Certification] -

adruet29, 2023-02-22:
1. Please do provide device information such as Brand and Model name.

OWON - Sonde de température déportée sur câble Zigbee
Référence: THS-317-ET
1. Provide the URL where you have purchase the device
https://www.domadoo.fr/fr/peripheriques/5998-owon-sonde-de-temperature-deportee-sur-cable-zigbee.html

1. Finaly copy-paste all information provided in the log

{"PluginInfos":{"HardwareID":13,"HomeFolder":"/opt/domoticz/plugins/Zigate/","StartupFolder":"/opt/domoticz/","UserDataFolder":"/opt/domoticz/","WebRoot":"","Database":"/opt/domoticz/domoticz.db","Language":"fr","Version":"6.3","Author":"pipiche38","Name":"Zigate","Address":"192.168.10.50","Port":"9999","SerialPort":"/dev/serial/by-id/usb-Silicon_Labs_ZiGate_0001-if00-port0","Username":"","Password":"","Key":"Zigate","Mode1":"V1","Mode2":"USB","Mode3":"False","Mode4":"9440","Mode5":"http://alex:MaDomot1c@192.168.10.50:8085","Mode6":"null","DomoticzVersion":"2023.1","DomoticzHash":"f9b9ac774","DomoticzBuildTime":"2023-02-14 15:06:40","PluginBranch":"stable6","PluginVersion":"6.3.009","TimeStamp":1677090494,"available":"6.3.010","available-firmMajor":"04","available-firmMinor":"0321","FirmwareVersion":"0321","FirmwareUpdate":false,"PluginUpdate":true,"LogLevel":4,"Zigpy":false,"CoordinatorModel":"ZiGate Classic PDM (OptiPDM)","CoordinatorFirmwareVersion":"0321","DisplayFirmwareVersion":"Zigate - 0321","CoordinatorIEEE":"00158d0000ce21b1","NetworkSize":"Total: 12 | Routers: 3 | End Devices: 9","NetworkDevices":{"1002":{"_TZ3000_dowj6gyi":["TS0201"],"_TZ3000_wamqdr3f":["TS011F"]},"1037":{"LUMI":["lumi.weather"]},"1286":{"SONOFF":["S26R2ZB"]},"117c":{"IKEA of Sweden":["TRADFRI onoff switch","TRADFRI control outlet"]},"0000":{"eWeLink":["WB01"]}}},"Analytics":1,"Device":{"Version":"3","ZDeviceName":"","Status":"inDB","SQN":"05","Ep":{"01":{"0000":{"0004":"OWON","0005":"THS317-ET","0000":"3","0001":"3","0002":"0","0003":"1","0006":"20220721","0007":"3"},"0001":{"0020":27,"0021":112},"0020":{},"0003":{},"0402":{"0000":2070},"ClusterType":{"2870":"Voltage","2871":"Temp"}}},"Heartbeat":"15","RIA":"4","LQI":156,"Battery":112,"Model":"THS317-ET","ForceAckCommands":[],"MacCapa":"80","IEEE":"3c6a2cfffed245cc","Type":"Voltage/Temp","ProfileID":"0104","ZDeviceID":"0302","App Version":"3","Attributes List":{},"DeviceType":"RFD","HW Version":"1","Last Cmds":[[593,"02","68a6","0120","01010001000000000300210000012CFFFE00000001002900200E100E1000000001002100210E100E10000001"],[594,"02","68a6","0100","0101040200000000010000"],[595,"02","68a6","0100","010100010000000003000000010003"],[596,"02","68a6","0100","010100010000000003002000210033"],[597,"02","68a6","0100","0101000100000000010035"],[598,"02","68a6","0100","010100000000000003000400050000"],[599,"02","68a6","0100","010100000000000003000100020003"],[600,"02","68a6","0100","01010000000000000300060007000a"],[601,"02","68a6","0100","010100000000000003000f00100015"],[602,"02","68a6","0100","0101000000000000024000f000"]],"LogicalType":"End Device","Manufacturer":"113c","Manufacturer Name":"OWON","NbEp":"1","PowerSource":"Battery","ReadAttributes":{"Ep":{"01":{"0000":{"TimeStamp":1677090438,"iSQN":{"4000":602,"0004":598,"0005":598,"0000":598,"0001":599,"0002":599,"0003":599,"0006":600,"0007":600,"000a":600,"000f":601,"0010":601,"0015":601,"f000":602},"Attributes":{"4000":"86","0004":"00","0005":"00","0000":"00","0001":"00","0002":"00","0003":"00","0006":"00","0007":"00","000a":"86","000f":"86","0010":"86","0015":"86","f000":"86"},"ZigateRequest":{}},"0402":{"TimeStamp":1677090477,"iSQN":{"0000":594},"Attributes":{"0000":"00"},"ZigateRequest":{}},"0001":{"TimeStamp":1677090456,"iSQN":{"0000":595,"0001":595,"0003":595,"0020":596,"0021":596,"0033":596,"0035":597},"Attributes":{"0000":"86","0001":"86","0003":"86","0020":"00","0021":"00","0033":"86","0035":"86"},"ZigateRequest":{}}}}},"ReceiveOnIdle":"Off","Stack Version":"0","Stamp":{"time":1677090482.3619244,"Time":"2023-02-22 19:28:02","MsgType":"8011","LastSeen":1677090482},"ZCL Version":"3","Health":"Live","Capability":["Reduced-Function Device","Standard security","NwkAddr should be allocated"],"ZDPSQN":"04","RollingLQI":[96,96,96,96,96,96,93,93,96,153,156],"Epv2":{"01":{"ClusterIn":{"0001":{},"0020":{},"0000":{},"0003":{},"0402":{}},"ClusterOut":{"0003":{}},"ProfileID":"0104","ZDeviceID":"0302"}},"WriteAttributes":{"Ep":{"01":{"0000":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0001":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0020":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0003":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0402":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}}}}},"PairingInProgress":false,"ZDeviceVersion":"0001","ConfigSource":"8043","_rawNodeDescriptor":"113c005200522c000080524002","Max Buffer Size":"52","Max Rx":"0052","Max Tx":"0052","macapa":"80","bitfield":"4002","server_mask":"2c00","descriptor_capability":"00","PairingTime":1677090419.0664358,"Bind":{"01":{"0001":{"Target":"0000","Stamp":1677090419,"Phase":"binded","Status":"00","i_sqn":586},"0020":{"Target":"0000","Stamp":1677090420,"Phase":"binded","Status":"00","i_sqn":587},"0000":{"Target":"0000","Stamp":1677090421,"Phase":"binded","Status":"00","i_sqn":588},"0003":{"Target":"0000","Stamp":1677090422,"Phase":"binded","Status":"00","i_sqn":589},"0402":{"Target":"0000","Stamp":1677090423,"Phase":"binded","Status":"00","i_sqn":590}}},"BindingTable":{"SQN":0,"Status":"Success","TimeStamp":1677090440.6743467,"BindingTableSize":5,"Devices":[{"3c6a2cfffed245cc":{"sourceEp":"01","Cluster":"0001","targetIEEE":"00158d0000ce21b1","targetNickName":"Zigbee Coordinator","targetEp":"01"}},{"3c6a2cfffed245cc":{"sourceEp":"01","Cluster":"0020","targetIEEE":"00158d0000ce21b1","targetNickName":"Zigbee Coordinator","targetEp":"01"}},{"3c6a2cfffed245cc":{"sourceEp":"01","Cluster":"0000","targetIEEE":"00158d0000ce21b1","targetNickName":"Zigbee Coordinator","targetEp":"01"}},{"3c6a2cfffed245cc":{"sourceEp":"01","Cluster":"0003","targetIEEE":"00158d0000ce21b1","targetNickName":"Zigbee Coordinator","targetEp":"01"}},{"3c6a2cfffed245cc":{"sourceEp":"01","Cluster":"0402","targetIEEE":"00158d0000ce21b1","targetNickName":"Zigbee Coordinator","targetEp":"01"}},{"3c6a2cfffed245cc":{"sourceEp":"01","Cluster":"0003","targetIEEE":"00158d0000ce21b1","targetNickName":"Zigbee Coordinator","targetEp":"01"}},{"3c6a2cfffed245cc":{"sourceEp":"01","Cluster":"0402","targetIEEE":"00158d0000ce21b1","targetNickName":"Zigbee Coordinator","targetEp":"01"}}]},"ConfigureReporting":{"Ep":{"01":{"0402":{"TimeStamp":1677090419.0684595,"iSQN":{"0000":592},"Attributes":{"0000":"00"},"ZigateRequest":{}},"0001":{"TimeStamp":1677090419.068924,"iSQN":{"0000":593,"0020":593,"0021":593},"Attributes":{"0000":"86","0020":"8d","0021":"8d"},"ZigateRequest":{}}}}},"CertifiedDevice":false,"ZCLSQN":"00","BatteryUpdateTime":1677090456,"SWBUILD_1":"20220721","ReadConfigureReporting":{"Ep":{"01":{"0402":{"0000":{"TimeStamp":1677090441.7383914,"Status":"00","DataType":"29","MinInterval":"000a","MaxInterval":"012c"}}}},"TimeStamp":1677090441.7383893}}}

pipiche38, 2023-02-22:
What is the problem ? Does it works and it is just seen as not certified ?

pipiche38, 2023-02-23:
Please provide a minimum of context so we can understand what is needed .

- what does work ?
- what does not work ?

pipiche38, 2023-02-24:
planned for 6.3.011
if you want you can switch to branch: `config-release-6.3.011`