zigbeefordomoticz/Domoticz-Zigbee #1545: Robb Smarrt Smart Plug Zigbee 3680watt no widget on/off made

mcmikev, 2023-02-23:
* Robb Smarrt Smart Plug Zigbee 3680watt
* Robb Smarrt
* https://www.robbshop.nl/robb-smarrt-slimme-stekker-zigbee-3680watt)
* I expect also an ON/OFF switch/widget being made
* 
[export.json.txt](https://github.com/zigbeefordomoticz/Domoticz-Zigbee/files/10814149/export.json.txt)
[_PluginZigbee_48.log](https://github.com/zigbeefordomoticz/Domoticz-Zigbee/files/10814167/_PluginZigbee_48.log)

mcmikev, 2023-02-24:
Removed the widgets/device from plugin and domoticz.

[_PluginZigbee_48.log](https://github.com/zigbeefordomoticz/Domoticz-Zigbee/files/10823132/_PluginZigbee_48.log)
[export.json.txt](https://github.com/zigbeefordomoticz/Domoticz-Zigbee/files/10823137/export.json.txt)

mcmikev, 2023-02-24:
updated plugin develop version , here are the logs

Deleted the device in domoticz, was gone in plugin, re-paired the plug
[export.txt](https://github.com/zigbeefordomoticz/Domoticz-Zigbee/files/10823935/export.txt)
[_PluginZigbee_48.log](https://github.com/zigbeefordomoticz/Domoticz-Zigbee/files/10823936/_PluginZigbee_48.log)

mcmikev, 2023-02-24:
Here is some log from domoticz

` 2023-02-24 13:11:57.391 Status: Zigbee: Request Accepting new Hardware for 240 seconds
2023-02-24 13:11:57.442 Status: Zigbee: Accepting new Hardware: Enable (On)
2023-02-24 13:12:07.585 Status: Zigbee: [None] NEW OBJECT: 7b7e None
2023-02-24 13:12:09.018 Status: Zigbee: [-] NEW OBJECT: 7b7e Active Endpoint Response Ep: 01 LQI: 0
2023-02-24 13:12:09.018 Status: Zigbee: [-] NEW OBJECT: 7b7e Active Endpoint Response Ep: f2 LQI: 0
2023-02-24 13:12:11.111 Status: Zigbee: [0] NEW OBJECT: 7b7e Model Name: ROB_200-017-0
2023-02-24 13:12:11.119 Status: Zigbee: [0] NEW OBJECT: 7b7e Manufacturer Name: ROBB smarrt
2023-02-24 13:12:11.131 Status: Zigbee: [None] NEW OBJECT: 7b7e Request Node Descriptor
2023-02-24 13:12:11.766 Status: Zigbee: (7b7e/04cd15fffe5bbd0d) sent a Leave indication and will be outside of the network. LQI: 0
2023-02-24 13:12:11.770 Status: Zigbee: (7b7e/04cd15fffe5bbd0d) cleanup key plugin data informations
2023-02-24 13:12:11.835 Status: Zigbee: (7b7e/04cd15fffe5bbd0d) sent a Leave indication and will be outside of the network. LQI: 0
2023-02-24 13:12:11.835 Status: Zigbee: (7b7e/04cd15fffe5bbd0d) cleanup key plugin data informations
2023-02-24 13:12:11.832 Error: Zigbee: Receiving a leave from 7b7e/04cd15fffe5bbd0d while device is Leave status
2023-02-24 13:12:12.336 Status: Zigbee: (7b7e/04cd15fffe5bbd0d) sent a Leave indication and will be outside of the network. LQI: 0
2023-02-24 13:12:12.337 Status: Zigbee: (7b7e/04cd15fffe5bbd0d) cleanup key plugin data informations
2023-02-24 13:12:12.328 Error: Zigbee: Receiving a leave from 7b7e/04cd15fffe5bbd0d while device is Leave status
2023-02-24 13:12:18.627 Status: Zigbee: reconnectNWkDevice - Update Status from inDB to 'inDB' for NetworkID : 6698
2023-02-24 13:12:18.704 Status: Zigbee: NetworkID: 6698 is replacing 7b7e for object: 04cd15fffe5bbd0d
2023-02-24 13:12:19.713 Status: Zigbee: Device Annoucement: NwkId: 6698 Ieee: 04cd15fffe5bbd0d MacCap: 8e
2023-02-24 13:12:30.247 Error: Zigbee: ReadCluster 0b04 - 6698/01 Attribute: 0000 Type: 1b Size: 0004 Data: 00010009
2023-02-24 13:12:30.672 Error: Zigbee: ReadCluster 0b04 - 6698/01 Attribute: 0600 Type: 21 Size: 0002 Data: 1
2023-02-24 13:12:30.682 Error: Zigbee: ReadCluster 0b04 - 6698/01 Attribute: 0601 Type: 21 Size: 0002 Data: 10
2023-02-24 13:12:30.802 Error: Zigbee: ReadCluster 0b04 - 6698/01 Attribute: 0602 Type: 21 Size: 0002 Data: 1
2023-02-24 13:12:30.818 Error: Zigbee: ReadCluster 0b04 - 6698/01 Attribute: 0603 Type: 21 Size: 0002 Data: 1000
2023-02-24 13:12:30.835 Error: Zigbee: ReadCluster 0b04 - 6698/01 Attribute: 0604 Type: 21 Size: 0002 Data: 1
2023-02-24 13:12:30.929 Error: Zigbee: ReadCluster 0b04 - 6698/01 Attribute: 0605 Type: 21 Size: 0002 Data: 10
2023-02-24 13:12:32.525 Error: Zigbee: ReadCluster 0b04 - 6698/01 Attribute: 0000 Type: 1b Size: 0004 Data: 00010009
2023-02-24 13:12:32.767 Error: Zigbee: ReadCluster 0b04 - 6698/01 Attribute: 0600 Type: 21 Size: 0002 Data: 1
2023-02-24 13:12:32.809 Error: Zigbee: ReadCluster 0b04 - 6698/01 Attribute: 0601 Type: 21 Size: 0002 Data: 10
2023-02-24 13:12:32.855 Error: Zigbee: ReadCluster 0b04 - 6698/01 Attribute: 0602 Type: 21 Size: 0002 Data: 1
2023-02-24 13:12:32.893 Error: Zigbee: ReadCluster 0b04 - 6698/01 Attribute: 0603 Type: 21 Size: 0002 Data: 1000
2023-02-24 13:12:32.935 Error: Zigbee: ReadCluster 0b04 - 6698/01 Attribute: 0604 Type: 21 Size: 0002 Data: 1
2023-02-24 13:12:32.987 Error: Zigbee: ReadCluster 0b04 - 6698/01 Attribute: 0605 Type: 21 Size: 0002 Data: 10
2023-02-24 13:15:31.066 Status: User: mdevries (IP: 192.168.150.1) initiated a switch command (4529/Zigbee - ROB_200-017-0_Plug-04cd15fffe5bbd0d-01/On)
2023-02-24 13:15:33.331 Status: User: mdevries (IP: 192.168.150.1) initiated a switch command (4529/Zigbee - ROB_200-017-0_Plug-04cd15fffe5bbd0d-01/Off)
2023-02-24 13:15:58.004 Status: Zigbee: Request Disabling Accepting new Hardware
2023-02-24 13:15:58.081 Status: Zigbee: Accepting new Hardware: Disable (Off) `

pipiche38, 2023-02-25:
Closing as I understood it is up an running

closed 2023-02-25 by pipiche38