zigbeefordomoticz/Domoticz-Zigbee #1554: [Model Certification] - IMMAX - PRISE INTELLIGENTE DOUBLE ZIGBEE

yahtala75, 2023-03-02:
IMMAX - PRISE INTELLIGENTE DOUBLE ZIGBEE TUYA 16A + MESURE DE CONSOMMATION - Référence 07752L
https://www.domadoo.fr/fr/peripheriques/6429-immax-prise-intelligente-double-zigbee-tuya-16a-mesure-de-consommation.html
{"PluginInfos":{"HardwareID":9,"HomeFolder":"/opt/domoticz/userdata/plugins/Zigate/","StartupFolder":"/opt/domoticz/","UserDataFolder":"/opt/domoticz/userdata/","WebRoot":"","Database":"/opt/domoticz/userdata/domoticz.db","Language":"en","Version":"6.3","Author":"pipiche38","Name":"Zigate","Address":"0.0.0.0","Port":"9999","SerialPort":"/dev/ttyUSB0","Key":"Zigate","Mode1":"V2","Mode2":"USB","Mode3":"False","Mode4":"9440","Mode6":"2","DomoticzVersion":"2023.1","DomoticzHash":"f9b9ac774","DomoticzBuildTime":"2023-02-14 15:06:40","PluginBranch":"stable6","PluginVersion":"6.3.011","TimeStamp":1677792151,"available":null,"available-firmMajor":null,"available-firmMinor":null,"FirmwareVersion":"03a0","FirmwareUpdate":null,"PluginUpdate":null,"LogLevel":7,"Zigpy":false,"CoordinatorModel":"ZiGate+ (V2)","CoordinatorFirmwareVersion":"03a0","DisplayFirmwareVersion":"Zigate - 03a0","CoordinatorIEEE":"00158d0005d28534","NetworkSize":"Total: 31 | Routers: 25 | End Devices: 6","NetworkDevices":{"1002":{"_TZ3000_z8wel7lz":["TS0001"],"_TZ3000_2putqrmw":["TS011F-plug"]},"1037":{"LUMI":["lumi.sensor_magnet.aq2"],"LiXee":["ZLinky_TIC-historique-mono"]},"1141":{"_TZ3000_2putqrmw":["TS011F-plug"]},"1189":{"unknown":["Plug Z3"],"LEDVANCE":["Plug Z3"]},"117c":{"unknown":["TRADFRI bulb E27 WW 806lm","TRADFRIbulbE27WSglobeopal1055lm"],"IKEA of Sweden":["TRADFRIbulbE14WSglobeopal470lm","TRADFRI onoff switch"]},"100b":{"unknown":[]},"0000":{"eWeLink":["WB01"]}}},"Analytics":1,"Device":{"Version":"3","ZDeviceName":"","Status":"inDB","SQN":"8c","Ep":{"01":{"0702":{"0000":1},"0b04":{"0505":216,"0508":0,"050b":0},"e000":{"d001":"06","d002":"0a"},"0006":{"4001":"0","4002":"0","5000":"1","8000":"00","8001":"1","8002":"0","0000":"01"},"e001":{"d010":0,"d030":0,"d000":1,"d001":"00000000","d002":"6543696","d003":"111","d004":"3292","d005":"550"},"0000":{"ffdf":"3\u00129\u0016\u0012","0001":"77","ffe2":"54","ffe4":"0","0004":"_TZ3000_jak16dll","0000":"3","0005":"TS011F","0007":"1","fffe":"0"},"ClusterType":{"2610":"Power","2611":"Meter","2612":"Power","2613":"Meter","2614":"Ampere","2615":"Switch"}},"02":{"0006":{"4001":"0","4002":"0","0000":"01"},"e001":{"d010":0},"ClusterType":{"2616":"Switch"}},"f2":{}},"Heartbeat":"1207","RIA":"4","LQI":60,"Battery":{},"Model":"TS011F","ForceAckCommands":[],"MacCapa":"8e","IEEE":"a4c138ec684137df","Type":"Power/Meter/Ampere/Switch","ProfileID":{},"ZDeviceID":{},"App Version":"77","Attributes List":{},"DeviceType":"FFD","HW Version":"","Last Cmds":[[277,"07","25d9","0100","0102000600000000010000"],[278,"02","25d9","0092","010200"],[279,"07","25d9","0100","0101000600000000010000"],[280,"07","25d9","0100","0102000600000000010000"],[623,"02","25d9","0092","010201"],[624,"07","25d9","0100","0101000600000000010000"],[625,"07","25d9","0100","0102000600000000010000"],[644,"02","25d9","0092","010101"],[645,"07","25d9","0100","0101000600000000010000"],[646,"07","25d9","0100","0102000600000000010000"]],"LogicalType":"Router","Manufacturer":"1141","Manufacturer Name":"_TZ3000_jak16dll","NbEp":"3","PowerSource":"Main","ReadAttributes":{"Ep":{"01":{"0000":{"TimeStamp":1677791598,"iSQN":{"0004":248,"0000":248,"0001":248,"0005":248,"0007":248,"fffe":248},"Attributes":{"ffdf":"00","ffe2":"00","ffe4":"00","0004":"00","0000":"00","0001":"00","0005":"00","0007":"00","fffe":"00"},"ZigateRequest":{}},"0702":{"TimeStamp":1677791535,"iSQN":{"0000":244,"0017":244,"0200":244,"0301":244,"0302":244,"0303":244,"0306":245,"0400":245},"Attributes":{"0000":"00"},"ZigateRequest":{}},"0b04":{"TimeStamp":1677791223,"iSQN":{"050b":249,"0505":249,"0508":249},"Attributes":{"050b":"00","0505":"00","0508":"00"},"ZigateRequest":{}},"e000":{"TimeStamp":1677789308,"iSQN":{},"Attributes":{"d001":"00","d002":"00"},"ZigateRequest":{}},"0006":{"TimeStamp":1677791514,"iSQN":{"4001":246,"4002":246,"0000":645},"Attributes":{"4001":"00","4002":"00","5000":"00","8000":"00","8001":"00","8002":"00","0000":"00"},"ZigateRequest":{}},"e001":{"TimeStamp":1677785657,"iSQN":{},"Attributes":{"d010":"00","d030":"00","d000":"00","d001":"00","d002":"00","d003":"00","d004":"00","d005":"00"},"ZigateRequest":{}}},"02":{"0006":{"TimeStamp":1677786121,"iSQN":{"4001":247,"4002":247,"0000":646},"Attributes":{"4001":"00","4002":"00","0000":"00"},"ZigateRequest":{}},"e001":{"TimeStamp":1677785657,"iSQN":{},"Attributes":{"d010":"00"},"ZigateRequest":{}},"0000":{"TimeStamp":1677776816,"iSQN":{},"Attributes":{},"ZigateRequest":{}}},"f2":{"0000":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}}}}},"ReceiveOnIdle":"","Stack Version":"","Stamp":{"time":1677791598.9517524,"Time":"2023-03-02 22:13:18","MsgType":"8011","LastSeen":1677791598},"ZCL Version":"3","Health":"Live","Capability":["Full-Function Device","Main Powered","Receiver during Idle","Standard security","NwkAddr should be allocated"],"ZDPSQN":"02","PairingInProgress":false,"RollingLQI":[81,81,81,77,77,77,77,77,60,60,60],"SQN_000a":"68","Epv2":{"01":{"ClusterIn":{},"ClusterOut":{},"ProfileID":{},"ZDeviceID":{}},"02":{"ClusterIn":{},"ClusterOut":{},"ProfileID":{},"ZDeviceID":{}},"f2":{"ClusterIn":{},"ClusterOut":{},"ProfileID":{},"ZDeviceID":{}}},"PairingTime":1677776821.1087554,"BindingTable":{"SQN":0,"Status":"Requested","TimeStamp":1677780751.0799499,"BindingTableSize":0,"Devices":[]},"ConfigureReporting":{"Ep":{"01":{"0702":{"TimeStamp":1677776821.112808,"iSQN":{"0000":240,"0400":240},"Attributes":{},"ZigateRequest":{}},"0b04":{"TimeStamp":1677776821.1214378,"iSQN":{"0505":241,"0508":241,"050b":241},"Attributes":{"0505":"00","0508":"00","050b":"00"},"ZigateRequest":{}},"0006":{"TimeStamp":1677776821.1238735,"iSQN":{"0000":242},"Attributes":{"0000":"00"},"ZigateRequest":{}}},"02":{"0006":{"TimeStamp":1677776821.1256747,"iSQN":{"0000":243},"Attributes":{"0000":"00"},"ZigateRequest":{}}}}},"CertifiedDevice":false,"ZCLSQN":"04","WriteAttributes":{"Ep":{"01":{"0702":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0b04":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"e000":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0006":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"e001":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0000":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}}},"02":{"0006":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"e001":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}}}}},"ReadConfigureReporting":{"Ep":{"01":{"0b04":{"0505":{"TimeStamp":1677776874.545692,"Status":"00","DataType":"21","MinInterval":"012c","MaxInterval":"0001","Change":"0001"},"0508":{"TimeStamp":1677776874.5462828,"Status":"00","DataType":"21","MinInterval":"012c","MaxInterval":"0001","Change":"0001"},"050b":{"TimeStamp":1677776874.546898,"Status":"00","DataType":"29","MinInterval":"012c","MaxInterval":"0005","Change":"0001"}}}},"TimeStamp":1677776874.5468948},"_rawNodeDescriptor":"1141004200422a00008e424001","Max Buffer Size":"42","Max Rx":"0042","Max Tx":"0042","macapa":"8e","bitfield":"4001","server_mask":"2a00","descriptor_capability":"00"}}

pipiche38, 2023-03-03:
will be provided on 'develop' branch for the 7.1 release

pipiche38, 2023-03-03:
So at first, please switch to the 'develop' branch on which I have push an update

```
git pull
git checkout develop
sudo python3 -m pip install -r requirements.txt --upgrade
```

Then restart the plugin, check that everything works, and after do the pairing.
You should get 2 plugs created and for each they have their own Power. The Meter, Ampere and Voltage is only for the first plugin. Let see that we get power on each ( which I'm pessimistic on that)

let see what is the outcome, I would be interested of the device information as you paste in this issue, just make sure that when doing it you have some load on each plug

yahtala75, 2023-03-03:
The checkout to the develop branch was OK : no error
after pairing I had some errors so I tried a second time and it was the same.
Even with that all the objects are created normally in Domotiz and I can manage the plugs on/off independently but the Power stay at 0, nothing came up and no update shown in the logs.

[domoticz.log](https://github.com/zigbeefordomoticz/Domoticz-Zigbee/files/10885504/domoticz.log)
[PluginZigbee_09.log](https://github.com/zigbeefordomoticz/Domoticz-Zigbee/files/10885505/PluginZigbee_09.log)

pipiche38, 2023-03-04:
I do not see any errors, except that you have issues between Zigate and devices. I cannot help on that.

>  ERROR   : [   ZiGateSerial_9] NXP_Extended_Error_Code - Extended Error Code: [c0] 

May be stop the plugin , disconnect the zigate, reset the device
then restart everything

Otherwise if you still have the issue please contact the Zigate vendor, as this is his firmware

yahtala75, 2023-03-04:
Hi, effectively I saw this error also. I will change my coordinateur from Zigate to SonOff -e during the weekend. 
But I was talking about the error during the pairing : 
2023-03-03 19:49:48.104  Status: Zigate: [None] NEW OBJECT: 080d Request Node Descriptor
2023-03-03 19:49:48.136  Error: Zigate: Call to function 'onHeartbeat' failed, exception details:
2023-03-03 19:49:48.145  Error: Zigate: Traceback (most recent call last):
2023-03-03 19:49:48.145  Error: Zigate:   File "/opt/domoticz/userdata/plugins/Zigate/plugin.py", line 1600, in onHeartbeat
2023-03-03 19:49:48.145  Error: Zigate:     _plugin.onHeartbeat()
2023-03-03 19:49:48.146  Error: Zigate:   File "/opt/domoticz/userdata/plugins/Zigate/plugin.py", line 987, in onHeartbeat
2023-03-03 19:49:48.146  Error: Zigate:     processListOfDevices(self, Devices)
2023-03-03 19:49:48.146  Error: Zigate:   File "/opt/domoticz/userdata/plugins/Zigate/Modules/heartbeat.py", line 881, in processListOfDevices
2023-03-03 19:49:48.146  Error: Zigate:     processNotinDBDevices(self, Devices, NWKID, status, RIA)
2023-03-03 19:49:48.146  Error: Zigate:   File "/opt/domoticz/userdata/plugins/Zigate/Modules/pairingProcess.py", line 116, in processNotinDBDevices
2023-03-03 19:49:48.146  Error: Zigate:     interview_state_createDB(self, Devices, NWKID, RIA, status)
2023-03-03 19:49:48.146  Error: Zigate:   File "/opt/domoticz/userdata/plugins/Zigate/Modules/pairingProcess.py", line 339, in interview_state_createDB
2023-03-03 19:49:48.146  Error: Zigate:     full_provision_device(self, Devices, NWKID, RIA, status)
2023-03-03 19:49:48.146  Error: Zigate:   File "/opt/domoticz/userdata/plugins/Zigate/Modules/pairingProcess.py", line 386, in full_provision_device
2023-03-03 19:49:48.146  Error: Zigate:     zigbee_provision_device(self, Devices, NWKID, RIA, status)
2023-03-03 19:49:48.146  Error: Zigate:   File "/opt/domoticz/userdata/plugins/Zigate/Modules/pairingProcess.py", line 429, in zigbee_provision_device
2023-03-03 19:49:48.146  Error: Zigate:     device_interview(self, NWKID)
2023-03-03 19:49:48.146  Error: Zigate:   File "/opt/domoticz/userdata/plugins/Zigate/Modules/pairingProcess.py", line 556, in device_interview
2023-03-03 19:49:48.146  Error: Zigate:     func(self, Nwkid)
2023-03-03 19:49:48.146  Error: Zigate:   File "/opt/domoticz/userdata/plugins/Zigate/Modules/readAttributes.py", line 1254, in ReadAttributeRequest_0702
2023-03-03 19:49:48.146  Error: Zigate:     for iterAttr in retreive_ListOfAttributesByCluster(self, key, EPout, "0702"):
2023-03-03 19:49:48.146  Error: Zigate:   File "/opt/domoticz/userdata/plugins/Zigate/Modules/readAttributes.py", line 226, in retreive_ListOfAttributesByCluster
2023-03-03 19:49:48.146  Error: Zigate:     targetAttribute = retreive_attributes_from_default_plugin_list(self, key, Ep, cluster)
2023-03-03 19:49:48.146  Error: Zigate:   File "/opt/domoticz/userdata/plugins/Zigate/Modules/readAttributes.py", line 283, in retreive_attributes_from_default_plugin_list
2023-03-03 19:49:48.146  Error: Zigate:     targetAttribute = retreive_attributes_from_zcl_standard( self, cluster)
2023-03-03 19:49:48.146  Error: Zigate:   File "/opt/domoticz/userdata/plugins/Zigate/Modules/readAttributes.py", line 207, in retreive_attributes_from_zcl_standard
2023-03-03 19:49:48.146  Error: Zigate:     return int(self.readZclClusters[ cluster ]["Attributes"].keys(), 16)
2023-03-03 19:49:48.146  Error: Zigate: TypeError: int() can't convert non-string with explicit base

2023-03-03 19:49:54.531  Zigate: Assisted Pairing: Polling: ['080d']
2023-03-03 19:49:54.531  Zigate: --> New device: 080d
2023-03-03 19:49:54.531  Zigate: --> inDB , removed 080d from List
2023-03-03 19:49:54.531  Zigate: ------> New Cluster: {'ClusterId': '0702', 'ClusterDesc': 'Unknown'}
2023-03-03 19:49:54.531  Zigate: ------> New Cluster: {'ClusterId': '0b04', 'ClusterDesc': 'Electrical Measurement'}
2023-03-03 19:49:54.531  Zigate: ------> New Cluster: {'ClusterId': 'e000', 'ClusterDesc': 'Unknown'}
2023-03-03 19:49:54.531  Zigate: ------> New Cluster: {'ClusterId': '0006', 'ClusterDesc': 'On/Off'}
2023-03-03 19:49:54.532  Zigate: ------> New Cluster: {'ClusterId': 'e001', 'ClusterDesc': 'Unknown'}
2023-03-03 19:49:54.532  Zigate: ------> New Cluster: {'ClusterId': '0000', 'ClusterDesc': 'Basic'}
2023-03-03 19:49:54.532  Zigate: ----> New Ep: {'Ep': '01', 'Clusters': [{'ClusterId': '0702', 'ClusterDesc': 'Unknown'}, {'ClusterId': '0b04', 'ClusterDesc': 'Electrical Measurement'}, {'ClusterId': 'e000', 'ClusterDesc': 'Unknown'}, {'ClusterId': '0006', 'ClusterDesc': 'On/Off'}, {'ClusterId': 'e001', 'ClusterDesc': 'Unknown'}, {'ClusterId': '0000', 'ClusterDesc': 'Basic'}]}
2023-03-03 19:49:54.532  Zigate: ------> New Cluster: {'ClusterId': '0006', 'ClusterDesc': 'On/Off'}
2023-03-03 19:49:54.532  Zigate: ------> New Cluster: {'ClusterId': 'e001', 'ClusterDesc': 'Unknown'}
2023-03-03 19:49:54.532  Zigate: ----> New Ep: {'Ep': '02', 'Clusters': [{'ClusterId': '0006', 'ClusterDesc': 'On/Off'}, {'ClusterId': 'e001', 'ClusterDesc': 'Unknown'}]}
2023-03-03 19:49:54.532  Zigate: ----> New Ep: {'Ep': 'f2', 'Clusters': []}
2023-03-03 19:49:54.532  Zigate:  --> New Device: {'NwkId': '080d', 'ProvisionStatus': 'inDB', 'ProvisionStatusDesc': 'inDB', 'IEEE': 'a4c138ec684137df', 'ProfileId': '', 'ProfileIdDesc': 'Unknow', 'ZDeviceID': '', 'ZDeviceIDDesc': 'Unknow', 'Model': '', 'PluginCertified': 'Unknow', 'Ep': [{'Ep': '01', 'Clusters': [{'ClusterId': '0702', 'ClusterDesc': 'Unknown'}, {'ClusterId': '0b04', 'ClusterDesc': 'Electrical Measurement'}, {'ClusterId': 'e000', 'ClusterDesc': 'Unknown'}, {'ClusterId': '0006', 'ClusterDesc': 'On/Off'}, {'ClusterId': 'e001', 'ClusterDesc': 'Unknown'}, {'ClusterId': '0000', 'ClusterDesc': 'Basic'}]}, {'Ep': '02', 'Clusters': [{'ClusterId': '0006', 'ClusterDesc': 'On/Off'}, {'ClusterId': 'e001', 'ClusterDesc': 'Unknown'}]}, {'Ep': 'f2', 'Clusters': []}]}

We will see after I finished to reconfigurate all my zigbee network

pipiche38, 2023-03-04:
this error should be fixed. please get 6.4.307

pipiche38, 2023-03-07:
I did contact the vendor and they confirm that the power measurement is for both, you don't have one per plug.
I have updated the conf file accordingly

pipiche38, 2023-03-14:
Any update , could confirm it works ?

closed 2023-04-14 by pipiche38