zigbeefordomoticz/Domoticz-Zigbee #1557: TUYA Zigbee presence sensor ZY-M100

nivekamoud, 2023-03-08:
Hi All,
See below, can u please have a look at it. It is now a switch with temp measuremnt what is not correct. 
Hope to here from u, 
gr kevin

* Product Name: ZY-M100
* Brand: Tuya
* URL on a web site,:  https://smarthomescene.com/reviews/tuya-zigbee-human-presence-sensor-zy-m100-review/
* What do you expect:  expext presence sensor and a Lux measurement
* The logs from the pairing process
[_PluginZigbee_34.log](https://github.com/zigbeefordomoticz/Domoticz-Zigbee/files/10923124/_PluginZigbee_34.log)
[image]

* export.json file as described here above

pipiche38, 2023-03-08:
which version of the Plugin do you use ?

pipiche38, 2023-03-08:
This device has been integrated and is available on the 'develop' branch
please switch to it ( taking assumption that you are on 6.3.011

git checkout develop
git pull
sudo python3 -m pip --ignore-requires-python --no-input install -r requirements.txt --upgrade

then  restart

you'll have to remove the existing device, and then  when well removed, you can redo the pairing

closed 2023-03-08 by pipiche38

nivekamoud, 2023-03-09:
Ok thanks will try-out tonight

nivekamoud, 2023-03-09:
ok got everything working again, now on develop branch, 6.4.3.12, radar distance gives a lot of updates. Can i set this somewhere in the parameters.

deennoo, 2023-03-09:
> 

i'll add info on wiki very soon

nivekamoud, 2023-03-09:
Ok, for now i have deleted the distance. I will test it for a couple of dans and let u kmow

deennoo, 2023-03-09:
> Ok, for now i have deleted the distance. I will test it for a couple of dans and let u kmow

Distance is send only if device get a body on his scope.

Earsing distance widget will only "clean" your log but device keep sending thé value when présence is détected.

deennoo, 2023-03-09:
> > 
> 
> i'll add info on wiki very soon

Wiki Param is update for this device

nivekamoud, 2023-03-09:
Wow thats fast.