zigbeefordomoticz/Domoticz-Zigbee #1562: problem with charset_normalizer

pipiche38, 2023-03-10:
> 2023-03-10 13:42:12.046 Error: Zigbee_RDC: Call to function 'onStart' failed, exception details:
> 2023-03-10 13:42:12.048 Error: Zigbee_RDC: Traceback (most recent call last):
> 2023-03-10 13:42:12.048 Error: Zigbee_RDC: File "/usr/local/lib/python3.9/dist-packages/aiohttp/client_reqrep.py", line 70, in <module>
> 2023-03-10 13:42:12.048 Error: Zigbee_RDC: import cchardet as chardet
> 2023-03-10 13:42:12.048 Error: Zigbee_RDC: ModuleNotFoundError: No module named 'cchardet'
> 2023-03-10 13:42:12.048 Error: Zigbee_RDC:
> 2023-03-10 13:42:12.048 Error: Zigbee_RDC: During handling of the above exception, another exception occurred:
> 2023-03-10 13:42:12.048 Error: Zigbee_RDC:
> 2023-03-10 13:42:12.048 Error: Zigbee_RDC: Traceback (most recent call last):
> 2023-03-10 13:42:12.048 Error: Zigbee_RDC: File "/opt/domoticz/userdata/plugins/Domoticz-Zigbee/plugin.py", line 1536, in onStart
> 2023-03-10 13:42:12.048 Error: Zigbee_RDC: _plugin.onStart()
> 2023-03-10 13:42:12.048 Error: Zigbee_RDC: File "/opt/domoticz/userdata/plugins/Domoticz-Zigbee/plugin.py", line 569, in onStart
> 2023-03-10 13:42:12.048 Error: Zigbee_RDC: from Classes.ZigpyTransport.Transport import ZigpyTransport
> 2023-03-10 13:42:12.048 Error: Zigbee_RDC: File "/opt/domoticz/userdata/plugins/Domoticz-Zigbee/Classes/ZigpyTransport/Transport.py", line 12, in <module>
> 2023-03-10 13:42:12.048 Error: Zigbee_RDC: import zigpy.application
> 2023-03-10 13:42:12.048 Error: Zigbee_RDC: File "/usr/local/lib/python3.9/dist-packages/zigpy/application.py", line 22, in <module>
> 2023-03-10 13:42:12.048 Error: Zigbee_RDC: import zigpy.ota
> 2023-03-10 13:42:12.048 Error: Zigbee_RDC: File "/usr/local/lib/python3.9/dist-packages/zigpy/ota/__init__.py", line 19, in <module>
> 2023-03-10 13:42:12.048 Error: Zigbee_RDC: import zigpy.ota.provider
> 2023-03-10 13:42:12.048 Error: Zigbee_RDC: File "/usr/local/lib/python3.9/dist-packages/zigpy/ota/provider.py", line 16, in <module>
> 2023-03-10 13:42:12.048 Error: Zigbee_RDC: import aiohttp
> 2023-03-10 13:42:12.048 Error: Zigbee_RDC: File "/usr/local/lib/python3.9/dist-packages/aiohttp/__init__.py", line 6, in <module>
> 2023-03-10 13:42:12.048 Error: Zigbee_RDC: from .client import (
> 2023-03-10 13:42:12.048 Error: Zigbee_RDC: File "/usr/local/lib/python3.9/dist-packages/aiohttp/client.py", line 59, in <module>
> 2023-03-10 13:42:12.048 Error: Zigbee_RDC: from .client_reqrep import (
> 2023-03-10 13:42:12.048 Error: Zigbee_RDC: File "/usr/local/lib/python3.9/dist-packages/aiohttp/client_reqrep.py", line 72, in <module>
> 2023-03-10 13:42:12.048 Error: Zigbee_RDC: import charset_normalizer as chardet # type: ignore[no-redef]
> 2023-03-10 13:42:12.048 Error: Zigbee_RDC: File "/usr/local/lib/python3.9/dist-packages/charset_normalizer/__init__.py", line 24, in <module>
> 2023-03-10 13:42:12.048 Error: Zigbee_RDC: from .api import from_bytes, from_fp, from_path
> 2023-03-10 13:42:12.048 Error: Zigbee_RDC: File "/usr/local/lib/python3.9/dist-packages/charset_normalizer/api.py", line 5, in <module>
> 2023-03-10 13:42:12.048 Error: Zigbee_RDC: from .cd import (
> 2023-03-10 13:42:12.048 Error: Zigbee_RDC: File "/usr/local/lib/python3.9/dist-packages/charset_normalizer/cd.py", line 9, in <module>
> 2023-03-10 13:42:12.048 Error: Zigbee_RDC: from .md import is_suspiciously_successive_range
> 2023-03-10 13:42:12.048 Error: Zigbee_RDC: AttributeError: partially initialized module 'charset_normalizer' has no attribute 'md__mypyc' (most likely due to a circular import)
>

pipiche38, 2023-03-10:
sudo pip3 install --force-reinstall --upgrade charset-normalizer==2.0.12

pipiche38, 2023-03-13:
added charset-normalizer==2.0.12 in, requirements.txt

closed 2023-03-13 by pipiche38

waltervl, 2025-09-08:
@pipiche38  There seems to be a new issue with this charset_normalizer issue as installed version is 3.0.1: https://forum.domoticz.com/viewtopic.php?p=328166 (not my issue!)

[image]

pipiche38, 2025-09-08:
With bookworm, you need to configure a python virtual environnement.
Please have a look here : https://zigbeefordomoticz.github.io/wik ... alEnv.html