zigbeefordomoticz/Domoticz-Zigbee #1564: TypeError: ias_sensitivity() takes 3 positional arguments but 4 were given

Achterhoeker, 2023-03-14:
After updating from branch issue/1353 to /develop the following error is shown in the log:

2023-03-14 15:52:48.079 Error: zigbee: Call to function 'onHeartbeat' failed, exception details:
2023-03-14 15:52:48.082 Error: zigbee: Traceback (most recent call last):
2023-03-14 15:52:48.082 Error: zigbee: File "/home/ronald/domoticz/plugins/Domoticz-Zigbee/plugin.py", line 1571, in onHeartbeat
2023-03-14 15:52:48.082 Error: zigbee: _plugin.onHeartbeat()
2023-03-14 15:52:48.082 Error: zigbee: File "/home/ronald/domoticz/plugins/Domoticz-Zigbee/plugin.py", line 965, in onHeartbeat
2023-03-14 15:52:48.082 Error: zigbee: processListOfDevices(self, Devices)
2023-03-14 15:52:48.082 Error: zigbee: File "/home/ronald/domoticz/plugins/Domoticz-Zigbee/Modules/heartbeat.py", line 822, in processListOfDevices
2023-03-14 15:52:48.082 Error: zigbee: processKnownDevices(self, Devices, NWKID)
2023-03-14 15:52:48.082 Error: zigbee: File "/home/ronald/domoticz/plugins/Domoticz-Zigbee/Modules/heartbeat.py", line 565, in processKnownDevices
2023-03-14 15:52:48.082 Error: zigbee: sanity_check_of_param(self, NWKID)
2023-03-14 15:52:48.082 Error: zigbee: File "/home/ronald/domoticz/plugins/Domoticz-Zigbee/Modules/paramDevice.py", line 330, in sanity_check_of_param
2023-03-14 15:52:48.082 Error: zigbee: func(self, NwkId, value)
2023-03-14 15:52:48.082 Error: zigbee: File "/home/ronald/domoticz/plugins/Domoticz-Zigbee/Modules/paramDevice.py", line 241, in ias_sensitivity
2023-03-14 15:52:48.082 Error: zigbee: self.iaszonemgt.ias_sensitivity(self, nwkid, sensitivity)
2023-03-14 15:52:48.082 Error: zigbee: TypeError: ias_sensitivity() takes 3 positional arguments but 4 were given

I've already enabled the debug log in the function ias_sensitivity. The device causing it seems to be:
 
2023-03-14 16:30:26.096 zigbee: sanity_check_of_param for eb7b
2023-03-14 16:30:26.096 zigbee: sanity_check_of_param - calling TuyaPIRKeepTime
2023-03-14 16:30:26.097 zigbee: sanity_check_of_param - calling IASsensitivity

[eb7b.zip](https://github.com/zigbeefordomoticz/Domoticz-Zigbee/files/10970530/eb7b.zip)
Device info as attachment. 

Is there some old (wrong) configuration in my setup, if so, where do i have to make some changes ?

Thanks for your support,
Ronald.

pipiche38, 2023-03-14:
@Achterhoeker thanks for reporting the issue.
please do update to the latest develop branch and that should be fixed

closed 2023-03-14 by pipiche38

Achterhoeker, 2023-03-14:
@pipiche38 It is indeed solved, thanks for the quick solution.