zigbeefordomoticz/Domoticz-Zigbee #1566: [Model Certification] - Fl LED Adeo Lexman E14 470lm CCT ADEO Lexman

LD40, 2023-03-17:
Just another Adeo/LexMan/Enki model sold by Leroy Merlin. [https://www.leroymerlin.fr/produits/decoration-eclairage/ampoule-et-led/ampoule-led/ampoule-e14/ampoule-led-connectee-flamme-e14-470lm-40w-variation-de-blancs-enki-lexman-84372285.html](url)
Works perfectly directly but just to have it certified/optimized

Logs :
`{"PluginInfos":{"HardwareID":17,"HomeFolder":"/home/pi/domoticz/plugins/Domoticz-Zigate/","StartupFolder":"/home/pi/domoticz/","UserDataFolder":"/home/pi/domoticz/","WebRoot":"","Database":"/home/pi/domoticz/domoticz.db","Language":"fr","Version":"6.3","Author":"pipiche38","Name":"Zigate","Address":"0.0.0.0","Port":"9999","SerialPort":"/dev/serial/by-id/usb-Prolific_Technology_Inc._USB-Serial_Controller-if00-port0","Key":"Zigate","Mode1":"V1","Mode2":"USB","Mode3":"False","Mode4":"9440","Mode6":"2","DomoticzVersion":"2023.1","DomoticzHash":"f9b9ac774","DomoticzBuildTime":"2023-02-14 15:06:40","PluginBranch":"stable6","PluginVersion":"6.3.011","TimeStamp":1679048442,"available":0,"available-firmMajor":0,"available-firmMinor":0,"FirmwareVersion":"0320","FirmwareUpdate":false,"PluginUpdate":false,"Zigpy":false,"CoordinatorModel":"ZiGate Classic PDM (OptiPDM)","CoordinatorFirmwareVersion":"0320","DisplayFirmwareVersion":"Zigate - 0320","CoordinatorIEEE":"00158d0001edb6bf","NetworkSize":"Total: 9 | Routers: 6 | End Devices: 3","NetworkDevices":{"1110":{"Profalux":["BSO-Profalux","MAI-ZTS"]},"1277":{"Adeo":["ZBEK-14"]}}},"Analytics":1,"Device":{"Version":"3","ZDeviceName":"","Status":"inDB","SQN":"3f","Ep":{"01":{"1000":{},"0000":{"4000":"2.09.00","0004":"Adeo","0005":"ZBEK-14","0000":"3","0001":"2","0002":"2","0003":"1","0006":"20210714","0007":"1","000a":"303130313030333435"},"0003":{},"0004":{},"0005":{},"0006":{"4001":"0","4002":"0","0000":"00"},"0008":{"0000":"fe"},"0300":{"0003":"28363","0004":"26397","0007":"331","0008":"2","400a":"16"},"fc57":{},"fc82":{},"0019":{},"ClusterType":{"458":"ColorControlWW"}},"f2":{"0021":{}}},"Heartbeat":"6928","RIA":"4","LQI":135,"Battery":{},"Model":"ZBEK-14","ForceAckCommands":[],"MacCapa":"8e","IEEE":"2c1165fffe2d28db","Type":"Switch/LvlControl/ColorControl","ProfileID":"a1e0","ZDeviceID":"0061","App Version":"2","Attributes List":{"Request":{"01":{"1000":1677344694.1100135,"0000":1677344694.1014616,"0003":1677344694.1019647,"0004":1677344694.1025515,"0005":1677344694.1030402,"0006":1677344694.1035295,"0008":1677344694.1040256,"0300":1677344694.109345,"fc57":1677344694.110569,"fc82":1677344694.1111262,"0019":1677344694.1117113},"f2":{"0021":1677344694.1122336}}},"DeviceType":"FFD","HW Version":"1","Last Cmds":[],"LogicalType":"Router","Manufacturer":"1277","Manufacturer Name":"Adeo","NbEp":"2","PowerSource":"Main","ReadAttributes":{"Ep":{"01":{"0000":{"TimeStamp":1677341061,"iSQN":{"4000":20922,"0004":20918,"0005":20918,"0000":21338,"0001":20919,"0002":20919,"0003":20919,"0006":20920,"0007":20920,"000a":20920,"000f":20921,"0010":20921,"0015":20921,"f000":20922},"Attributes":{"4000":"00","0004":"00","0005":"00","0000":"00","0001":"00","0002":"00","0003":"00","0006":"00","0007":"00","000a":"00","000f":"86","0010":"86","0015":"86","f000":"86"},"ZigateRequest":{}},"0019":{"TimeStamp":1677341060,"iSQN":{"0001":20909,"0002":20909,"0003":20909,"0004":20910,"0005":20910,"0006":20910,"0007":20911,"0008":20911,"0009":20911,"000a":20912},"Attributes":{"0001":"86","0002":"86","0003":"86","0004":"86","0005":"86","0006":"86","0007":"86","0008":"86","0009":"86","000a":"86"},"ZigateRequest":{}},"0006":{"TimeStamp":1679048357,"iSQN":{"4001":20913,"4002":20913,"0000":4831},"Attributes":{"4001":"00","4002":"00","0000":"00"},"ZigateRequest":{}},"0008":{"TimeStamp":1679048278,"iSQN":{"0000":4832},"Attributes":{"0000":"00"},"ZigateRequest":{}},"0300":{"TimeStamp":1679048377,"iSQN":{"0000":20915,"0001":20915,"0003":20915,"0004":20916,"0007":20916,"0008":20916,"400a":20917},"Attributes":{"0000":"86","0001":"86","0003":"00","0004":"00","0007":"00","0008":"00","400a":"00"},"ZigateRequest":{}}},"f2":{"0000":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}}}}},"ReceiveOnIdle":"On","Stack Version":"2","Stamp":{"time":1679048377.590468,"Time":"2023-03-17 11:19:37","MsgType":"8011","LastSeen":1679048377,"LastPing":1677377059},"ZCL Version":"3","Health":"Live","_rawNodeDescriptor":"1277005200522c00008e524001","bitfield":"4001","ReadConfigureReporting":{"Ep":{"01":{"0006":{"0000":{"TimeStamp":1677401758.9618077,"Status":"00","DataType":"10","MinInterval":"0001","MaxInterval":"012c"}},"0008":{"0000":{"TimeStamp":1677401759.9327905,"Status":"00","DataType":"20","MinInterval":"0005","MaxInterval":"012c"}},"0300":{"0008":{"TimeStamp":1677401759.986213,"Status":"00","DataType":"30","MinInterval":"0001","MaxInterval":"012c"}}}},"TimeStamp":1677401759.986197},"Capability":["Full-Function Device","Main Powered","Receiver during Idle","Standard security","NwkAddr should be allocated"],"Max Rx":"0052","macapa":"8e","Max Tx":"0052","descriptor_capability":"00","Max Buffer Size":"52","SWBUILD_1":"20210714","server_mask":"2c00","CertifiedDevice":false,"SWBUILD_3":"2.09.00","BindingTable":{"SQN":0,"Status":"Success","TimeStamp":1677341062.072282,"BindingTableSize":11,"Devices":[{"2c1165fffe2d28db":{"sourceEp":"01","Cluster":"2000","targetGroupId":"fffd"}},{"2c1165fffe2d28db":{"sourceEp":"01","Cluster":"0000","targetIEEE":"00158d0001edb6bf","targetNickName":"Zigate","targetEp":"01"}},{"2c1165fffe2d28db":{"sourceEp":"01","Cluster":"0003","targetIEEE":"00158d0001edb6bf","targetNickName":"Zigate","targetEp":"01"}},{"2c1165fffe2d28db":{"sourceEp":"01","Cluster":"0004","targetIEEE":"00158d0001edb6bf","targetNickName":"Zigate","targetEp":"01"}},{"2c1165fffe2d28db":{"sourceEp":"01","Cluster":"0005","targetIEEE":"00158d0001edb6bf","targetNickName":"Zigate","targetEp":"01"}},{"2c1165fffe2d28db":{"sourceEp":"01","Cluster":"0006","targetIEEE":"00158d0001edb6bf","targetNickName":"Zigate","targetEp":"01"}},{"2c1165fffe2d28db":{"sourceEp":"01","Cluster":"0008","targetIEEE":"00158d0001edb6bf","targetNickName":"Zigate","targetEp":"01"}},{"2c1165fffe2d28db":{"sourceEp":"01","Cluster":"0300","targetIEEE":"00158d0001edb6bf","targetNickName":"Zigate","targetEp":"01"}},{"2c1165fffe2d28db":{"sourceEp":"01","Cluster":"1000","targetIEEE":"00158d0001edb6bf","targetNickName":"Zigate","targetEp":"01"}},{"2c1165fffe2d28db":{"sourceEp":"01","Cluster":"fc57","targetIEEE":"00158d0001edb6bf","targetNickName":"Zigate","targetEp":"01"}},{"2c1165fffe2d28db":{"sourceEp":"01","Cluster":"fc82","targetIEEE":"00158d0001edb6bf","targetNickName":"Zigate","targetEp":"01"}}]},"Epv2":{"01":{"ClusterIn":{"1000":{},"0000":{},"0003":{},"0004":{},"0005":{},"0006":{},"0008":{},"0300":{},"fc57":{},"fc82":{}},"ClusterOut":{"0019":{}},"ProfileID":"0104","ZDeviceID":"010c"},"f2":{"ClusterIn":{},"ClusterOut":{"0021":{}},"ProfileID":"a1e0","ZDeviceID":"0061"}},"ConfigSource":"8043","WriteAttributes":{"Ep":{"01":{"1000":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0000":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0003":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0004":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0005":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0006":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0008":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0300":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"fc57":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"fc82":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0019":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}}},"f2":{"0021":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}}}}},"ConfigureReporting":{"Ep":{"01":{"0006":{"TimeStamp":1677341058.1491165,"iSQN":{"0000":20906},"Attributes":{"0000":"00"},"ZigateRequest":{}},"0008":{"TimeStamp":1677341058.1539168,"iSQN":{"0000":20907},"Attributes":{"0000":"00"},"ZigateRequest":{}},"0300":{"TimeStamp":1677341058.158767,"iSQN":{"0008":20908},"Attributes":{"0008":"00"},"ZigateRequest":{}}}}},"ColorInfos":{"X":"28363","Y":"26397","ColorTemperatureMireds":"331","ColorMode":2,"ColorCapabilities":"16"},"Bind":{"01":{"1000":{"Target":"0000","Stamp":1677341059,"Phase":"binded","Status":"00","i_sqn":20902},"0000":{"Target":"0000","Stamp":1677341058,"Phase":"binded","Status":"00","i_sqn":20895},"0003":{"Target":"0000","Stamp":1677341058,"Phase":"binded","Status":"00","i_sqn":20896},"0004":{"Target":"0 [... cut ...] "Status":"00","i_sqn":20899},"0008":{"Target":"0000","Stamp":1677341058,"Phase":"binded","Status":"00","i_sqn":20900},"0300":{"Target":"0000","Stamp":1677341058,"Phase":"binded","Status":"00","i_sqn":20901},"fc57":{"Target":"0000","Stamp":1677341059,"Phase":"binded","Status":"00","i_sqn":20903},"fc82":{"Target":"0000","Stamp":1677341059,"Phase":"binded","Status":"00","i_sqn":20904}}},"GroupMemberShip":{"01":{}},"ConsistencyCheck":"ok","RollingLQI":[129,129,160,132,132,132,115,132,132,132,135]}}`

pipiche38, 2023-03-19:
https://github.com/zigbeefordomoticz/z4d-certified-devices/commit/5bf339ecc59980df10313e8edbdf71a462c32d0a

pipiche38, 2023-03-19:
If you want to test, a fix has been pushed to develop branch.
you need to be with python3.8 or above. In case you are on buster, you must upgrade to bullseye

if you are on python3.8 or above please do the following to switch to develop branch

```
git checkout develop
git pull
sudo python3 -m pip install -r requirements.txt --upgrade
```

And simply restart the plugin

pipiche38, 2023-03-23:
The best would be to redo a full pairing if you can by removing the existing device and the doing the pairing.

LD40, 2023-03-24:
OK after removing and new pairing

closed 2023-03-24 by LD40