zigbeefordomoticz/Domoticz-Zigbee #1567: [Model Certification] - Profalux remote control MAI-ZTS

LD40, 2023-03-17:
Just to have certified/optimized my Profalux remote controlers 
Profalux MAI-ZTP20C (identification on remote) seen as MAI-ZTS by the Zigate
[https://www.profalux-pieces-detachees.com/telecommandes/242-telecommande-individuelle-profalux.html](url)
Currently I do not really use it in Domoticz but could be interesting at least for for battery level and avoid any conflicts/errors if optimized.

Logs:
`{"PluginInfos":{"HardwareID":17,"HomeFolder":"/home/pi/domoticz/plugins/Domoticz-Zigate/","StartupFolder":"/home/pi/domoticz/","UserDataFolder":"/home/pi/domoticz/","WebRoot":"","Database":"/home/pi/domoticz/domoticz.db","Language":"fr","Version":"6.3","Author":"pipiche38","Name":"Zigate","Address":"0.0.0.0","Port":"9999","SerialPort":"/dev/serial/by-id/usb-Prolific_Technology_Inc._USB-Serial_Controller-if00-port0","Key":"Zigate","Mode1":"V1","Mode2":"USB","Mode3":"False","Mode4":"9440","Mode6":"2","DomoticzVersion":"2023.1","DomoticzHash":"f9b9ac774","DomoticzBuildTime":"2023-02-14 15:06:40","PluginBranch":"stable6","PluginVersion":"6.3.011","TimeStamp":1679049352,"available":0,"available-firmMajor":0,"available-firmMinor":0,"FirmwareVersion":"0320","FirmwareUpdate":false,"PluginUpdate":false,"Zigpy":false,"CoordinatorModel":"ZiGate Classic PDM (OptiPDM)","CoordinatorFirmwareVersion":"0320","DisplayFirmwareVersion":"Zigate - 0320","CoordinatorIEEE":"00158d0001edb6bf","NetworkSize":"Total: 9 | Routers: 6 | End Devices: 3","NetworkDevices":{"1110":{"Profalux":["BSO-Profalux","MAI-ZTS"]},"1277":{"Adeo":["ZBEK-14"]}}},"Analytics":1,"Device":{"Version":"3","ZDeviceName":"BSO-Bureau-Telec","Status":"inDB","SQN":"67","Ep":{"01":{"0000":{"4000":"Release","0004":"Profalux","0005":"MAI-ZTS","0000":"3","0001":"25","0003":"29","0006":"Jun 10 2021","0007":"3","0010":"TS","000d":"None","0011":"0","fffd":"1"},"0003":{},"0015":{},"fc02":{},"fc20":{},"0004":{},"0005":{},"0006":{},"0008":{},"0100":{},"fc21":{},"ClusterType":{"382":"LvlControl","383":"BSO-Orientation"}},"02":{"0000":{"4000":"Release","0004":"Profalux","0005":"MAI-ZTS","0000":"3","0001":"25","0003":"29","0006":"Jun 10 2021","0007":"3","0010":"TS","000d":"None","0011":"0","fffd":"1"},"0001":{"0000":321.6,"0020":26,"0035":0,"0036":20,"003e":"00000000","0038":25,"fffd":"1","0037":22},"0002":{},"0003":{},"0009":{},"0015":{},"0020":{},"fc02":{},"fc20":{},"0004":{},"0005":{},"0019":{},"0102":{},"fc21":{},"ClusterType":{"384":"Voltage","385":"Alarm","386":"BSO-Orientation"}}},"Heartbeat":"28321","RIA":"1","LQI":123,"Battery":20,"Model":"MAI-ZTS","ForceAckCommands":[],"MacCapa":"80","IEEE":"20918a0000106d94","Type":"Switch/LvlControl/BSO-Orientation/Voltage/Alarm/WindowCovering","ProfileID":"0104","ZDeviceID":"0203","App Version":"25","Attributes List":{"Ep":{"01":{"0000":{"4000":"42","0000":"20","0001":"20","0003":"20","0004":"42","0005":"42","0006":"42","0007":"30","000d":"42","0010":"42","0011":"30","fffd":"21"},"0003":{"0000":"21","fffd":"21"},"0015":{"2100":"00"}},"02":{"0000":{"4000":"42","0000":"20","0001":"20","0003":"20","0004":"42","0005":"42","0006":"42","0007":"30","000d":"42","0010":"42","0011":"30","fffd":"21"},"0001":{"0000":"21","0020":"20","0035":"18","0036":"20","0037":"20","0038":"20","003e":"1b","fffd":"21"},"0002":{"2900":"00"},"0003":{"0000":"21","fffd":"21"},"0015":{"2100":"00"},"0020":{"2300":"00"}}}},"DeviceType":"RFD","HW Version":"29","Last Cmds":[],"LogicalType":"End Device","Manufacturer":"1110","Manufacturer Name":"Profalux","NbEp":"2","PowerSource":"Battery","ReadAttributes":{"Ep":{"01":{"0000":{"TimeStamp":1672684583,"iSQN":{"4000":7670,"0000":7660,"0001":7661,"0003":7662,"0004":7663,"0005":7664,"0006":7665,"0007":7666,"000d":7667,"0010":7668,"0011":7669,"fffd":7671},"Attributes":{"4000":"00","0002":"86","000a":"86","000f":"86","0015":"86","f000":"86","0000":"00","0001":"00","0003":"00","0005":"00","0004":"00","0006":"00","000d":"00","0010":"00","0007":"00","0011":"00","fffd":"00"},"ZigateRequest":{}},"0100":{"TimeStamp":1633972173,"iSQN":{"0000":1699,"0001":1700,"0002":1701,"0010":1702,"0011":1703},"Attributes":{"0000":"86","0001":"86","0002":"86","0010":"86","0011":"86"},"ZigateRequest":{}},"0006":{"TimeStamp":1633972176,"iSQN":{"4001":1705,"4002":1706,"0000":1704},"Attributes":{"4001":"86","4002":"86","0000":"86"},"ZigateRequest":{}},"0008":{"TimeStamp":1633972177,"iSQN":{"0000":1707},"Attributes":{"0000":"86"},"ZigateRequest":{}}},"02":{"0000":{"TimeStamp":1672684588,"iSQN":{"4000":7682,"0000":7672,"0001":7673,"0003":7674,"0004":7675,"0005":7676,"0006":7677,"0007":7678,"000d":7679,"0010":7680,"0011":7681,"fffd":7683},"Attributes":{"0002":"86","000a":"86","000f":"86","0015":"86","f000":"86","0000":"00","0001":"00","0003":"00","0004":"00","0005":"00","0006":"00"},"ZigateRequest":{}},"0001":{"TimeStamp":1672684660,"iSQN":{"0001":1737,"0003":1738,"0021":1740,"0033":1741,"0000":7684,"0020":7685,"0035":7686,"0036":7687,"0037":7688,"0038":7689,"003e":7690,"fffd":7691},"Attributes":{"0001":"86","0003":"86","0021":"86","0033":"86","0020":"00","0035":"00","0036":"00","0037":"00","003e":"00","fffd":"00","0038":"00"},"ZigateRequest":{}},"0019":{"TimeStamp":1672684659,"iSQN":{"0002":1743,"0001":1211,"0003":1212,"0004":1213,"0005":1214,"0006":1215,"0007":1216,"0008":1217,"000a":1219,"0009":7692},"Attributes":{"0002":"86","0001":"86","0003":"86","0004":"86","0005":"86","0006":"86","0007":"86","0008":"86","000a":"86","0009":"86"},"ZigateRequest":{}},"0102":{"TimeStamp":1633972229,"iSQN":{"fffd":1758},"Attributes":{"0000":"86","0001":"86","0002":"86","0003":"86","0004":"86","0007":"86","0008":"86","0009":"86","000a":"86","000b":"86","0010":"86","0011":"86","0014":"86","0017":"86","fffd":"86"},"ZigateRequest":{}}}}},"ReceiveOnIdle":"","Stack Version":"","Stamp":{"time":1679049112.8605998,"Time":"2023-03-17 11:31:52","MsgType":"8011","LastSeen":1679049112},"ZCL Version":"3","Health":"Live","_rawNodeDescriptor":"1110005200522c000080524002","bitfield":"4002","Capability":["Reduced-Function Device","Standard security","NwkAddr should be allocated"],"Max Rx":"0052","macapa":"80","Max Tx":"0052","descriptor_capability":"00","Max Buffer Size":"52","SWBUILD_1":"Jun 10 2021","BatteryUpdateTime":1672684660,"server_mask":"2c00","CertifiedDevice":false,"SWBUILD_3":"Release","Epv2":{"01":{"ClusterIn":{"0000":{},"0003":{},"0015":{},"fc02":{},"fc20":{}},"ClusterOut":{"0003":{},"0004":{},"0005":{},"0006":{},"0008":{},"0100":{},"fc20":{},"fc21":{}},"ProfileID":"0104","ZDeviceID":"0201"},"02":{"ClusterIn":{"0000":{},"0001":{},"0002":{},"0003":{},"0009":{},"0015":{},"0020":{},"fc02":{},"fc20":{}},"ClusterOut":{"0003":{},"0004":{},"0005":{},"0019":{},"0102":{},"fc20":{},"fc21":{}},"ProfileID":"0104","ZDeviceID":"0203"}},"ConfigSource":"8043","WriteAttributes":{"Ep":{"01":{"0000":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0003":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0015":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"fc02":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"fc20":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0004":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0005":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0006":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0008":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0100":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"fc21":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}}},"02":{"0000":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0001":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0002":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0003":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0009":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0015":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0020":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"fc02":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"fc20":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0004":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0005":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0019":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0102":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"fc21":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}}}}},"ConfigureReporting":{"Ep":{"01":{"0006":{"TimeStamp":1672 [... cut ...] 72159,"Phase":"binded","Status":"00","i_sqn":1689},"0015":{"Target":"0000","Stamp":1633972160,"Phase":"binded","Status":"00","i_sqn":1690},"0020":{"Target":"0000","Stamp":1633972161,"Phase":"binded","Status":"00","i_sqn":1691},"fc02":{"Target":"0000","Stamp":1633972162,"Phase":"binded","Status":"00","i_sqn":1692},"fc20":{"Target":"0000","Stamp":1633972163,"Phase":"binded","Status":"00","i_sqn":1693}}},"Param":{},"ConsistencyCheck":"ok","RollingLQI":[135,135,132,132,61,132,132,126,126,123,123]}}`

pipiche38, 2023-03-17:
To make it short, I see what I can do but the remote are not sending any information to the coordinator, they are binded to the Blind, so even the battery level cannot be reported.

pipiche38, 2023-03-19:
If you want to test, a fix has been pushed to develop branch.
you need to be with python3.8 or above. In case you are on buster, you must upgrade to bullseye

if you are on python3.8 or above please do the following to switch to develop branch

```
git checkout develop
git pull
sudo python3 -m pip install -r requirements.txt --upgrade
```

And simply restart the plugin

LD40, 2023-03-23:
Finally it seems I have a regression on Profalux BSO orientation (perhaps related to other modifications made on dev branch)
```
 2023-03-23 18:48:00.211 Error: Zigate: Call to function 'onCommand' failed, exception details:
2023-03-23 18:48:00.214 Error: Zigate: Traceback (most recent call last):
2023-03-23 18:48:00.214 Error: Zigate: File "/home/pi/domoticz/plugins/Domoticz-Zigate/plugin.py", line 1562, in onCommand
2023-03-23 18:48:00.214 Error: Zigate: _plugin.onCommand(Unit, Command, Level, Hue)
2023-03-23 18:48:00.214 Error: Zigate: File "/home/pi/domoticz/plugins/Domoticz-Zigate/plugin.py", line 881, in onCommand
2023-03-23 18:48:00.214 Error: Zigate: mgtCommand(self, Devices, Unit, Command, Level, Color)
2023-03-23 18:48:00.214 Error: Zigate: File "/home/pi/domoticz/plugins/Domoticz-Zigate/Modules/command.py", line 1179, in mgtCommand
2023-03-23 18:48:00.214 Error: Zigate: profalux_MoveToLiftAndTilt(self, NWKID, tilt=Tilt)
2023-03-23 18:48:00.214 Error: Zigate: File "/home/pi/domoticz/plugins/Domoticz-Zigate/Modules/profalux.py", line 266, in profalux_MoveToLiftAndTilt
2023-03-23 18:48:00.214 Error: Zigate: level = getLevel(self, nwkid)
2023-03-23 18:48:00.214 Error: Zigate: File "/home/pi/domoticz/plugins/Domoticz-Zigate/Modules/profalux.py", line 224, in getLevel
2023-03-23 18:48:00.214 Error: Zigate: level = int(self.ListOfDevices[nwkid]["Ep"]["01"]["0008"]["0000"], 16)
2023-03-23 18:48:00.214 Error: Zigate: TypeError: int() can't convert non-string with explicit base 
```

LD40, 2023-03-24:
That's OK now.

closed 2023-03-24 by LD40