zigbeefordomoticz/Domoticz-Zigbee #1568: Error: zigbee: zcl_attribute_discovery_request not implemented for RAW mode

Achterhoeker, 2023-03-17:
Hi,

After updating from branch issue/1353 to /develop i encounter the following problem.
My Adurosmart motion detector was not working anymore. I'm trying to delete it and re-pair it to see if then it works again. Re-pair gives the following errors in the log:
 
2023-03-17 16:17:15.006 Status: zigbee: NetworkID: ef46 is replacing 30a6 for object: 00158d000195e149
2023-03-17 16:17:16.056 Error: zigbee: zcl_attribute_discovery_request not implemented for RAW mode
2023-03-17 16:17:16.063 Error: zigbee: zcl_attribute_discovery_request not implemented for RAW mode
2023-03-17 16:17:16.070 Error: zigbee: zcl_attribute_discovery_request not implemented for RAW mode
2023-03-17 16:17:16.076 Error: zigbee: zcl_attribute_discovery_request not implemented for RAW mode
2023-03-17 16:17:16.084 Error: zigbee: zcl_attribute_discovery_request not implemented for RAW mode
2023-03-17 16:17:20.044 zigbee: Assisted Pairing: Polling: []

The device is still seen in the zigbee webbased interface, but has been deleted in Domoticz.
What is the best approach for me to tackle this issue ?

Thanks,
Ronald.

pipiche38, 2023-03-17:
this error is not an issue, you need to make sure the device is removed also from the plugin.
If all widgets have been deleted from Domoticz, then you should have a trash icon to remove it
otherwise a restart of the plugin should also make the trick

Achterhoeker, 2023-03-17:
@pipiche38 You were right, there was still an device in Domoticz (battery state) After deleting that one the device was also correctly deleted in the plugin. Repair did now the trick, it is functional again. Thx for your advice, i'll close this ticket.

closed 2023-03-17 by Achterhoeker