zigbeefordomoticz/Domoticz-Zigbee #1569: TS0601-_TZE200_t1blo2bj no battery

OlivierC93360, 2023-03-21:
**Description the bug**
I added the Tuya Alarm detected as TS0601-_TZE200_t1blo2bj and it works as expected including volume, melody, duration, but i can't get the battery level.

**Expected behavior**
I'd like to get the  battery level from device to have it available in the domoticz device.
In the RAW object detail (cf. details below), the Battery entry returns "Object {}" instead of a value.

**Screenshots**
As shown below the battery is not available for the siren but is for the door contact.
[image]

**Plugin (please complete the following information):**
 - OS: Linux BullsEye
 - Plugin version : Version 6.3.011
 - Plugin branch: stable 6
 - Domoticz Version: 2023.1
 - Coordinator Zigate+ V2 USB Firmware: Zigate - 0320

**RAW object**
Version: "3"
ZDeviceName: ""
Status: "inDB"
SQN: "71"
Ep: Object {"01":{"0000":{"0004":"_TZE200_t1blo2bj","0000":"3","0001":"70","0005":"TS0601","0007":"1","fffe":"0","ffe2":"54","ffe4":"0","ffdf":"Q}\u0007+fR}\u0007+\u0012f\u0012eee\u0012è+fè+\u0012è+eè+e"},"0004":{},"ef00":{},"000a":{},"0019":{"0001":"ffffffff","0002":"00000046","0003":"0002","0004":"ffffffff","0005":"ffff","0006":"00","0007":"1141","0008":"ffff","0009":"0000"},"Type":"SwitchAlarm","ClusterType":{"124":"SwitchAlarm"}},"f2":{}}
Heartbeat: "35479"
RIA: "0"
LQI: 81
**Battery: Object {}**
Model: "TS0601-_TZE200_t1blo2bj"
ForceAckCommands: Array[0] []
MacCapa: "8e"
IEEE: "a4c138010ce1807d"
Type: "SwitchAlarm"
ProfileID: "a1e0"
ZDeviceID: "0061"
App Version: "70"
Attributes List: Object {}
DeviceType: "FFD"
HW Version: ""
Last Cmds: Array[0] []
LogicalType: "Router"
Manufacturer: "1141"
Manufacturer Name: "_TZE200_t1blo2bj"
NbEp: "2"
PowerSource: "Main"
ReadAttributes: Object {"Ep":{"01":{"0019":{"TimeStamp":"18/03/2023 18:47:06","iSQN":{"0001":80,"0002":80,"0003":80,"0004":80,"0005":80,"0006":80,"0007":81,"0008":81,"0009":81,"000a":81},"Attributes":{"0001":"00","0002":"00","0003":"00","0004":"00","0005":"00","0006":"00","0007":"00","0008":"00","0009":"00","000a":"86"},"ZigateRequest":{}},"0000":{"TimeStamp":"21/03/2023 23:06:12","iSQN":{"0004":82,"0000":82,"0001":82,"0005":82,"0007":82,"fffe":82},"Attributes":{"0004":"00","0000":"00","0001":"00","0005":"00","0007":"00","fffe":"00","ffe2":"00","ffe4":"00","ffdf":"00"},"ZigateRequest":{}}}}}
ReceiveOnIdle: ""
Stack Version: ""
Stamp: Object {"time":1679436372.9767065,"Time":"2023-03-21 23:06:12","MsgType":"8011","LastSeen":"21/03/2023 23:06:12"}
ZCL Version: "3"
Health: "Live"
descriptor_capability: "00"
bitfield: "4001"
ConfigSource: "DeviceConf"
Capability: Array[5] ["Full-Function Device","Main Powered","Receiver during Idle","Standard security","NwkAddr should be allocated"]
Param: Object {"Disabled":0,"TuyaAlarmDuration":90,"TuyaAlarmLevel":2,"TuyaAlarmMelody":7}
_rawNodeDescriptor: "1141004200422a00008e424001"
Epv2: Object {"01":{"ClusterIn":{},"ClusterOut":{},"ProfileID":{},"ZDeviceID":{}},"f2":{"ClusterIn":{},"ClusterOut":{"0021":{}},"ProfileID":"a1e0","ZDeviceID":"0061"}}
Max Rx: "0042"
BindingTable: Object {"SQN":0,"Status":"Success","TimeStamp":"18/03/2023 18:47:05","BindingTableSize":0,"Devices":[]}
CertifiedDevice: true
server_mask: "2a00"
macapa: "8e"
Max Tx: "0042"
WriteAttributes: Object {"Ep":{"01":{"0000":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0004":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"ef00":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"000a":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0019":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}}}}}
Tuya: Object {"TuyaTransactionId":174}
Max Buffer Size: "42"
CheckParam: false
ConsistencyCheck: "ok"
ZCLSQN: "39"
RollingLQI: Array[11] [85,84,84,84,84,84,84,84,81,81,81]
SQN_000a: "6e"
NwkID: "53d9"

Thanks for your job !!!
Oliv

pipiche38, 2023-03-22:
Could you please provide again the device information as you did ?
As the current was probably done just after the pairing and some information are missing.
Otherwise did you try to remove all widgets in order to repo the device, restart the plugin and then do a reset of the device to repair

OlivierC93360, 2023-03-22:
Hi

thanks for your answer

I removed the devices, added it again, reload the plugin.... as requested and still the same situation no battery level available.

Is there a way to send easyly a RESTAPI (or equivalent) to the device to check the read values ? Using the Tuya SmartPhone app it should provide the battery... i guess there should be a missing information to read somewhere...

**here is the RAW object;**
Version: "3"
ZDeviceName: ""
Status: "inDB"
SQN: "5e"
Heartbeat: "2"
RIA: "0"
LQI: 52
**Battery: Object {}**
Model: "TS0601-_TZE200_t1blo2bj"
ForceAckCommands: Array[0] []
MacCapa: "8e"
IEEE: "a4c138010ce1807d"
Type: "SwitchAlarm"
ProfileID: "a1e0"
ZDeviceID: "0061"
App Version: "70"
Attributes List: Object {}
DeviceType: "FFD"
HW Version: ""
Last Cmds: Array[7] [[2022,"02","2dbd","0100","01010000000000000600040000000100050007fffe"],[2033,"02","2dbd","0100","010100190000000006000100020003000400050006"],[2034,"02","2dbd","0100","010100190000000004000700080009000a"],[2035,"02","2dbd","0100","01010000000000000600040000000100050007fffe"],[2056,"02","fd82","0100","010100190000000006000100020003000400050006"],[2057,"02","fd82","0100","010100190000000004000700080009000a"],[2058,"02","fd82","0100","01010000000000000600040000000100050007fffe"]]
LogicalType: "Router"
Manufacturer: "1141"
Manufacturer Name: "_TZE200_t1blo2bj"
NbEp: "2"
PowerSource: "Main"
ReceiveOnIdle: ""
Stack Version: ""
Stamp: Object {"time":1679479805.1854928,"Time":"2023-03-22 11:10:05","MsgType":"8002","LastSeen":"22/03/2023 11:10:05"}
ZCL Version: "3"
Health: "Live"
Capability: Array[5] ["Full-Function Device","Main Powered","Receiver during Idle","Standard security","NwkAddr should be allocated"]
ZDPSQN: "04"
RollingLQI: Array[11] [52,52,52,52,52,52,52,52,52,52,52]
Param: Object {"Disabled":0,"TuyaAlarmDuration":10,"TuyaAlarmLevel":2,"TuyaAlarmMelody":3}
ConfigSource: "DeviceConf"
Ep: Object {"01":{"0000":{"0004":"_TZE200_t1blo2bj","0000":"3","0001":"70","0005":"TS0601","0007":"1","fffe":"0","ffe2":"54","ffe4":"0","ffdf":"-+f.+\u0012"},"0004":{},"ef00":{},"000a":{},"0019":{"0001":"ffffffff","0002":"00000046","0003":"0002","0004":"ffffffff","0005":"ffff","0006":"00","0007":"1141","0008":"ffff","0009":"0000"},"Type":"SwitchAlarm","ClusterType":{"130":"SwitchAlarm"}},"f2":{}}
Epv2: Object {"01":{"ClusterIn":{},"ClusterOut":{},"ProfileID":{},"ZDeviceID":{}},"f2":{"ClusterIn":{},"ClusterOut":{"0021":{}},"ProfileID":"a1e0","ZDeviceID":"0061"}}
ZDeviceVersion: "0000"
PairingInProgress: false
PairingTime: 1679474049.1090813
CheckParam: true
CertifiedDevice: true
_rawNodeDescriptor: "1141004200422a00008e424001"
Max Buffer Size: "42"
Max Rx: "0042"
Max Tx: "0042"
macapa: "8e"
bitfield: "4001"
server_mask: "2a00"
descriptor_capability: "00"
SQN_000a: "48"
PreviousStatus: "Leave"
BindingTable: Object {"SQN":0,"Status":"Success","TimeStamp":"22/03/2023 09:34:44","BindingTableSize":0,"Devices":[]}
ReadAttributes: Object {"Ep":{"01":{"0019":{"TimeStamp":"22/03/2023 09:34:44","iSQN":{"0001":2056,"0002":2056,"0003":2056,"0004":2056,"0005":2056,"0006":2056,"0007":2057,"0008":2057,"0009":2057,"000a":2057},"Attributes":{"0001":"00","0002":"00","0003":"00","0004":"00","0005":"00","0006":"00","0007":"00","0008":"00","0009":"00","000a":"86"},"ZigateRequest":{}},"0000":{"TimeStamp":"22/03/2023 11:09:46","iSQN":{"0004":2058,"0000":2058,"0001":2058,"0005":2058,"0007":2058,"fffe":2058},"Attributes":{"0004":"00","0000":"00","0001":"00","0005":"00","0007":"00","fffe":"00","ffe2":"00","ffe4":"00","ffdf":"00"},"ZigateRequest":{}}}}}
WriteAttributes: Object {"Ep":{"01":{"0000":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0004":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"ef00":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"000a":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0019":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}}}}}
ZCLSQN: "0d"
Tuya: Object {"TuyaTransactionId":14}
NwkID: "fd82"

pipiche38, 2023-03-22:
This device and I have one works perfectly, the extract you are giving are missing key information.
We cannot read value , we are relying on the device to send the information.  And here the device is not sending its battery level.

Please switch to develop branch and then we will see how/what to debug
you need to have python3.8 at least (no Buster, you must be on Bullseye)

from the Domoticz-Zigbee folder

```
git checkout develop
git pull
sudo python3 -m pip install -r requirements.txt --upgrade
```

Then  redo the pairing, and please provide the device informations as a file the copy paste is not usable

OlivierC93360, 2023-03-22:
I've run the requested command (without any difficulties due to several libpython installed in my box) and re-appaired the device.
Also i upgraded the FirmWare of my ZiGate+V2 to version 3.A0 (previous 3.20)

In the RAW object i can see the battery level:
**RAW**
Version: "3"
ZDeviceName: ""
Status: "inDB"
SQN: "e1"
Heartbeat: "7"
RIA: "0"
LQI: 55
**Battery: 97**
Model: "TS0601-_TZE200_t1blo2bj"
ForceAckCommands: Array[0] []
MacCapa: "8e"
IEEE: "a4c138010ce1807d"
Type: "SwitchAlarm"
ProfileID: "a1e0"
ZDeviceID: "0061"
App Version: 70
Attributes List: Object {}
DeviceType: "FFD"
HW Version: ""
Last Cmds: Array[4] [[12,"02","dae1","0100","01010000000000000600040000000100050007fffe"],[36,"02","dae1","0100","010100190000000006000100020003000400050006"],[37,"02","dae1","0100","010100190000000004000700080009000a"],[38,"02","dae1","0100","01010000000000000600040000000100050007fffe"]]
LogicalType: "Router"
Manufacturer: "1141"
Manufacturer Name: "_TZE200_t1blo2bj"
NbEp: "2"
PowerSource: "Main"
ReadAttributes: Object {"Ep":{"01":{"0000":{"TimeStamp":"22/03/2023 19:48:14","iSQN":{"0004":38,"0000":38,"0001":38,"0005":38,"0007":38,"fffe":38},"Attributes":{"0004":"00","0000":"00","0001":"00","0005":"00","0007":"00","fffe":"00","ffe2":"00","ffe4":"00","ffdf":"00"},"ZigateRequest":{}},"0019":{"TimeStamp":"22/03/2023 19:48:05","iSQN":{"0001":36,"0002":36,"0003":36,"0004":36,"0005":36,"0006":36,"0007":37,"0008":37,"0009":37,"000a":37},"Attributes":{"0001":"00","0002":"00","0003":"00","0004":"00","0005":"00","0006":"00","0007":"00","0008":"00","0009":"00","000a":"86"},"ZigateRequest":{}}}}}
ReceiveOnIdle: ""
Stack Version: ""
Stamp: Object {"time":1679510921.7107093,"Time":"2023-03-22 19:48:41","MsgType":"8011","LastSeen":"22/03/2023 19:48:41"}
ZCL Version: 3
Health: "Live"
Capability: Array[5] ["Full-Function Device","Main Powered","Receiver during Idle","Standard security","NwkAddr should be allocated"]
ZDPSQN: "02"
RollingLQI: Array[11] [55,54,55,54,54,54,55,54,54,55,55]
WriteAttributes: Object {"Ep":{"01":{"0000":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0004":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"ef00":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"000a":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0019":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}}}}}
Param: Object {"TuyaAlarmDuration":10,"TuyaAlarmLevel":0,"TuyaAlarmMelody":1}
ConfigSource: "DeviceConf"
Ep: Object {"01":{"0000":{"0004":"_TZE200_t1blo2bj","0000":3,"0001":70,"0005":"TS0601","0007":1,"fffe":0,"ffe2":"54","ffe4":"0","ffdf":"`\u0016+\u0013\u0017+e\u0017+\u0012"},"0004":{},"ef00":{},"000a":{},"0019":{"0001":"ffffffff","0002":"00000046","0003":"0002","0004":"ffffffff","0005":"ffff","0006":"00","0007":"1141","0008":"ffff","0009":"0000"},"Type":"SwitchAlarm","ClusterType":{"131":"SwitchAlarm"}},"f2":{}}
Epv2: Object {"01":{"ClusterIn":{},"ClusterOut":{},"ProfileID":{},"ZDeviceID":{}},"f2":{"ClusterIn":{},"ClusterOut":{"0021":{}},"ProfileID":"a1e0","ZDeviceID":"0061"}}
ZDeviceVersion: "0000"
PairingInProgress: false
PairingTime: 1679510884.1122813
BindingTable: Object {"SQN":0,"Status":"Success","TimeStamp":"22/03/2023 19:48:04","BindingTableSize":0,"Devices":[]}
CheckParam: true
CertifiedDevice: true
_rawNodeDescriptor: "1141004200422a00008e424001"
Max Buffer Size: "42"
Max Rx: "0042"
Max Tx: "0042"
macapa: "8e"
bitfield: "4001"
server_mask: "2a00"
descriptor_capability: "00"
Tuya: Object {"TUYA_MCU_VERSION_RSP":"43","AlarmLevel":"02","AlarmDuration":"0000000a","Battery":"00000061","AlarmMelody":"05"}
SQN_000a: "f2"
NwkID: "dae1"

In the PlugIn the device is not recognized with Battery
[image]

Furthermore in Domoticz the device has not yet the battery visible as shown below:
[image]

any ideas ??

pipiche38, 2023-03-26:
If I'm not mistaken, the Siren is also Main powered, that is why it is not reported in the Dahsboard on the battery section.

closed 2023-03-26 by pipiche38