zigbeefordomoticz/Domoticz-Zigbee #1571: lumi.remote.b286acn01 (dual button Xiao mi Aqara button. Domoticz device not created. Non optimized

hkhk12, 2023-03-31:
I have a dual button switch (Xiao mi Aqara dual button) with mac address 00158d0002b48bc5

This button is not recognized as a optimized device and creation of the Domoticz device fails.
Attached the export file.

[export.zip](https://github.com/zigbeefordomoticz/Domoticz-Zigbee/files/11120751/export.zip)

The device was recognized when using Zigate in teh past. Not in Zigbee for Domoticz plugin mode.
The similar single button device from same brand is recognized fine.

pipiche38, 2023-03-31:
Please kindly use the Domoticz forum to solve this issue.

Check for instance that Domoticz allow the creation of Widget (Device) .
Hardware/Devices: -> Accept new Hardware Devices must be enabled

hkhk12, 2023-03-31:
Sorry but I don't think thats the issue. Although I'm not sure what you are replying to (my issue of Achterhoekers comment).
I followed the suggestion on how to handle when you have an "none optimized device".
That the domoticz device is not created is just a consequence of an "none optimized device" I believe.

pipiche38, 2023-03-31:
Please do use the forum also,this device is Certified device.
And from the export.json you have provided, the device is in a status "UNKNOW".

hkhk12, 2023-04-02:
After upgrading to latest **stable6** (was on an about month old **dev**) and upgrading domoticz to **2023.1** the device was added correctly.

Achterhoeker, 2023-04-03:
@hkhk12 @pipiche38  I've tested again and can confirm that on the stable6 branch the issue is not reproducible. On the develop branch i was never able to pair it. So the develop branch seems to have an pairing issue for this particular device.

pipiche38, 2023-04-03:
please test against beta6-6.4.412 (thanks for insisting)

Achterhoeker, 2023-04-04:
@pipiche38 beta6-6.4.412 works with pairing again. Thx for fixing. This ticket can be closed for me (I assume it will also work for @hkhk12 )

Pairing log:

> 2023-04-04 21:55:28.360 zigbee: --> Empty queue
> 2023-04-04 21:55:28.311 Status: zigbee: Request Accepting new Hardware for 240 seconds
> 2023-04-04 21:55:28.397 Status: zigbee: Accepting new Hardware: Enable (On)
> 2023-04-04 21:55:38.380 zigbee: Assisted Pairing: Polling: []
> 2023-04-04 21:55:38.381 zigbee: --> Empty queue
> 2023-04-04 21:55:44.023 Status: zigbee: [None] NEW OBJECT: bd1f None
> 2023-04-04 21:55:44.106 Status: zigbee: [0] NEW OBJECT: bd1f Model Name: lumi.remote.b286acn02
> 2023-04-04 21:55:44.124 Status: zigbee: [None] NEW OBJECT: bd1f Request Node Descriptor
> 2023-04-04 21:55:49.195 zigbee: Assisted Pairing: Polling: ['bd1f']
> 2023-04-04 21:55:49.195 zigbee: --> New device: bd1f
> 2023-04-04 21:55:49.195 zigbee: --> inDB , removed bd1f from List
> 2023-04-04 21:55:49.195 zigbee: ------> New Cluster: {'ClusterId': '0000', 'ClusterDesc': 'Basic'}
> 2023-04-04 21:55:49.195 zigbee: ------> New Cluster: {'ClusterId': '0003', 'ClusterDesc': 'Identify'}
> 2023-04-04 21:55:49.195 zigbee: ------> New Cluster: {'ClusterId': '0004', 'ClusterDesc': 'Groups'}
> 2023-04-04 21:55:49.195 zigbee: ------> New Cluster: {'ClusterId': '0005', 'ClusterDesc': 'Scenes'}
> 2023-04-04 21:55:49.195 zigbee: ------> New Cluster: {'ClusterId': '0006', 'ClusterDesc': 'On/Off'}
> 2023-04-04 21:55:49.195 zigbee: ------> New Cluster: {'ClusterId': '0012', 'ClusterDesc': 'Multistate Input (Basic)'}
> 2023-04-04 21:55:49.195 zigbee: ------> New Cluster: {'ClusterId': '0019', 'ClusterDesc': 'Over-the-Air Upgrade'}
> 2023-04-04 21:55:49.195 zigbee: ------> New Cluster: {'ClusterId': 'ffff', 'ClusterDesc': 'Unknown'}
> 2023-04-04 21:55:49.195 zigbee: ------> New Cluster: {'ClusterId': '0001', 'ClusterDesc': 'Power Configuration'}
> 2023-04-04 21:55:49.195 zigbee: ----> New Ep: {'Ep': '01', 'Clusters': [{'ClusterId': '0000', 'ClusterDesc': 'Basic'}, {'ClusterId': '0003', 'ClusterDesc': 'Identify'}, {'ClusterId': '0004', 'ClusterDesc': 'Groups'}, {'ClusterId': '0005', 'ClusterDesc': 'Scenes'}, {'ClusterId': '0006', 'ClusterDesc': 'On/Off'}, {'ClusterId': '0012', 'ClusterDesc': 'Multistate Input (Basic)'}, {'ClusterId': '0019', 'ClusterDesc': 'Over-the-Air Upgrade'}, {'ClusterId': 'ffff', 'ClusterDesc': 'Unknown'}, {'ClusterId': '0001', 'ClusterDesc': 'Power Configuration'}]}
> 2023-04-04 21:55:49.195 zigbee: ----> New Ep: {'Ep': '02', 'Clusters': []}
> 2023-04-04 21:55:49.195 zigbee: ------> New Cluster: {'ClusterId': '0000', 'ClusterDesc': 'Basic'}
> 2023-04-04 21:55:49.195 zigbee: ------> New Cluster: {'ClusterId': '0003', 'ClusterDesc': 'Identify'}
> 2023-04-04 21:55:49.195 zigbee: ------> New Cluster: {'ClusterId': '0006', 'ClusterDesc': 'On/Off'}
> 2023-04-04 21:55:49.195 zigbee: ------> New Cluster: {'ClusterId': '000c', 'ClusterDesc': 'Analog Input (basic)'}
> 2023-04-04 21:55:49.195 zigbee: ------> New Cluster: {'ClusterId': '0012', 'ClusterDesc': 'Multistate Input (Basic)'}
> 2023-04-04 21:55:49.195 zigbee: ----> New Ep: {'Ep': '03', 'Clusters': [{'ClusterId': '0000', 'ClusterDesc': 'Basic'}, {'ClusterId': '0003', 'ClusterDesc': 'Identify'}, {'ClusterId': '0006', 'ClusterDesc': 'On/Off'}, {'ClusterId': '000c', 'ClusterDesc': 'Analog Input (basic)'}, {'ClusterId': '0012', 'ClusterDesc': 'Multistate Input (Basic)'}]}
> 2023-04-04 21:55:49.196 zigbee: --> New Device: {'NwkId': 'bd1f', 'ProvisionStatus': 'inDB', 'ProvisionStatusDesc': 'inDB', 'IEEE': '00158d000890dfa3', 'ProfileId': '', 'ProfileIdDesc': 'Unknow', 'ZDeviceID': '', 'ZDeviceIDDesc': 'Unknow', 'Model': 'lumi.remote.b286acn02', 'PluginCertified': 'yes', 'Ep': [{'Ep': '01', 'Clusters': [{'ClusterId': '0000', 'ClusterDesc': 'Basic'}, {'ClusterId': '0003', 'ClusterDesc': 'Identify'}, {'ClusterId': '0004', 'ClusterDesc': 'Groups'}, {'ClusterId': '0005', 'ClusterDesc': 'Scenes'}, {'ClusterId': '0006', 'ClusterDesc': 'On/Off'}, {'ClusterId': '0012', 'ClusterDesc': 'Multistate Input (Basic)'}, {'ClusterId': '0019', 'ClusterDesc': 'Over-the-Air Upgrade'}, {'ClusterId': 'ffff', 'ClusterDesc': 'Unknown'}, {'ClusterId': '0001', 'ClusterDesc': 'Power Configuration'}]}, {'Ep': '02', 'Clusters': []}, {'Ep': '03', 'Clusters': [{'ClusterId': '0000', 'ClusterDesc': 'Basic'}, {'ClusterId': '0003', 'ClusterDesc': 'Identify'}, {'ClusterId': '0006', 'ClusterDesc': 'On/Off'}, {'ClusterId': '000c', 'ClusterDesc': 'Analog Input (basic)'}, {'ClusterId': '0012', 'ClusterDesc': 'Multistate Input (Basic)'}]}]}
> 2023-04-04 21:55:59.225 zigbee: Assisted Pairing: Polling: []
> 2023-04-04 21:55:59.225 zigbee: --> Empty queue

commit 0fc4d9f3 (2023-04-03): fix #1571
closed 2023-04-04 by pipiche38