zigbeefordomoticz/Domoticz-Zigbee #1575: Woox 4 button Zigbee remote control

olive6707, 2023-04-06:
1. Please do provide device information such as Brand and Model name. Woox zigbee smart remote : ref R7054
1. Provide the URL where you have purchase the device https://www.domadoo.fr/en/peripheriques/5684-woox-zigbee-30-4-button-smart-remote-control-8435606701136.html
1. Finaly copy-paste all information provided in the log
{"Version":"3","ZDeviceName":"","Status":"inDB","SQN":"44","Ep":{"01":{"0000":{"0004":"_TYZB01_ynsiasng","0005":"TS0219","0007":"2","0000":"3","0001":"80","0002":"2","0003":"2","0006":"20190430"},"0001":{"0020":35,"0021":1},"0003":{},"0004":{},"0005":{},"0500":{"0002":"alarm1: 0, alarm2: 0, tamper: 0, batter: 0, srepor: 1, rrepor: 0, troubl: 0, acmain: 0, test: 0, batdef: 0, doorbell: 0","0000":"01","0001":"0225","0010":"8cf681fffec7d318"},"0502":{"0000":"240","0001":"50","0002":"50","0003":"00"},"eeff":{},"0019":{},"ClusterType":{"183":"Voltage","185":"AlarmWD"}}},"Heartbeat":"62708","RIA":"4","LQI":80,"Battery":1,"Model":"TS0219","ForceAckCommands":[],"MacCapa":"8e","IEEE":"847127fffe096c12","Type":"Voltage/Door/AlarmWD","ProfileID":"0104","ZDeviceID":"0403","App Version":"80","Attributes List":{"Request":{"01":{"0000":1673266568.06221,"0001":1673266568.0625045,"0003":1673266568.0627959,"0004":1673266568.0630865,"0005":1673266568.0633755,"0500":1673266568.0636678,"0502":1673266568.0639567,"eeff":0}}},"DeviceType":"FFD","HW Version":"2","Last Cmds":[],"LogicalType":"Router","Manufacturer":1002,"Manufacturer Name":"_TYZB01_ynsiasng","NbEp":"1","PowerSource":"Main","ReadAttributes":{"Ep":{"01":{"0000":{"TimeStamp":1673262809,"iSQN":{"4000":"10","0004":"0e","0005":"0e","0000":"00","0001":"0e","0002":"0e","0003":"0f","0006":"0f","0007":"0f","000a":"0f","000f":"0f","0010":"10","0015":"10","f000":"10"},"Attributes":{"4000":"86","0004":"00","0005":"00","0000":"00","0001":"00","0002":"00","0003":"00","0006":"00","0007":"00","000a":"86","0010":"86","0015":"86"},"ZigateRequest":{}},"0001":{"TimeStamp":1673284841,"iSQN":{"0000":"09","0001":"09","0003":"09","0020":"09","0021":"09","0033":"0a","0035":"0a"},"Attributes":{"0000":"86","0001":"86","0003":"86","0020":"00","0021":"00","0033":"86"},"ZigateRequest":{}},"0500":{"TimeStamp":1673284841,"iSQN":{},"Attributes":{"0002":"00","0000":"00","0001":"00","0010":"00"},"ZigateRequest":{}},"0502":{"TimeStamp":1673284719,"iSQN":{"0000":"0d"},"Attributes":{"0001":"00","0002":"00","0003":"00","0000":"00"},"ZigateRequest":{}},"0019":{"TimeStamp":1673251160,"iSQN":{"0001":"0b","0002":"0b","0003":"0b","0004":"0b","0005":"0b","0006":"0c","0007":"0c","0008":"0c","0009":"0c","000a":"0c"},"Attributes":{"0001":"86","0002":"86","0003":"86","0004":"86","0006":"86","0007":"86","0008":"86","0009":"86"},"ZigateRequest":{}}}}},"ReceiveOnIdle":"On","Stack Version":"2","Stamp":{"time":1673284841.7048151,"Time":"2023-01-09 18:20:41","MsgType":"8102","LastSeen":1673284841,"LastPing":1680785579},"ZCL Version":"3","Health":"Not Reachable","ConfigSource":"8043","BatteryUpdateTime":1673284841,"IAS":{"01":{"EnrolledStatus":1,"ZoneType":549,"ZoneTypeName":"standard_warning","ZoneStatus":{"alarm1":0,"alarm2":0,"tamper":0,"battery":0,"Support Reporting":1,"Restore Reporting":0,"trouble":0,"acmain":0,"test":0,"battdef":0,"doorbell":0,"GlobalInfos":"0;0;0;0;1;0;0;0;0;0;0","TimeStamp":1673284841}},"Auto-Enrollment":{"Status":"Enrolled","Ep":{"01":{"Status":"Enrolled","TimeStamp":1673251138.0952215,"ZoneId":"00"}}}},"CertifiedDevice":false,"Epv2":{"01":{"ClusterIn":{"0000":{},"0001":{},"0003":{},"0004":{},"0005":{},"0500":{},"0502":{},"eeff":{}},"ClusterOut":{"0003":{},"0019":{}},"ProfileID":"0104","ZDeviceID":"0403"}},"Bind":{"01":{"0500":{"Target":"0000","Stamp":1673251141,"Phase":"binded","Status":"00","i_sqn":"03"},"0000":{"Target":"0000","Stamp":1673251153,"Phase":"binded","Status":"00","i_sqn":"04"},"0001":{"Target":"0000","Stamp":1673251153,"Phase":"binded","Status":"00","i_sqn":"05"},"0003":{"Target":"0000","Stamp":1673251154,"Phase":"binded","Status":"8c","i_sqn":"06"},"0004":{"Target":"0000","Stamp":1673251154,"Phase":"binded","Status":"8c","i_sqn":"07"},"0005":{"Target":"0000","Stamp":1673251155,"Phase":"binded","Status":"8c","i_sqn":"08"},"0502":{"Target":"0000","Stamp":1673251155,"Phase":"binded","Status":"8c","i_sqn":"09"},"eeff":{"Target":"0000","Stamp":1673251156,"Phase":"binded","Status":"8c","i_sqn":"0a"}}},"ConfigureReporting":{"Ep":{"01":{"0500":{"TimeStamp":1673251153.175188,"iSQN":{"0000":"06","0001":"06","0002":"06"},"Attributes":{"0000":"00","0001":"00","0002":"00"},"ZigateRequest":{}},"0502":{"TimeStamp":1673251153.1822715,"iSQN":{"0000":"07"},"Attributes":{"0000":"00"},"ZigateRequest":{}},"0001":{"TimeStamp":1673251153.1843574,"iSQN":{"0000":"08","0020":"08","0021":"08"},"Attributes":{"0000":"86","0020":"8d","0021":"8d"},"ZigateRequest":{}}}}},"SWBUILD_1":"20190430","BindingTable":{"SQN":0,"Status":"Success","TimeStamp":1673251156.8966572,"BindingTableSize":3,"Devices":[{"847127fffe096c12":{"sourceEp":"01","Cluster":"0500","targetIEEE":"8cf681fffec7d318","targetNickName":"Zigbee Coordinator","targetEp":"01"}},{"847127fffe096c12":{"sourceEp":"01","Cluster":"0000","targetIEEE":"8cf681fffec7d318","targetNickName":"Zigbee Coordinator","targetEp":"01"}},{"847127fffe096c12":{"sourceEp":"01","Cluster":"0001","targetIEEE":"8cf681fffec7d318","targetNickName":"Zigbee Coordinator","targetEp":"01"}}]},"ReadConfigureReporting":{"Ep":{"01":{"0500":{"0000":{"TimeStamp":1673251162.9554746,"Status":"00","DataType":"30","MinInterval":"003c","MaxInterval":"0384"},"0001":{"TimeStamp":1673251162.9560387,"Status":"00","DataType":"31","MinInterval":"003c","MaxInterval":"0384"},"0002":{"TimeStamp":1673251162.9564912,"Status":"00","DataType":"19","MinInterval":"003c","MaxInterval":"0384"}},"0502":{"0000":{"TimeStamp":1673251163.3578336,"Status":"00","DataType":"21","MinInterval":"003c","MaxInterval":"0384","Change":"0001"}}}},"TimeStamp":1673251163.3578196},"WriteAttributes":{"Ep":{"01":{"0000":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0001":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0003":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0004":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0005":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0500":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0502":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"eeff":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0019":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}}}}},"Capability":["Full-Function Device","Main Powered","Receiver during Idle","Standard security","NwkAddr should be allocated"],"Log_UnknowDeviceFlag":1673255184.329431,"IASBattery":100,"ConsistencyCheck":"ok","ZCLSQN":"00","pingDeviceRetry":{"Retry":0,"TimeStamp":1680785588}}

olive6707, 2023-04-06:
Erratum, bellow right infomation 
{"Version":"3","ZDeviceName":"","Status":"inDB","SQN":"2a","Ep":{"01":{"0000":{"0004":"_TZ3000_0zrccfgx","0005":"TS0215A","0000":"3","0001":"68","0002":"0","0003":"1","0006":"","0007":"3"},"0001":{"0000":0,"0020":30,"0021":200},"0003":{},"0501":{},"0019":{"0001":"ffffffff","0002":"00000044","0003":"0002","0004":"ffffffff","0005":"ffff","0006":"00","0007":"1141","0008":"ffff","0009":"0000"},"000a":{},"ClusterType":{"275":"Voltage","276":"Door"},"0500":{"0000":"01","0001":"010f","0002":"alarm1: 0, alarm2: 0, tamper: 0, batter: 0, srepor: 0, rrepor: 0, troubl: 0, acmain: 0, test: 0, batdef: 0, doorbell: 0","0010":"8cf681fffec7d318"}}},"Heartbeat":"302","RIA":"4","LQI":176,"Battery":200,"Model":"TS0215A","ForceAckCommands":[],"MacCapa":"80","IEEE":"a4c138a296abbb20","Type":"Voltage/Door","ProfileID":"0104","ZDeviceID":"0401","App Version":"68","Attributes List":{},"DeviceType":"RFD","HW Version":"1","Last Cmds":[],"LogicalType":"End Device","Manufacturer":1002,"Manufacturer Name":"_TZ3000_0zrccfgx","NbEp":"1","PowerSource":"Battery","ReadAttributes":{"Ep":{"01":{"0001":{"TimeStamp":1680785383,"iSQN":{"0000":"17","0001":"17","0003":"17","0020":"18","0021":"18","0033":"18","0035":"19"},"Attributes":{"0000":"00","0001":"86","0020":"00","0021":"00","0033":"86","0035":"86"},"ZigateRequest":{}},"0019":{"TimeStamp":1680785387,"iSQN":{"0001":"1a","0002":"1a","0003":"1a","0004":"1b","0005":"1b","0006":"1b","0007":"1c","0008":"1c","0009":"1c","000a":"1d"},"Attributes":{"0001":"00","0002":"00","0003":"00","0004":"00","0005":"00","0006":"00","0007":"00","0008":"00","0009":"00","000a":"86"},"ZigateRequest":{}},"0000":{"TimeStamp":1680785392,"iSQN":{"4000":"22","0004":"1e","0005":"1e","0000":"1e","0001":"1f","0002":"1f","0003":"1f","0006":"20","0007":"20","000a":"20","000f":"21","0010":"21","0015":"21","f000":"22"},"Attributes":{"4000":"86","0004":"00","0005":"00","0000":"00","0001":"00","0002":"00","0003":"00","0006":"00","0007":"00","000a":"86","000f":"86","0010":"86"},"ZigateRequest":{}},"0500":{"TimeStamp":1680785408,"iSQN":{},"Attributes":{"0000":"00","0001":"00","0002":"00","0010":"00"},"ZigateRequest":{}}}}},"ReceiveOnIdle":"Off","Stack Version":"0","Stamp":{"time":1680785408.6205704,"Time":"2023-04-06 14:50:08","MsgType":"8011","LastSeen":1680785408},"ZCL Version":"3","Health":"","ConfigSource":"8043","BatteryUpdateTime":1680785382,"IAS":{"Auto-Enrollment":{"Status":"Enrolled","Ep":{"01":{"Status":"Enrolled","TimeStamp":1680785373.75823,"ZoneId":"00"}}},"01":{"EnrolledStatus":1,"ZoneType":271,"ZoneTypeName":"remote_control","ZoneStatus":{"alarm1":0,"alarm2":0,"tamper":0,"battery":0,"Support Reporting":0,"Restore Reporting":0,"trouble":0,"acmain":0,"test":0,"battdef":0,"doorbell":0,"GlobalInfos":"0;0;0;0;0;0;0;0;0;0;0","TimeStamp":1680785406}}},"CertifiedDevice":false,"Epv2":{"01":{"ClusterIn":{"0001":{},"0003":{},"0500":{},"0000":{}},"ClusterOut":{"0501":{},"0019":{},"000a":{}},"ProfileID":"0104","ZDeviceID":"0401"}},"Bind":{"01":{"0001":{"Target":"0000","Stamp":1680785375,"Phase":"binded","Status":"00","i_sqn":"0c"},"0003":{"Target":"0000","Stamp":1680785376,"Phase":"binded","Status":"00","i_sqn":"0d"},"0000":{"Target":"0000","Stamp":1680785377,"Phase":"binded","Status":"00","i_sqn":"0e"},"0500":{"Target":"0000","Stamp":1680785407,"Phase":"binded","Status":"00","i_sqn":"13"}}},"ConfigureReporting":{"Ep":{"01":{"0500":{"TimeStamp":1680785373.7760398,"iSQN":{"0000":"15","0001":"15","0002":"15"},"Attributes":{"0000":"00","0001":"00","0002":"00","100e":"86"},"ZigateRequest":{}},"0001":{"TimeStamp":1680785373.7791643,"iSQN":{"0000":"16","0020":"16","0021":"16"},"Attributes":{"0000":"00","0020":"8d","0021":"8d"},"ZigateRequest":{}}}}},"SWBUILD_1":"","BindingTable":{"SQN":0,"Status":"Success","TimeStamp":1680785396.8399873,"BindingTableSize":4,"Devices":[{"a4c138a296abbb20":{"sourceEp":"01","Cluster":"0001","targetIEEE":"8cf681fffec7d318","targetNickName":"Zigbee Coordinator","targetEp":"01"}},{"a4c138a296abbb20":{"sourceEp":"01","Cluster":"0003","targetIEEE":"8cf681fffec7d318","targetNickName":"Zigbee Coordinator","targetEp":"01"}},{"a4c138a296abbb20":{"sourceEp":"01","Cluster":"0000","targetIEEE":"8cf681fffec7d318","targetNickName":"Zigbee Coordinator","targetEp":"01"}},{"a4c138a296abbb20":{"sourceEp":"01","Cluster":"0500","targetIEEE":"8cf681fffec7d318","targetNickName":"Zigbee Coordinator","targetEp":"01"}}]},"ReadConfigureReporting":{"Ep":{"01":{"0500":{"0000":{"TimeStamp":1680785401.4120767,"Status":"00","DataType":"30","MinInterval":"003c","MaxInterval":"0384"},"0001":{"TimeStamp":1680785401.4125264,"Status":"00","DataType":"31","MinInterval":"003c","MaxInterval":"0384"},"0002":{"TimeStamp":1680785401.4129677,"Status":"00","DataType":"19","MinInterval":"003c","MaxInterval":"0384"}},"0001":{"0000":{"TimeStamp":1680785402.4805148,"Status":"00","DataType":"21","MinInterval":"012c","MaxInterval":"fffe","Change":"0001"},"0020":{"TimeStamp":1680785402.4807293,"Status":"8b"},"0021":{"TimeStamp":1680785402.4809353,"Status":"8b"}}}},"TimeStamp":1680785402.4809275},"WriteAttributes":{"Ep":{"01":{"0500":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0000":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0001":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0003":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0501":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0019":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"000a":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}}}}},"Capability":["Reduced-Function Device","Standard security","NwkAddr should be allocated"],"IASBattery":100,"ConsistencyCheck":"ok","Log_UnknowDeviceFlag":1680785574.1623049}

pipiche38, 2023-04-08:
is the device working correctly ?

olive6707, 2023-04-08:
Hi pipiche

No it doesn't work.
2 devices created in domoticz, but no selector. See enclosed id=
a4c138a296abbb20
One Switch and one voltage.
No effect with thé remote control
Thanks for your help

pipiche38, 2023-04-08:
please switch to 'develop' branch

```
git checkout develop
git pull
sudo python3 -m pip install -r requirements.txt --upgrade
```

then  you need to remove the 2 widgets, make sure that the device has been remove from the plugin

then restart the plugin

and do the pairing

olive6707, 2023-04-08:
Ok
I am now absent for the easter WE. I will do it next week and take you
informe
Rgds

pipiche38, 2023-04-11:
please provides logs ? ( Domoticz logs and plugin logs )

olive6707, 2023-04-11:
[PluginZigbee_log_error_history.json.txt](https://github.com/zigbeefordomoticz/Domoticz-Zigbee/files/11203208/PluginZigbee_log_error_history.json.txt)

olive6707, 2023-04-11:
don t know where to find domoticz log file

pipiche38, 2023-04-11:
Plugin logs are available under the Logs/ folder
For domoticz it depends how you have configure it

olive6707, 2023-04-11:
maybe it can help : https://www.zigbee2mqtt.io/devices/TS0215A_remote.html#tuya-ts0215a_remote

pipiche38, 2023-04-12:
please follow https://github.com/zigbeefordomoticz/Domoticz-Zigbee/issues/1575#issuecomment-1500890812
the logs you provided shows that you are on the plugin version 6.3.008 ;, which is an out-dated version of stable6 and you need to be on the develop branch

olive6707, 2023-04-12:
Afer redo it I check the branch and I am always under stable 6 release
here bellow error message after git checkout command

sudo git checkout develop
error: Your local changes to the following files would be overwritten by checkout:
	.github/FUNDING.yml
	.github/stale.yml
	.github/workflows/codeql-analysis.yml
	.github/workflows/jsoncheck.yml
	.github/workflows/lint_python.yml
	.gitmodules
	.hidden/VERSION
	Classes/DomoticzDB.py
	Classes/GroupMgtv2/GrpDomoticz.py
	Classes/GroupMgtv2/GrpResponses.py
	Classes/IAS.py
	Classes/LoggingManagement.py
	Classes/NetworkEnergy.py
	Classes/NetworkMap.py
	Classes/OTA.py
	Classes/PluginConf.py
	Classes/WebServer/WebServer.py
	Classes/WebServer/dispatcher.py
	Classes/WebServer/onMessage.py
	Classes/WebServer/rest_Casaia.py
	Classes/WebServer/rest_CfgReporting.py
	Classes/WebServer/rest_Groups.py
	Classes/WebServer/rest_Ota.py
	Classes/WebServer/rest_PluginUpgrade.py
	Classes/WebServer/rest_Topology.py
	Classes/WebServer/rest_ZLinky.py
	Classes/ZigateTransport/Transport.py
	Classes/ZigateTransport/decode8012.py
	Classes/ZigateTransport/handleProtocol.py
	Classes/ZigateTransport/readDecoder.py
	Classes/ZigateTransport/readwriteTcp.py
	Classes/ZigateTransport/writerThread.py
	Classes/ZigpyTransport/AppBellows.py
	Classes/ZigpyTransport/AppDeconz.py
	Classes/ZigpyTransport/AppGeneric.py
	Classes/ZigpyTransport/AppZigate.py
	Classes/ZigpyTransport/AppZnp.py
	Classes/ZigpyTransport/Transport.py
	Classes/ZigpyTransport/instrumentation.py
	Classes/ZigpyTransport/zigpyThread.py
	Conf/Certified/00Local/.PRECIOUS
	Conf/Certified/Adeo/LDSENK10.json
	Conf/Certified/Bitron/90201021A.json
	Conf/Certified/Bitron/90201021B.json
	Conf/Certified/Bitron/90201024.json
	Conf/Certified/Bitron/AV201021B.json
	Conf/Certified/Bitron/AV201024A.json
	Conf/Certified/Bitron/AV201029A.json
	Conf/Certified/CLEODE/ZHUM.json
	Conf/Certified/CLEODE/ZMOVE.json
	Conf/Certified/CLEODE/ZPLUG.json
	Conf/Certified/Casa.ia/CAC221.json
	Conf/Certified/Casa.ia/CCB432.json
	Conf/Certified/Casa.ia/CDWS312.json
	Conf/Certified/Casa.ia/CKF204.json
	Conf/Certified/Casa.ia/CKF205.json
	Conf/Certified/Casa.ia/CMS323.json
	Conf/Certified/Casa.ia/CPB206.json
	Conf/Certified/Casa.ia/CPC321.json
	Conf/Certified/Casa.ia/CPR412-E.json
	Conf/Certified/Casa.ia/CPR412.json
	Conf/Certified/Casa.ia/CSAC451-E.json
	Conf/Certified/Casa.ia/CSAC451-WTC-E.json
	Conf/Certified/Casa.ia/CSLC601-D-E.json
	Conf/Certified/Casa.ia/CSLC601-E.json
	Conf/Certified/Casa.ia/CSLC631-3.json
	Conf/Certified/Casa.ia/CSP403.json
	Conf/Certified/Casa.ia/CTHS317ET.json
	Conf/Certified/Casa.ia/Dual_Relay_Module.json
	Conf/Certified/Danalock/V3-BTZB.json
	Conf/Certified/Danfoss/RV001.json
	Conf/Certified/Danfoss/eT093WRO.json
	Conf/Certified/Danfoss/eTRV0100.json
	Conf/Certified/Develco/HESZB120.json
	Conf/Certified/Develco/HMSZB-110.json
	Conf/Certified/Develco/MOSZB-140.json
	Conf/Certified/Develco/SMSZB-120.json
	Conf/Certified/Develco/SPLZB-131.json
	Conf/Certified/Develco/SPLZB-132.json
	Conf/Certified/Develco/ZHEMI101.json
	Conf/Certified/ENKI-LEXMAN/LXEK-1.json
	Conf/Certified/ENKI-LEXMAN/LXEK-2.json
	Conf/Certified/ENKI-LEXMAN/LXEK-7.json
	Conf/Certified/Eurotronics/SPZB0001.json
	Conf/Certified/GLEDOPTO/GL-B-007Z.json
	Conf/Certified/GLEDOPTO/GL-B-008Z.json
	Conf/Certified/GLEDOPTO/GL-C-007-2ID.json
	Conf/Certified/GLEDOPTO/GL-C-007.json
	Conf/Certified/GLEDOPTO/GL-C-008.json
	Conf/Certified/GLEDOPTO/GL-C-008S.json
	Conf/Certified/GLEDOPTO/GL-C-008Z.json
	Conf/Certified/GLEDOPTO/GL-C-009.json
	Conf/Certified/GLEDOPTO/GL-D-003Z.json
	Conf/Certified/GLEDOPTO/GL-FL-004TZS.json
	Conf/Certified/GLEDOPTO/GL-MC-001.json
	Conf/Certified/GLEDOPTO/GLEDOPTO.json
	Conf/Certified/Heiman/COSensor-EF-3.0.json
	Conf/Certified/Heiman/COSensor-EM.json
	Conf/Certified/Heiman/COSensor-N.json
	Conf/Certified/Heiman/CO_V15.json
	Conf/Certified/Heiman/CO_YDLV10.json
	Conf/Certified/Heiman/GASSensor-EF-3.0.json
	Conf/Certified/Heiman/GAS_V15.json
	Conf/Certified/Heiman/RC-EF-3.0.json
	Conf/Certified/Heiman/RC-EM.json
	Conf/Certified/Heiman/SMOK_V16.json
	Conf/Certified/Heiman/SMOK_YDLV10.json
	Conf/Certified/Heiman/SmokeSensor-EF-3.0.json
	Conf/Certified/Heiman/SmokeSensor-EM.json
	Conf/Certified/Heiman/SmokeSensor-N-3.0.json
	Conf/
Aborting

pipiche38, 2023-04-12:
can you try the following command

`Tools/big-clean-git.sh`

and then redo
```
git checkout develop
git pull
sudo python3 -m pip install -r requirements.txt --upgrade
```

olive6707, 2023-04-12:
Tools directory doesn't exist

olive6707, 2023-04-13:
ok I installed again the plugin from scratched switch to develop branch.
here the log
[_PluginZigbee_28.log](https://github.com/zigbeefordomoticz/Domoticz-Zigbee/files/11225383/_PluginZigbee_28.log)

pipiche38, 2023-04-14:
This is still 6.3. which is used. Don't you have several vertsion of the plugin under plugins/ folder ?

olive6707, 2023-04-14:
very strange:
I did again:
- stop domoticz, switch to branch stable, start domoticz: log domoticz = ok
- stop domoticz, switch to branch develop, start domoticz: error, see bellow: seems to have an issue with python

2023-04-14 10:45:44.167 Error: Zigbee: (Zigate) failed to load 'plugin.py', Python Path used was '/home/pi/domoticz/plugins/Domoticz-Zigbee/:/usr/lib/python39.zip:/usr/lib/python3.9:/usr/lib/python3.9/lib-dynload:/usr/local/lib/python3.9/dist-packages:/usr/lib/python3/dist-packages:/usr/lib/python3.9/dist-packages'.
2023-04-14 10:45:44.170 Error: Zigbee: Traceback (most recent call last):
2023-04-14 10:45:44.170 Error: Zigbee: File "/home/pi/domoticz/plugins/Domoticz-Zigbee/plugin.py", line 98, in <module>
2023-04-14 10:45:44.170 Error: Zigbee: import z4d_certified_devices
2023-04-14 10:45:44.170 Error: Zigbee: ModuleNotFoundError: No module named 'z4d_certified_devices'

pipiche38, 2023-04-14:
looks like you are missing a python module
you need to run this command from the plugin home folder

`sudo python3 -m pip install -r requirements.txt --upgrade`

olive6707, 2023-04-14:
already done but attribute error, and no change on domoticz log
[image]

pipiche38, 2023-04-14:
`sudo python3 -m pip install pip pyopenssl --upgrade`
`sudo python3 -m pip install -r requirements.txt --upgrade`

olive6707, 2023-04-14:
no change

pipiche38, 2023-04-14:
`sudo python3 -m pip install wheel`
and then
`sudo python3 -m pip install -r requirements.txt --upgrade`

olive6707, 2023-04-14:
no effect, I also tried apt update and apt upgrade
[image]
[image]

olive6707, 2023-04-14:
I think I will re install python3, which release is it suitable?

pipiche38, 2023-04-14:
Python3.9 is fine. 
Do you have a lite-raspbian version ?
If that is the case, I think you should install also the following

git-all, python3-dev et python3-pip

olive6707, 2023-04-14:
no it is Raspbian linux11 (bullseye)

olive6707, 2023-04-15:
Hi pipiche
After numerous trouble (broke my instalation difficulties to come back with a backup.) I finaly resolved my issue with python thanks to google:

udo rm -rf /usr/lib/python3/dist-packages/OpenSSL
sudo pip3 install pyopenssl
sudo pip3 install pyopenssl --upgrade

olive6707, 2023-04-15:
now I have some errors in the domoticz log when starting
2023-04-15 14:21:48.025 Error: Zigbee: Updating path from /home/pi/domoticz/plugins/Domoticz-Zigbee/ to /home/pi/domoticz/plugins/Domoticz-Zigbee is required, but no backward compatibility
2023-04-15 14:21:48.025 Error: Zigbee: Updating path from /home/pi/domoticz/plugins/Domoticz-Zigbee/ to /home/pi/domoticz/plugins/Domoticz-Zigbee is required, but no backward compatibility
2023-04-15 14:21:48.025 Error: Zigbee: Updating path from /home/pi/domoticz/plugins/Domoticz-Zigbee/Data/ to /home/pi/domoticz/plugins/Domoticz-Zigbee/Data is required, but no backward compatibility
2023-04-15 14:21:48.025 Error: Zigbee: Updating path from /home/pi/domoticz/plugins/Domoticz-Zigbee/Conf/ to /home/pi/domoticz/plugins/Domoticz-Zigbee/Conf is required, but no backward compatibility
2023-04-15 14:21:48.025 Error: Zigbee: Updating path from /home/pi/domoticz/plugins/Domoticz-Zigbee/OTAFirmware/ to /home/pi/domoticz/plugins/Domoticz-Zigbee/OTAFirmware is required, but no backward compatibility
2023-04-15 14:21:48.026 Error: Zigbee: Updating path from /home/pi/domoticz/plugins/Domoticz-Zigbee/Reports/ to /home/pi/domoticz/plugins/Domoticz-Zigbee/Reports is required, but no backward compatibility
2023-04-15 14:21:48.026 Error: Zigbee: Updating path from /home/pi/domoticz/plugins/Domoticz-Zigbee/www/ to /home/pi/domoticz/plugins/Domoticz-Zigbee/www is required, but no backward compatibility
2023-04-15 14:21:48.026 Error: Zigbee: Updating path from /home/pi/domoticz/plugins/Domoticz-Zigbee/Logs/ to /home/pi/domoticz/plugins/Domoticz-Zigbee/Logs is required, but no backward compatibility

pipiche38, 2023-04-15:
The error messages are normal.
as regards to the device , 

are you sure that the device is not in the plugin ? Check via the Plugin WebUI Device Management ?
Make sure to remove it, then  restart the plugin and finaly redo the pairing

olive6707, 2023-04-15:
I confirm no device created and devices are enables in domoticz

olive6707, 2023-04-15:
[_PluginZigbee_28 (2).log](https://github.com/zigbeefordomoticz/Domoticz-Zigbee/files/11239640/_PluginZigbee_28.2.log)
one device created after a new trial

olive6707, 2023-04-15:
and the device is a selector with 4 levels we are progressing

olive6707, 2023-04-18:
after switching to stable branch device still reconized and s/eems works well : the four button change the selector level.
Thanks for your help. Now I have an issue with another zigbee euipment (a siren). I think i will open a diffetent issue_

olive6707, 2023-04-18:
I had some random issues with other devices but maybe nothing to do with the develop banch
I just switch again on develop branch and this morning all seems to work correctly

olive6707, 2023-04-22:
other issue I just fiound on this device: change of selector in domoticz has no effect (either by script or directly on the widget). Only button activation works on the remote control 
[_PluginZigbee_28 (11).log](https://github.com/zigbeefordomoticz/Domoticz-Zigbee/files/11300668/_PluginZigbee_28.11.log)

pipiche38, 2023-04-22:
You need to 'Allow Domoticz actions on Widgets from remote devices' enable it in the Plugin settings.

olive6707, 2023-04-22:
good it works

pipiche38, 2023-04-22:
Closed has completed

closed 2023-04-22 by pipiche38