zigbeefordomoticz/Domoticz-Zigbee #1576: Aqara 2button switch E1

Hezker, 2023-04-07:
Aqara QBKG39LM
lumi.switch.b2lc0
Aqara Smart Wall Switch 2 buttons E1 With NO Neutral
https://www.smartbest.pl/en/switches/311-aqara-e1-light-switch-2-buttons-without-n-6970504214989.html

Logs
`2023-04-08 01:34:53.856 Status: UsbCoordinator: Request Accepting new Hardware for 240 seconds
2023-04-08 01:34:58.487 Status: UsbCoordinator: [None] NEW OBJECT: 44eb None
2023-04-08 01:34:59.113 Status: UsbCoordinator: [-] NEW OBJECT: 44eb Active Endpoint Response Ep: 01 LQI: 192
2023-04-08 01:34:59.113 Status: UsbCoordinator: [-] NEW OBJECT: 44eb Active Endpoint Response Ep: 02 LQI: 192
2023-04-08 01:34:59.114 Status: UsbCoordinator: [-] NEW OBJECT: 44eb Active Endpoint Response Ep: f2 LQI: 192
2023-04-08 01:34:59.115 Status: UsbCoordinator: [-] NEW OBJECT: 44eb Request Simple Descriptor for Ep: 01
2023-04-08 01:34:59.871 Status: UsbCoordinator: [-] NEW OBJECT: 44eb Simple Descriptor Response EP: 0x01 LQI: 180
2023-04-08 01:34:59.872 Status: UsbCoordinator: [-] NEW OBJECT: 44eb ProfileID 0104
2023-04-08 01:34:59.872 Status: UsbCoordinator: [-] NEW OBJECT: 44eb ZDeviceID 0100
2023-04-08 01:34:59.872 Status: UsbCoordinator: [-] NEW OBJECT: 44eb Application Version 0001
2023-04-08 01:34:59.872 Status: UsbCoordinator: [-] NEW OBJECT: 44eb Cluster IN Count: 07
2023-04-08 01:34:59.872 Status: UsbCoordinator: [-] NEW OBJECT: 44eb Cluster In 1: 0000 (Basic)
2023-04-08 01:34:59.872 Status: UsbCoordinator: [-] NEW OBJECT: 44eb Cluster In 2: 0002
2023-04-08 01:34:59.872 Status: UsbCoordinator: [-] NEW OBJECT: 44eb Cluster In 3: 0003 (Identify)
2023-04-08 01:34:59.872 Status: UsbCoordinator: [-] NEW OBJECT: 44eb Cluster In 4: 0004 (Groups)
2023-04-08 01:34:59.873 Status: UsbCoordinator: [-] NEW OBJECT: 44eb Cluster In 5: 0005 (Scenes)
2023-04-08 01:34:59.873 Status: UsbCoordinator: [-] NEW OBJECT: 44eb Cluster In 6: 0006 (On/Off)
2023-04-08 01:34:59.873 Status: UsbCoordinator: [-] NEW OBJECT: 44eb Cluster In 7: 0009 (Alarms)
2023-04-08 01:34:59.873 Status: UsbCoordinator: [-] NEW OBJECT: 44eb Cluster OUT Count: 02
2023-04-08 01:34:59.873 Status: UsbCoordinator: [-] NEW OBJECT: 44eb Cluster Out 1: 000a (Time)
2023-04-08 01:34:59.873 Status: UsbCoordinator: [-] NEW OBJECT: 44eb Cluster Out 2: 0019 (Over-the-Air Upgrade)
2023-04-08 01:35:04.104 Status: UsbCoordinator: [1] NEW OBJECT: 44eb Model Name: lumi.switch.b2lc04
2023-04-08 01:35:04.104 Status: UsbCoordinator: [1] NEW OBJECT: 44eb Manufacturer Name: LUMI
2023-04-08 01:35:04.105 Status: UsbCoordinator: [None] NEW OBJECT: 44eb Request Node Descriptor
2023-04-08 01:35:09.094 Status: UsbCoordinator: [2] NEW OBJECT: 44eb Model Name: lumi.switch.b2lc04
2023-04-08 01:35:09.094 Status: UsbCoordinator: [2] NEW OBJECT: 44eb Manufacturer Name: LUMI
2023-04-08 01:35:14.104 Status: UsbCoordinator: [3] NEW OBJECT: 44eb Model Name: lumi.switch.b2lc04
2023-04-08 01:35:14.104 Status: UsbCoordinator: [3] NEW OBJECT: 44eb Manufacturer Name: LUMI
2023-04-08 01:35:14.127 Error: UsbCoordinator: zcl_attribute_discovery_request not implemented for RAW mode
2023-04-08 01:35:14.128 Error: UsbCoordinator: zcl_attribute_discovery_request not implemented for RAW mode
2023-04-08 01:35:14.132 Error: UsbCoordinator: zcl_attribute_discovery_request not implemented for RAW mode
2023-04-08 01:35:14.134 Error: UsbCoordinator: zcl_attribute_discovery_request not implemented for RAW mode
2023-04-08 01:35:14.135 Error: UsbCoordinator: zcl_attribute_discovery_request not implemented for RAW mode
2023-04-08 01:35:14.136 Error: UsbCoordinator: zcl_attribute_discovery_request not implemented for RAW mode
2023-04-08 01:35:14.137 Error: UsbCoordinator: zcl_attribute_discovery_request not implemented for RAW mode
2023-04-08 01:35:14.144 Error: UsbCoordinator: zcl_attribute_discovery_request not implemented for RAW mode
2023-04-08 01:35:14.152 Error: UsbCoordinator: zcl_attribute_discovery_request not implemented for RAW mode
2023-04-08 01:35:14.153 Error: UsbCoordinator: zcl_attribute_discovery_request not implemented for RAW mode`

Only one button appears and one Alarm device.

Tested on stable6 and develop branches.

pipiche38, 2023-04-08:
Can you kindly switch to develop branche

```
git checkout develop
git pull
sudo python3 -m pip install -r requirements.txt --upgrade
```

enable the debug parameters

Pairing
Lumi

[image]

remove any existing widgets related to that device, and make sure it has been also removed from the plugin ( Device Management)

restart the plugin

redo the pairing

It should create 2 switches 

please provide the plugin log file after the pairing

Hezker, 2023-04-08:
It works now. 2 switches was created and works fine.
Thx!
[PluginZigbee_03.txt](https://github.com/zigbeefordomoticz/Domoticz-Zigbee/files/11182887/PluginZigbee_03.txt)

pipiche38, 2023-04-08:
excelent. Thanks for the infos.

closed 2023-04-08 by pipiche38