zigbeefordomoticz/Domoticz-Zigbee #1577: Loratrap 6 button switch

marktn, 2023-04-11:
[image]

1. Got to the Tools Menu
1. Select Plugin Raw Devices (json)
1. Export

This will create an export.json file that you can save.
   
Finally you create an issue on GitHub with the following details

* Product Name 
* Brand
* URL on a web site
* What do you expect
* The logs from the pairing process
* export.json file as described here above

[export.zip](https://github.com/zigbeefordomoticz/Domoticz-Zigbee/files/11202372/export.zip)

Loratrap 6 button switch
https://nl.aliexpress.com/item/1005004988714663.html?spm=a2g0o.order_list.order_list_main.5.61e179d2LsXsTJ&gatewayAdapt=glo2nld
It has 6 buttons, it shows 3. None of them are working

From the same brand, i have a 1 button switch, thats works fine.

Log of the pairing:
` 2023-04-11 17:40:47.844 Status: Zigate: Request Accepting new Hardware for 240 seconds
2023-04-11 17:41:23.748 Status: Zigate: [None] NEW OBJECT: 6752 None
2023-04-11 17:41:24.341 Status: Zigate: [-] NEW OBJECT: 6752 Active Endpoint Response Ep: 01 LQI: 240
2023-04-11 17:41:24.341 Status: Zigate: [-] NEW OBJECT: 6752 Active Endpoint Response Ep: 02 LQI: 240
2023-04-11 17:41:24.342 Status: Zigate: [-] NEW OBJECT: 6752 Active Endpoint Response Ep: 03 LQI: 240
2023-04-11 17:41:24.342 Status: Zigate: [-] NEW OBJECT: 6752 Active Endpoint Response Ep: 04 LQI: 240
2023-04-11 17:41:24.342 Status: Zigate: [-] NEW OBJECT: 6752 Request Simple Descriptor for Ep: 02
2023-04-11 17:41:24.658 Status: Zigate: [-] NEW OBJECT: 6752 Simple Descriptor Response EP: 0x02 LQI: 240
2023-04-11 17:41:24.659 Status: Zigate: [-] NEW OBJECT: 6752 ProfileID 0104
2023-04-11 17:41:24.659 Status: Zigate: [-] NEW OBJECT: 6752 ZDeviceID 0000
2023-04-11 17:41:24.659 Status: Zigate: [-] NEW OBJECT: 6752 Application Version 0001
2023-04-11 17:41:24.659 Status: Zigate: [-] NEW OBJECT: 6752 Cluster IN Count: 03
2023-04-11 17:41:24.659 Status: Zigate: [-] NEW OBJECT: 6752 Cluster In 1: 0006 (On/Off)
2023-04-11 17:41:24.659 Status: Zigate: [-] NEW OBJECT: 6752 Cluster In 2: 0001 (Power Configuration)
2023-04-11 17:41:24.659 Status: Zigate: [-] NEW OBJECT: 6752 Cluster In 3: 0006 (On/Off)
2023-04-11 17:41:24.660 Status: Zigate: [-] NEW OBJECT: 6752 Cluster OUT Count: 00
2023-04-11 17:41:24.660 Status: Zigate: [-] NEW OBJECT: 6752 Request Simple Descriptor for Ep: 03
2023-04-11 17:41:24.869 Status: Zigate: [-] NEW OBJECT: 6752 Simple Descriptor Response EP: 0x03 LQI: 240
2023-04-11 17:41:24.869 Status: Zigate: [-] NEW OBJECT: 6752 ProfileID 0104
2023-04-11 17:41:24.869 Status: Zigate: [-] NEW OBJECT: 6752 ZDeviceID 0000
2023-04-11 17:41:24.869 Status: Zigate: [-] NEW OBJECT: 6752 Application Version 0001
2023-04-11 17:41:24.869 Status: Zigate: [-] NEW OBJECT: 6752 Cluster IN Count: 03
2023-04-11 17:41:24.869 Status: Zigate: [-] NEW OBJECT: 6752 Cluster In 1: 0006 (On/Off)
2023-04-11 17:41:24.869 Status: Zigate: [-] NEW OBJECT: 6752 Cluster In 2: 0001 (Power Configuration)
2023-04-11 17:41:24.870 Status: Zigate: [-] NEW OBJECT: 6752 Cluster In 3: 0006 (On/Off)
2023-04-11 17:41:24.870 Status: Zigate: [-] NEW OBJECT: 6752 Cluster OUT Count: 00
2023-04-11 17:41:24.870 Status: Zigate: [-] NEW OBJECT: 6752 Request Simple Descriptor for Ep: 04
2023-04-11 17:41:25.309 Status: Zigate: [-] NEW OBJECT: 6752 Simple Descriptor Response EP: 0x04 LQI: 240
2023-04-11 17:41:25.309 Status: Zigate: [-] NEW OBJECT: 6752 ProfileID 0104
2023-04-11 17:41:25.309 Status: Zigate: [-] NEW OBJECT: 6752 ZDeviceID 0000
2023-04-11 17:41:25.309 Status: Zigate: [-] NEW OBJECT: 6752 Application Version 0001
2023-04-11 17:41:25.309 Status: Zigate: [-] NEW OBJECT: 6752 Cluster IN Count: 03
2023-04-11 17:41:25.309 Status: Zigate: [-] NEW OBJECT: 6752 Cluster In 1: 0006 (On/Off)
2023-04-11 17:41:25.310 Status: Zigate: [-] NEW OBJECT: 6752 Cluster In 2: 0001 (Power Configuration)
2023-04-11 17:41:25.310 Status: Zigate: [-] NEW OBJECT: 6752 Cluster In 3: 0006 (On/Off)
2023-04-11 17:41:25.310 Status: Zigate: [-] NEW OBJECT: 6752 Cluster OUT Count: 00
2023-04-11 17:41:26.096 Status: Zigate: [0] NEW OBJECT: 6752 Model Name: TS0046
2023-04-11 17:41:26.096 Status: Zigate: [0] NEW OBJECT: 6752 Manufacturer Name: _TZ3000_iszegwpd
2023-04-11 17:41:26.096 Status: Zigate: [None] NEW OBJECT: 6752 Request Node Descriptor
2023-04-11 17:41:31.068 Status: Zigate: [1] NEW OBJECT: 6752 Model Name: TS0046
2023-04-11 17:41:31.068 Status: Zigate: [1] NEW OBJECT: 6752 Manufacturer Name: _TZ3000_iszegwpd
2023-04-11 17:41:36.092 Status: Zigate: [2] NEW OBJECT: 6752 Model Name: TS0046
2023-04-11 17:41:36.092 Status: Zigate: [2] NEW OBJECT: 6752 Manufacturer Name: _TZ3000_iszegwpd
2023-04-11 17:41:41.063 Status: Zigate: [3] NEW OBJECT: 6752 Model Name: TS0046
2023-04-11 17:41:41.063 Status: Zigate: [3] NEW OBJECT: 6752 Manufacturer Name: _TZ3000_iszegwpd
2023-04-11 17:41:41.092 Error: Zigate: zcl_attribute_discovery_request not implemented for RAW mode
2023-04-11 17:41:41.116 Error: Zigate: zcl_attribute_discovery_request not implemented for RAW mode
2023-04-11 17:41:41.125 Error: Zigate: zcl_attribute_discovery_request not implemented for RAW mode
2023-04-11 17:41:41.131 Error: Zigate: zcl_attribute_discovery_request not implemented for RAW mode
2023-04-11 17:41:41.135 Error: Zigate: zcl_attribute_discovery_request not implemented for RAW mode
2023-04-11 17:41:41.138 Error: Zigate: zcl_attribute_discovery_request not implemented for RAW mode
2023-04-11 17:41:41.149 Error: Zigate: zcl_attribute_discovery_request not implemented for RAW mode
2023-04-11 17:41:52.901 Dummy: Light/Switch (Telegram messages 2)
2023-04-11 17:43:53.899 Error: Zigate: MajDomoDevice - 6752/05 not known Endpoint
2023-04-11 17:43:53.902 Error: Zigate: forwarder_thread - Error while receiving a Coordinator command
2023-04-11 17:43:53.911 Error: Zigate: Issue in request 0180020022ff000104000605010267520200000191fd00f003, dumping stack: Traceback (most recent call last):
2023-04-11 17:43:53.911 File "/home/pi/domoticz/plugins/Domoticz-Zigbee/Classes/ZigpyTransport/forwarderThread.py", line 41, in forwarder_thread
2023-04-11 17:43:53.911 forward_message(self, message)
2023-04-11 17:43:53.911 File "/home/pi/domoticz/plugins/Domoticz-Zigbee/Classes/ZigpyTransport/instrumentation.py", line 30, in f_out
2023-04-11 17:43:53.911 result = f_in(self, message)
2023-04-11 17:43:53.911 File "/home/pi/domoticz/plugins/Domoticz-Zigbee/Classes/ZigpyTransport/forwarderThread.py", line 57, in forward_message
2023-04-11 17:43:53.911 self.F_out(message)
2023-04-11 17:43:53.911 File "/home/pi/domoticz/plugins/Domoticz-Zigbee/plugin.py", line 828, in processFrame
2023-04-11 17:43:53.911 ZigateRead(self, Devices, Data)
2023-04-11 17:43:53.911 File "/home/pi/domoticz/plugins/Domoticz-Zigbee/Modules/input.py", line 185, in ZigateRead
2023-04-11 17:43:53.911 _decoding(self, Devices, MsgData, MsgLQI)
2023-04-11 17:43:53.911 File "/home/pi/domoticz/plugins/Domoticz-Zigbee/Modules/input.py", line 4365, in Decode8095
2023-04-11 17:43:53.911 self.ListOfDevices[MsgSrcAddr]["Ep"][MsgEP][MsgClusterId]["0000"] = "Cmd: %s, %s" % (MsgCmd, unknown_)
2023-04-11 17:43:53.911 KeyError: '05'
2023-04-11 17:43:53.911
2023-04-11 17:43:55.922 Error: Zigate: MajDomoDevice - 6752/06 not known Endpoint
2023-04-11 17:43:55.926 Error: Zigate: forwarder_thread - Error while receiving a Coordinator command
2023-04-11 17:43:55.936 Error: Zigate: Issue in request 0180020022ff000104000606010267520200000192fd00f003, dumping stack: Traceback (most recent call last):
2023-04-11 17:43:55.936 File "/home/pi/domoticz/plugins/Domoticz-Zigbee/Classes/ZigpyTransport/forwarderThread.py", line 41, in forwarder_thread
2023-04-11 17:43:55.936 forward_message(self, message)
2023-04-11 17:43:55.936 File "/home/pi/domoticz/plugins/Domoticz-Zigbee/Classes/ZigpyTransport/instrumentation.py", line 30, in f_out
2023-04-11 17:43:55.936 result = f_in(self, message)
2023-04-11 17:43:55.936 File "/home/pi/domoticz/plugins/Domoticz-Zigbee/Classes/ZigpyTransport/forwarderThread.py", line 57, in forward_message
2023-04-11 17:43:55.936 self.F_out(message)
2023-04-11 17:43:55.936 File "/home/pi/domoticz/plugins/Domoticz-Zigbee/plugin.py", line 828, in processFrame
2023-04-11 17:43:55.936 ZigateRead(self, Devices, Data)
2023-04-11 17:43:55.936 File "/home/pi/domoticz/plugins/Domoticz-Zigbee/Modules/input.py", line 185, in ZigateRead
2023-04-11 17:43:55.936 _decoding(self, Devices, MsgData, MsgLQI)
2023-04-11 17:43:55.936 File "/home/pi/domoticz/plugins/Domoticz-Zigbee/Modules/input.py", line 4365, in Decode8095
2023-04-11 17:43:55.936 self.ListOfDevices[MsgSrcAddr]["Ep"][MsgEP][MsgClusterId]["0000"] = "Cmd: %s, %s" % (MsgCmd, unknown_)
2023-04-11 17:43:55.936 KeyError: '06'
2023-04-11 17:43:55.936 `

pipiche38, 2023-04-11:
can you clarify what type of Coordinator you are using ? Is that a Zigate ? if Zigate whihc firmware level are you ?

pipiche38, 2023-04-11:
could you kindly switch to develop branch if not yet on it

```
git checkout develop
git pull
sudo python3 -m pip install -r requirements.txt --upgrade
```

remove any remaining widgets related to this device

restart the plugin

check that the device doesn't exist in the Plugin Device Management section 

then redo the pairing

marktn, 2023-04-11:
Thanks, I will do that later this week.

marktn, 2023-04-12:
Hi,

Looks something is wrong,
`pi@domoticzVM:~/domoticz/plugins/Domoticz-Zigbee$ git checkout develop
error: Your local changes to the following files would be overwritten by checkou                                                                                                             t:
        .github/workflows/lint_python
        .gitmodules
        .hidden/VERSION
        Classes/GroupMgtv2/GrpDomoticz.py
        Classes/GroupMgtv2/GrpResponses.py
        Classes/OTA.py
        Classes/PluginConf.py
        Classes/WebServer/WebServer.py
        Classes/WebServer/dispatcher.py
        Classes/WebServer/rest_CfgReporting.py
        Classes/WebServer/rest_Groups.py
        Classes/WebServer/rest_Ota.py
        Classes/WebServer/rest_PluginUpgrade.py
        Classes/ZigpyTransport/AppGeneric.py
        Conf/Certified/00Local/.PRECIOUS
        Conf/Certified/Adeo/LDSENK02F.json
        Conf/Certified/Adeo/LDSENK10.json
        Conf/Certified/Adeo/LXEK-1.json
        Conf/Certified/Adeo/LXEK-2.json
        Conf/Certified/Adeo/LXEK-7.json
        Conf/Certified/Adeo/ZBEK-3.json
        Conf/Certified/Bitron/90201021A.json
        Conf/Certified/Bitron/90201021B.json
        Conf/Certified/Bitron/90201024.json
        Conf/Certified/Bitron/AV201021B.json
        Conf/Certified/Bitron/AV201024A.json
        Conf/Certified/Bitron/AV201029A.json
        Conf/Certified/CLEODE/ZHUM.json
        Conf/Certified/CLEODE/ZMOVE.json
        Conf/Certified/CLEODE/ZPLUG.json
        Conf/Certified/Casa.ia/CAC221.json
        Conf/Certified/Casa.ia/CCB432.json
        Conf/Certified/Casa.ia/CDWS312.json
        Conf/Certified/Casa.ia/CKF204.json
        Conf/Certified/Casa.ia/CKF205.json
        Conf/Certified/Casa.ia/CMS323.json
        Conf/Certified/Casa.ia/CPB206.json
        Conf/Certified/Casa.ia/CPC321.json
        Conf/Certified/Casa.ia/CPR412-E.json
        Conf/Certified/Casa.ia/CPR412.json
        Conf/Certified/Casa.ia/CSAC451-E.json
        Conf/Certified/Casa.ia/CSAC451-WTC-E.json
        Conf/Certified/Casa.ia/CSLC601-D-E.json
        Conf/Certified/Casa.ia/CSLC601-E.json
        Conf/Certified/Casa.ia/CSLC631-3.json
        Conf/Certified/Casa.ia/CSP403.json
        Conf/Certified/Casa.ia/CTHS317ET.json
        Conf/Certified/Casa.ia/Dual_Relay_Module.json
        Conf/Certified/Danalock/V3-BTZB.json
        Conf/Certified/Danfoss/RV001.json
        Conf/Certified/Danfoss/eT093WRO.json
        Conf/Certified/Danfoss/eTRV0100.json
        Conf/Certified/Develco/HESZB120.json
        Conf/Certified/Develco/HMSZB-110.json
        Conf/Certified/Develco/MOSZB-140.json
        Conf/Certified/Develco/SMSZB-120.json
        Conf/Certified/Develco/SPLZB-131.json
        Conf/Certified/Develco/SPLZB-132.json
        Conf/Certified/Develco/ZHEMI101.json
        Conf/Certified/Eurotronics/SPZB0001.json
        Conf/Certified/GLEDOPTO/GL-B-004P.json
        Conf/Certified/GLEDOPTO/GL-B-007Z.json
        Conf/Certified/GLEDOPTO/GL-B-008Z.json
        Conf/Certified/GLEDOPTO/GL-B-008ZS.json
        Conf/Certified/GLEDOPTO/GL-C-007-2ID.json
[... 295 lines cut ...]
Collecting yarl
  Using cached yarl-1.8.2-cp39-cp39-manylinux_2_17_x86_64.manylinux2014_x86_64.whl (264 kB)
  Using cached yarl-1.8.1-cp39-cp39-manylinux_2_17_x86_64.manylinux2014_x86_64.whl (264 kB)
Requirement already satisfied: cryptography in /usr/local/lib/python3.9/dist-packages (from -r requirements.txt (line 22)) (39.0.1)
Collecting cryptography
  Downloading cryptography-40.0.1-cp36-abi3-manylinux_2_28_x86_64.whl (3.7 MB)
     |████████████████████████████████| 3.7 MB 10.2 MB/s
Requirement already satisfied: dnspython==2.2.1 in /usr/local/lib/python3.9/dist-packages (from -r requirements.txt (line 23)) (2.2.1)
Collecting dnspython==2.2.1
  Using cached dnspython-2.2.1-py3-none-any.whl (269 kB)
Requirement already satisfied: charset-normalizer<4.0,>=2.0 in /usr/local/lib/python3.9/dist-packages (from aiohttp->-r requirements.txt (line 1)) (2.0.12)
Requirement already satisfied: idna>=2.0 in /usr/lib/python3/dist-packages (from yarl->-r requirements.txt (line 21)) (2.10)
Requirement already satisfied: humanfriendly>=9.1 in /usr/local/lib/python3.9/dist-packages (from coloredlogs->-r requirements.txt (line 9)) (10.0)
Requirement already satisfied: cffi>=1.12 in /usr/local/lib/python3.9/dist-packages (from cryptography->-r requirements.txt (line 22)) (1.15.1)
Requirement already satisfied: pycparser in /usr/local/lib/python3.9/dist-packages (from cffi>=1.12->cryptography->-r requirements.txt (line 22)) (2.21)
Requirement already satisfied: pyrsistent!=0.17.0,!=0.17.1,!=0.17.2,>=0.14.0 in /usr/local/lib/python3.9/dist-packages (from jsonschema->-r requirements.txt (line 10)) (0.19.3)
Requirement already satisfied: setuptools in /usr/lib/python3/dist-packages (from pure-pcapy3->-r requirements.txt (line 14)) (52.0.0)
Installing collected packages: cryptography
  Attempting uninstall: cryptography
    Found existing installation: cryptography 39.0.1
    Uninstalling cryptography-39.0.1:
      Successfully uninstalled cryptography-39.0.1
Successfully installed cryptography-40.0.1
pi@domoticzVM:~/domoticz/plugins/Domoticz-Zigbee$
`

pipiche38, 2023-04-12:
can you try the following command

`Tools/big-clean-git.sh`

and then redo
```
git checkout develop
git pull
sudo python3 -m pip install -r requirements.txt --upgrade
```

marktn, 2023-04-12:
Hi,

before i continue, this is normal?

pi@domoticzVM:~/domoticz/plugins/Domoticz-Zigbee$ Tools/big-clean-git.sh
HEAD is now at 8001fbd4 hot fix: when API base url is using https they were a coding issue

pipiche38, 2023-04-12:
so from their you should do the switch to develop

marktn, 2023-04-12:
Yes it is working!! 

Only see some errors, no problem?
 2023-04-12 22:00:13.255 Error: Zigate: Updating path from /home/pi/domoticz/plugins/Domoticz-Zigbee/ to /home/pi/domoticz/plugins/Domoticz-Zigbee is required, but no backward compatibility
2023-04-12 22:00:13.255 Error: Zigate: Updating path from /home/pi/domoticz/plugins/Domoticz-Zigbee/ to /home/pi/domoticz/plugins/Domoticz-Zigbee is required, but no backward compatibility
2023-04-12 22:00:13.255 Error: Zigate: Updating path from /home/pi/domoticz/plugins/Domoticz-Zigbee/Data/ to /home/pi/domoticz/plugins/Domoticz-Zigbee/Data is required, but no backward compatibility
2023-04-12 22:00:13.255 Error: Zigate: Updating path from /home/pi/domoticz/plugins/Domoticz-Zigbee/Conf/ to /home/pi/domoticz/plugins/Domoticz-Zigbee/Conf is required, but no backward compatibility
2023-04-12 22:00:13.255 Error: Zigate: Updating path from /home/pi/domoticz/plugins/Domoticz-Zigbee/OTAFirmware/ to /home/pi/domoticz/plugins/Domoticz-Zigbee/OTAFirmware is required, but no backward compatibility
2023-04-12 22:00:13.255 Error: Zigate: Updating path from /home/pi/domoticz/plugins/Domoticz-Zigbee/Reports/ to /home/pi/domoticz/plugins/Domoticz-Zigbee/Reports is required, but no backward compatibility
2023-04-12 22:00:13.255 Error: Zigate: Updating path from /home/pi/domoticz/plugins/Domoticz-Zigbee/www/ to /home/pi/domoticz/plugins/Domoticz-Zigbee/www is required, but no backward compatibility
2023-04-12 22:00:13.256 Error: Zigate: Updating path from /home/pi/domoticz/plugins/Domoticz-Zigbee/Logs/ to /home/pi/domoticz/plugins/Domoticz-Zigbee/Logs is required, but no backward compatibility