zigbeefordomoticz/Domoticz-Zigbee #1578: [Model Certification] -

francouai, 2023-04-14:
1. Please do provide device information such as Brand and Model name.
AEOTEC "RANGE EXTENDER Zi" modele:ZGA001-C15

1. Provide the URL where you have purchase the device:
https://amzn.eu/d/2D4tBmo

1. Finaly copy-paste all information provided in the log:
{"PluginInfos":{"HardwareID":2,"HomeFolder":"/home/pi/domoticz/plugins/Domoticz-Zigbee/","StartupFolder":"/home/pi/domoticz/","UserDataFolder":"/home/pi/domoticz/","WebRoot":"","Database":"/home/pi/domoticz/domoticz.db","Language":"fr","Version":"6.3","Author":"pipiche38","Name":"ZigBeeForDomoticZ","Address":"0.0.0.0","Port":"9999","SerialPort":"/dev/serial/by-id/usb-Silicon_Labs_Sonoff_Zigbee_3.0_USB_Dongle_Plus_0001-if00-port0","Key":"Zigate","Mode1":"ZigpyZNP","Mode2":"USB","Mode3":"False","Mode4":"9440","Mode6":"2","DomoticzVersion":"2023.1","DomoticzHash":"f9b9ac774","DomoticzBuildTime":"2023-02-14 15:06:40","PluginBranch":"stable6","PluginVersion":"6.3.012","TimeStamp":1681460959,"available":"6.3.012","available-firmMajor":0,"available-firmMinor":0,"FirmwareVersion":"Z-Stack 3.30+ (build 20210708)","FirmwareUpdate":false,"PluginUpdate":false,"LogLevel":7,"Zigpy":true,"CoordinatorModel":"CC1352/CC2652, Z-Stack 3.30+ (znp)","CoordinatorFirmwareVersion":"Z-Stack 3.30+ (build 20210708)","DisplayFirmwareVersion":"Znp - Z-Stack 3.30+ (build 20210708)","CoordinatorIEEE":"00124b002a2e963d","NetworkSize":"Total: 9 | Routers: 3 | End Devices: 6","NetworkDevices":{"1141":{"_TZ3000_xr3htd96":["TS0201"],"_TZ3000_2putqrmw":["TS011F-plug"]}}},"Analytics":1,"Device":{"Version":"3","ZDeviceName":"","Status":"UNKNOW","SQN":"02","Ep":{"01":{"0000":{"0004":"AL001","0005":"WG001-Z01"},"0003":{},"0019":{}},"02":{"0000":{"0004":"AL001","0005":"WG001-Z01"}},"f2":{}},"Heartbeat":"22","RIA":"4","LQI":90,"Battery":{},"Model":"WG001-Z01","ForceAckCommands":[],"MacCapa":"8e","IEEE":"dc8e95fffe83685a","Type":{},"ProfileID":"0104","ZDeviceID":"0008","App Version":"","Attributes List":{},"DeviceType":"FFD","HW Version":"","Last Cmds":[],"LogicalType":"Router","Manufacturer":"1310","Manufacturer Name":"AL001","NbEp":"3","PowerSource":"Main","ReceiveOnIdle":"On","Stack Version":"","Stamp":{"time":1681460927.2385876,"Time":"2023-04-14 10:28:47","MsgType":"8048","LastSeen":1681460854},"ZCL Version":"","Health":"Live","Capability":["Full-Function Device","Main Powered","Receiver during Idle","Standard security","NwkAddr should be allocated"],"ZDPSQN":"03","Epv2":{"01":{"ClusterIn":{"0000":{},"0003":{}},"ClusterOut":{"0003":{},"0019":{}},"ProfileID":"0104","ZDeviceID":"0008"},"02":{"ClusterIn":{},"ClusterOut":{},"ProfileID":{},"ZDeviceID":{}},"f2":{"ClusterIn":{},"ClusterOut":{},"ProfileID":{},"ZDeviceID":{}}},"ZCLSQN":"01","ZCL-IN-SQN":{"0000":"01"},"RollingLQI":[90,90,90,90,90],"ZDeviceVersion":"0001","ConfigSource":"8043","PairingInProgress":true,"_rawNodeDescriptor":"1310005200522c00008e524001","Max Buffer Size":"52","Max Rx":"0052","Max Tx":"0052","macapa":"8e","bitfield":"4001","server_mask":"2c00","descriptor_capability":"00"}}

pipiche38, 2023-04-14:
This is available on the 'develop' branch. To do so you need to switch

```
git checkout develop
git pull
sudo python3 -m pip install -r requirements.txt --upgrade
```

Then restart the plugin and redo the pairing

In all case, the extender is only visible at the zigbee level, so no widget will be created in Domoticz, but you'll see it in the Plugin

francouai, 2023-04-14:
not sure why, but "git checkout develop" is not recognized:
```

pi@raspberrypi:~ $ git checkout develop
fatal: not a git repository (or any of the parent directories): .git
```

pipiche38, 2023-04-14:
you need to do it from the zigbee for Domoticz plugin home directory

in general under plugins/Domoticz-Zigbee

francouai, 2023-04-15:
I'm able to see it now. 
thanks!

closed 2023-04-15 by francouai

pipiche38, 2023-04-15:
Excellent, thanks for the feedback