zigbeefordomoticz/Domoticz-Zigbee #1580: [Model Certification] - Aqara RTCGQ11LM

rom289, 2023-04-18:
Bonjour,

Aqara RTCGQ11LM

`{"PluginInfos":{"HardwareID":82,"HomeFolder":"/home/pi/domoticz/plugins/Domoticz-Zigbee/","StartupFolder":"/home/pi/domoticz/","UserDataFolder":"/home/pi/domoticz/","WebRoot":"","Database":"/home/pi/domoticz/domoticz.db","Language":"fr","Version":"6.3","Author":"pipiche38","Name":"Zigbee","Address":"0.0.0.0","Port":"9999","SerialPort":"/dev/ttyUSB-sonoffzigbee2","Key":"Zigate","Mode1":"ZigpyEZSP","Mode2":"USB","Mode3":"False","Mode4":"9440","Mode6":"2","DomoticzVersion":"2023.1","DomoticzHash":"f9b9ac774","DomoticzBuildTime":"2023-02-14 15:06:40","PluginBranch":"stable6","PluginVersion":"6.3.012","TimeStamp":1681794573,"available":"6.3.012","available-firmMajor":0,"available-firmMinor":0,"FirmwareVersion":"6.10.3.0 build 297","FirmwareUpdate":false,"PluginUpdate":false,"LogLevel":2,"Zigpy":true,"CoordinatorModel":"Unknown Silicon Labs","DisplayFirmwareVersion":"UNK - 6.10.3.0 build 297","CoordinatorIEEE":"dc8e95fffe08a614","NetworkSize":"Total: 18 | Routers: 16 | End Devices: 2","NetworkDevices":{"1037":{"LUMI":["lumi.plug"]},"1141":{"_TZ3000_cehuw1lw":["TS011F-plug"],"_TZ3000_typdpbpg":["TS011F-plug"],"_TZ3000_54hjn4vs":["TS0002"]},"1166":{"innr":["SP 220"]},"1286":{"SONOFF":["S26R2ZB"]},"110c":{"OSRAM":["Plug 01"]},"0000":{"SONOFF":["BASICZBR3"]},"117c":{"IKEA of Sweden":["TRADFRI bulb E27 CWS opal 600lm","TRADFRI bulb E27 W opal 1000lm"]}}},"Analytics":1,"Device":{"Version":"3","ZDeviceName":"Presence_Sejour","Status":"inDB","SQN":"3a","Ep":{"01":{"0400":{"0000":1},"0406":{"0000":"01"},"ClusterType":{"658":"Lux","659":"Motion"},"0000":{"ff01":"0121b30b03281d0421a81305212e00062401000000000a2125376410000b210000"},"0001":{"0000":2995},"0006":{"0000":"00"}}},"Heartbeat":"6394","RIA":"4","LQI":148,"Battery":61,"Model":"","ForceAckCommands":[],"MacCapa":"80","IEEE":"00158d000204c921","Type":"Lux/Motion","ProfileID":{},"ZDeviceID":{},"App Version":"","Attributes List":{},"DeviceType":"RFD","HW Version":"","Last Cmds":[],"LogicalType":"End Device","Manufacturer":"","Manufacturer Name":"","NbEp":"","PowerSource":"Battery","ReadAttributes":{"Ep":{"01":{"0000":{"TimeStamp":1681792635,"iSQN":{"0004":"00","0005":"00"},"Attributes":{"ff01":"00"},"ZigateRequest":{}},"0400":{"TimeStamp":1681794416,"iSQN":{"0000":"05"},"Attributes":{"0000":"00"},"ZigateRequest":{}},"0406":{"TimeStamp":1681794354,"iSQN":{"0000":"03","0001":"03","0010":"03","0011":"04","0012":"04"},"Attributes":{"0000":"00"},"ZigateRequest":{}}}}},"ReceiveOnIdle":"","Stack Version":"","Stamp":{"time":1681794416.883445,"Time":"2023-04-18 07:06:56","MsgType":"8102","LastSeen":1681794416},"ZCL Version":"","Health":"Live","Capability":["Reduced-Function Device","Standard security","NwkAddr should be allocated"],"ZDPSQN":"06","RollingLQI":[108,108,108,112,108,144,148,160,128,124,148],"WriteAttributes":{"Ep":{"01":{"0400":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0406":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0000":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0001":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0006":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}}}}},"PairingInProgress":false,"ZCLSQN":"05","PairingTime":1681762603.0751872,"BindingTable":[{"Devices":[],"SQN":0,"Status":"","TimeStamp":1681762603,"Time":1681762603}],"ConfigureReporting":{"Ep":{"01":{"0406":{"TimeStamp":1681762603.076958,"iSQN":{"0000":"01"},"Attributes":{},"ZigateRequest":{}},"0400":{"TimeStamp":1681762603.078013,"iSQN":{"0000":"02"},"Attributes":{},"ZigateRequest":{}}}}},"CertifiedDevice":false,"Param":{},"CheckParam":false,"LUMI":{"DeviceTemperature":74.3,"RSSI dB":-210,"LQI":"0100","BatteryVoltage":2995},"BatteryUpdateTime":1681792635}}`

Merci

rom289, 2023-04-18:
Le module semble fonctionné correctement.

rom289, 2023-04-18:
Je viens de voir qu'il apparait dans la liste des équipements compatibles mais avec des erreurs sur la partie lux sensor.

pipiche38, 2023-04-18:
Can you please put logs, what are the errors .... otherwise hard to know

pipiche38, 2023-04-18:
Please switch to 'develop' branch

```
git checkout develop
git pull
sudo python3 -m pip install -r requirements.txt --upgrade
```

then restart the plugin

pipiche38, 2023-04-18:
Closing this as this device works has been integrated into 'develop 'branch

closed 2023-04-18 by pipiche38