zigbeefordomoticz/Domoticz-Zigbee #1581: Neo Siren issue

olive6707, 2023-04-18:
Device paired but no device created under domoticz

**1. Please do provide device information such as Brand and Model name.** 
 
Neo Outdoor Siren

**1. Provide the URL where you have purchase the device:**

https://www.domadoo.fr/fr/peripheriques/6532-neo-sirene-intelligente-exterieure-zigbee-tuya-alimentation-5v1a-ou-batterie-panneau-solaire.html

**1. Finaly copy-paste all information provided in the log**
{"PluginInfos":{"HardwareID":28,"HomeFolder":"/home/pi/domoticz/plugins/Domoticz-Zigbee/","StartupFolder":"/home/pi/domoticz/","UserDataFolder":"/home/pi/domoticz/","WebRoot":"","Database":"/home/pi/domoticz/domoticz.db","Language":"fr","Version":"6.4","Author":"pipiche38","Name":"Zigbee","Address":"0.0.0.0","Port":"9999","SerialPort":"/dev/ttyAMA0","Key":"Zigate","Mode1":"ZigpyEZSP","Mode2":"PI","Mode3":"False","Mode4":"9440","Mode6":"2","DomoticzVersion":"2023.1","DomoticzHash":"f9b9ac774","DomoticzBuildTime":"2023-02-14 15:06:40","PluginBranch":"beta6","PluginVersion":"6.4.420","CertifiedDbVersion":"2.45","TimeStamp":1681805280,"available":0,"available-firmMajor":0,"available-firmMinor":0,"FirmwareVersion":"6.10.0.0 build 169","FirmwareUpdate":false,"PluginUpdate":false,"LogLevel":7,"Zigpy":true,"CoordinatorModel":"Elelabs, ELR02x","CoordinatorFirmwareVersion":"6.10.0.0 build 169","DisplayFirmwareVersion":"Ezsp - 6.10.0.0 build 169","CoordinatorIEEE":"8cf681fffec7d318","NetworkSize":"Total: 14 | Routers: 4 | End Devices: 10","NetworkDevices":{"1002":{"_TZ3000_wamqdr3f":["TS011F"],"_TZ3000_49qchf10":["TS0502A"],"_TZ3000_lbtpiody":["TS0201"],"_TYZB01_ynsiasng":["TS0219"]},"1037":{"unknown":["lumi.sensor_motion.aq2"],"LUMI":["lumi.sensor_motion.aq2"]},"1141":{"_TZ3000_0zrccfgx":["TS0215A-secure-remote"]},"0000":{"eWeLink":["DS01"]}}},"Analytics":-1,"Device":{"Version":"3","ZDeviceName":"","Status":"UNKNOW","SQN":"94","Ep":{"01":{"0000":{"0004":"_TZE200_nlrfgpny","0000":3,"0001":72,"0005":"TS0601","0007":3,"fffe":0,"ffe2":56,"ffe4":0,"ffdf":"'+f'+\u0012"}}},"Heartbeat":"25","RIA":"6","LQI":116,"Battery":{},"Model":"TS0601","ForceAckCommands":[],"MacCapa":"80","IEEE":"a4c138967b221ecf","Type":{},"ProfileID":{},"ZDeviceID":{},"App Version":72,"Attributes List":{},"DeviceType":"RFD","HW Version":"","Last Cmds":[],"LogicalType":"End Device","Manufacturer":"","Manufacturer Name":"_TZE200_nlrfgpny","NbEp":"1","PowerSource":"","ReadAttributes":{"Ep":{"01":{"0000":{"TimeStamp":1681805128,"iSQN":{"0004":"00","0000":"00","0001":"00","0005":"00","0007":"00","fffe":"00"},"Attributes":{},"ZigateRequest":{}}}}},"ReceiveOnIdle":"","Stack Version":"","Stamp":{"time":1681805135.7080758,"Time":"2023-04-18 10:05:35","MsgType":"8011","LastSeen":1681805135},"ZCL Version":3,"Health":"Live","Capability":["Reduced-Function Device","Standard security","NwkAddr should be allocated"],"ZCLSQN":"00","ZDPSQN":"01","RollingLQI":[116,116,116,116,116,116,116,116,116,116,116],"WriteAttributes":{"Ep":{"01":{"0000":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}}}}},"Epv2":{"01":{"ClusterIn":{},"ClusterOut":{},"ProfileID":{},"ZDeviceID":{}}},"PairingInProgress":true,"ZCL-IN-SQN":{"ef00":"91","0000":"93","000a":"94"},"Tuya":{"TUYA_MCU_VERSION_RSP":"50"},"SQN_000a":"94","ConsistencyCheck":"Bad Pairing"}}

**Domoticz log:**
[domoticz.log](https://github.com/zigbeefordomoticz/Domoticz-Zigbee/files/11259455/domoticz.log)

**Zigbee log:**
[_PluginZigbee_28 (3).log](https://github.com/zigbeefordomoticz/Domoticz-Zigbee/files/11259466/_PluginZigbee_28.3.log)

pipiche38, 2023-04-18:
can you try to get the latest 'develop' branch by doing `git pull`

Then restart the plugin 
enable Debug 
Tuya

and redo the pairing

Please provides the logs if it doesn't work and provides also the new device information

olive6707, 2023-04-18:
[_PluginZigbee_28 (4).log](https://github.com/zigbeefordomoticz/Domoticz-Zigbee/files/11260539/_PluginZigbee_28.4.log)

device now created under domoticz but no effect by switching on/off the new device

olive6707, 2023-04-18:
device information
{"PluginInfos":{"HardwareID":28,"HomeFolder":"/home/pi/domoticz/plugins/Domoticz-Zigbee/","StartupFolder":"/home/pi/domoticz/","UserDataFolder":"/home/pi/domoticz/","WebRoot":"","Database":"/home/pi/domoticz/domoticz.db","Language":"fr","Version":"6.4","Author":"pipiche38","Name":"Zigbee","Address":"0.0.0.0","Port":"9999","SerialPort":"/dev/ttyAMA0","Key":"Zigate","Mode1":"ZigpyEZSP","Mode2":"PI","Mode3":"False","Mode4":"9440","Mode6":"2","DomoticzVersion":"2023.1","DomoticzHash":"f9b9ac774","DomoticzBuildTime":"2023-02-14 15:06:40","PluginBranch":"beta6","PluginVersion":"6.4.420","CertifiedDbVersion":"2.45","TimeStamp":1681812655,"available":0,"available-firmMajor":0,"available-firmMinor":0,"FirmwareVersion":"6.10.0.0 build 169","FirmwareUpdate":false,"PluginUpdate":false,"LogLevel":7,"Zigpy":true,"CoordinatorModel":"Elelabs, ELR02x","CoordinatorFirmwareVersion":"6.10.0.0 build 169","DisplayFirmwareVersion":"Ezsp - 6.10.0.0 build 169","CoordinatorIEEE":"8cf681fffec7d318","NetworkSize":"Total: 15 | Routers: 4 | End Devices: 11","NetworkDevices":{"1002":{"_TZ3000_wamqdr3f":["TS011F"],"_TZ3000_49qchf10":["TS0502A"],"_TZ3000_lbtpiody":["TS0201"],"_TYZB01_ynsiasng":["TS0219"]},"1037":{"unknown":["lumi.sensor_motion.aq2"],"LUMI":["lumi.sensor_motion.aq2"]},"1141":{"_TZ3000_0zrccfgx":["TS0215A-secure-remote"]},"0000":{"eWeLink":["DS01"]}}},"Analytics":-1,"Device":{"Version":"3","ZDeviceName":"","Status":"inDB","SQN":"a2","Heartbeat":"37","RIA":"0","LQI":112,"Battery":{},"Model":"TS0601-_TZE200_nlrfgpny","ForceAckCommands":[],"MacCapa":"80","IEEE":"a4c138967b221ecf","Type":"SwitchAlarm","ProfileID":{},"ZDeviceID":{},"App Version":72,"Attributes List":{},"DeviceType":"RFD","HW Version":"","Last Cmds":[],"LogicalType":"End Device","Manufacturer":"1141","Manufacturer Name":"_TZE200_nlrfgpny","NbEp":"1","PowerSource":"Battery","ReadAttributes":{"Ep":{"01":{"0000":{"TimeStamp":1681812330,"iSQN":{"0004":"04","0000":"04","0001":"04","0005":"04","0007":"04","fffe":"04"},"Attributes":{},"ZigateRequest":{}},"0019":{"TimeStamp":1681812330,"iSQN":{"0000":"01","0001":"01","0002":"01","0003":"01","0004":"01","0005":"01","0006":"02","0007":"02","0008":"02","0009":"02","000a":"02","fffd":"02","fffe":"03"},"Attributes":{},"ZigateRequest":{}}}}},"ReceiveOnIdle":"","Stack Version":"","Stamp":{"time":1681812472.3287926,"Time":"2023-04-18 12:07:52","MsgType":"8011","LastSeen":1681812472},"ZCL Version":3,"Health":"Live","Capability":["Reduced-Function Device","Standard security","NwkAddr should be allocated"],"ZCLSQN":"0b","ZDPSQN":"04","RollingLQI":[104,112,112,112,112,112,112,112,112,112,112],"Param":{"TuyaAlarmDuration":10,"TuyaAlarmLevel":0,"TuyaAlarmMelody":1},"ConfigSource":"DeviceConf","Ep":{"01":{"0000":{"0004":"_TZE200_nlrfgpny","0000":3,"0001":72,"0005":"TS0601","0007":3,"fffe":0,"ffe2":56,"ffe4":0},"0004":{},"ef00":{},"000a":{},"0019":{"0000":"ffffffffffffffff","0001":4294967295,"0002":72,"0003":2,"0004":4294967295,"0005":65535,"0006":0,"0007":4417,"0008":65535,"0009":0,"fffd":3},"Type":"SwitchAlarm","ClusterType":{"313":"SwitchAlarm"}}},"WriteAttributes":{"Ep":{"01":{"0000":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0004":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"ef00":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"000a":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0019":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}}}}},"Epv2":{"01":{"ClusterIn":{},"ClusterOut":{},"ProfileID":{},"ZDeviceID":{}}},"PairingInProgress":false,"PairingTime":1681812330.0748675,"BindingTable":{"SQN":0,"Status":"Success","TimeStamp":1681812330.515494,"BindingTableSize":0,"Devices":[]},"CheckParam":false,"CertifiedDevice":true,"_rawNodeDescriptor":"1141004200422a000080424002","Max Buffer Size":"42","Max Rx":"0042","Max Tx":"0042","macapa":"80","bitfield":"4002","server_mask":"2a00","descriptor_capability":"00","ZCL-IN-SQN":{"ef00":"9e","0000":"9f","000a":"a0"},"Tuya":{"TUYA_MCU_VERSION_RSP":"50","TuyaTransactionId":3,"UnknowDp_0x07_Dt_0x02":"0000000a","UnknowDp_0x15_Dt_0x04":"01"},"SQN_000a":"a0"}}

pipiche38, 2023-04-18:
Are you able to activate the Siren from the device itself ?
If so, can you do it and send me the plugin log with a time stamp when you activated the siren ?

olive6707, 2023-04-18:
No, not able to do it, that is the point

pipiche38, 2023-04-18:
let's try the following

update to the latest `git pull`
remove the already paired device (remove widget from domoticz and make sure the device is also removed from the plugin)
restart the plugin
keep the tuya debug log
redo the pairing

olive6707, 2023-04-18:
no change but note the git pull makes no updade

pipiche38, 2023-04-18:
please try again. I didn't push the update

olive6707, 2023-04-18:
after update no change

log domoticz
2023-04-18 14:32:23.491 Zigbee: UpdateDevice - (Zigbee - TS0601-Solar-Siren_SwitchAlarm-a4c138967b221ecf-01) 1:On
2023-04-18 14:32:23.431 Status: User: olivier (IP: 192.168.1.47) initiated a switch command (313/Zigbee - TS0601-Solar-Siren_SwitchAlarm-a4c138967b221ecf-01/On)
2023-04-18 14:32:30.205 Zigbee: UpdateDevice - (Zigbee - TS0601-Solar-Siren_SwitchAlarm-a4c138967b221ecf-01) 0:Off

log plugin
[_PluginZigbee_28 (7).log](https://github.com/zigbeefordomoticz/Domoticz-Zigbee/files/11261793/_PluginZigbee_28.7.log)

pipiche38, 2023-04-18:
can you try once more,

just to a `git pull`

restart the plugin, no need to remove and repair
Still keep the tuya debug log
then try to Switch On or Off

olive6707, 2023-04-18:
switch ON ok but no effect on switch OFF command : need tu pull out the battery to stop the siren

[_PluginZigbee_28 (8).log](https://github.com/zigbeefordomoticz/Domoticz-Zigbee/files/11264738/_PluginZigbee_28.8.log)

pipiche38, 2023-04-19:
In the log I do not see the Action of Switch Off.
Did few changes, please get the latest version

git pull
sudo python3 -m pip install -r requirements.txt --upgrade

then restart the plugin
enable debug Command

and try On and Off.

There is also a Duration, Level, Melody parameters in the Device Parameters you might want to update

olive6707, 2023-04-19:
On/Off commands work well: enclosed log with timeline highlighted in yellow : 22h12:07
but device is a switch On/Off type, no parameter for Duration, Level and Melody
[_PluginZigbee_28 (9).log](https://github.com/zigbeefordomoticz/Domoticz-Zigbee/files/11277320/_PluginZigbee_28.9.log)

pipiche38, 2023-04-20:
Check de device parameters via the webui Device Management Envoyé de mon iPhone

pipiche38, 2023-04-20:
Check here: https://zigbeefordomoticz.github.io/wiki/en-eng/HowTo_Device-parameters.html
You should have

>         "TuyaAlarmDuration": 10,    // Duration , not sure this apply
>         "TuyaAlarmLevel": 0,           // Sound level
>         "TuyaAlarmMelody": 1         // Melody

olive6707, 2023-04-20:
ok, I tried to change Level and Melody: no effect but not very important in my case

pipiche38, 2023-04-20:
Ok. So then we can consider done ?

olive6707, 2023-04-20:
yes thank you very much

closed 2023-04-20 by pipiche38