zigbeefordomoticz/Domoticz-Zigbee #1582: [Model Certification] - QS-Zigbee-D02-TRIAC-LN

rom289, 2023-04-18:
Bonjour Pipiche38,

- QS-Zigbee-D02-TRIAC-LN (1 Gang Zigbee Dimmer Module)
- https://fr.aliexpress.com/item/4001279149071.html

`{"PluginInfos":{"HardwareID":82,"HomeFolder":"/home/pi/domoticz/plugins/Domoticz-Zigbee/","StartupFolder":"/home/pi/domoticz/","UserDataFolder":"/home/pi/domoticz/","WebRoot":"","Database":"/home/pi/domoticz/domoticz.db","Language":"fr","Version":"6.4","Author":"pipiche38","Name":"Zigbee","Address":"0.0.0.0","Port":"9999","SerialPort":"/dev/ttyUSB-sonoffzigbee2","Key":"Zigate","Mode1":"ZigpyEZSP","Mode2":"USB","Mode3":"False","Mode4":"9440","Mode6":"2","DomoticzVersion":"2023.1","DomoticzHash":"f9b9ac774","DomoticzBuildTime":"2023-02-14 15:06:40","PluginBranch":"beta6","PluginVersion":"6.4.420","CertifiedDbVersion":"2.45","TimeStamp":1681843664,"available":"6.4.418","available-firmMajor":0,"available-firmMinor":0,"FirmwareVersion":"6.10.3.0 build 297","FirmwareUpdate":false,"PluginUpdate":false,"LogLevel":2,"Zigpy":true,"CoordinatorModel":"Unknown Silicon Labs","DisplayFirmwareVersion":"UNK - 6.10.3.0 build 297","CoordinatorIEEE":"dc8e95fffe08a614","NetworkSize":"Total: 24 | Routers: 19 | End Devices: 5","NetworkDevices":{"1037":{"LUMI":["lumi.plug","lumi.remote.b286acn02","lumi.remote.b1acn01","lumi.sensor_switch.aq2"]},"1141":{"_TZ3000_cehuw1lw":["TS011F-plug"],"_TZ3000_typdpbpg":["TS011F-plug"],"_TZ3000_54hjn4vs":["TS0002"]},"1166":{"innr":["SP 220"]},"1286":{"SONOFF":["S26R2ZB"]},"110c":{"OSRAM":["Plug 01"]},"0000":{"SONOFF":["BASICZBR3"]},"117c":{"IKEA of Sweden":["TRADFRI bulb E27 CWS opal 600lm","TRADFRI bulb E27 W opal 1000lm"]}}},"Analytics":1,"Device":{"Version":"3","ZDeviceName":"","Status":"inDB","SQN":"45","Ep":{"01":{"0000":{"0004":"_TZ3210_ngqk6jia","0000":3,"0001":66,"0005":"TS110E","0007":1,"fffe":0,"ffe2":32,"ffe4":0,"ffdf":"\u000fx\u0002\u0000\u0012\u0011x\u0002\u0000\u0016"},"0006":{"4001":0,"4002":0,"8002":"1","0000":"00","fffd":2},"0008":{"fc03":"10","fc04":"993","f000":"0188","0000":3},"ClusterType":{"665":"LvlControl"}},"f2":{}},"Heartbeat":"45","RIA":"4","LQI":200,"Battery":{},"Model":"TS110E","ForceAckCommands":[],"MacCapa":"8e","IEEE":"a4c1381e124b63a3","Type":"Switch/LvlControl","ProfileID":{},"ZDeviceID":{},"App Version":66,"Attributes List":{},"DeviceType":"FFD","HW Version":"","Last Cmds":[],"LogicalType":"Router","Manufacturer":"1141","Manufacturer Name":"_TZ3210_ngqk6jia","NbEp":"2","PowerSource":"Main","ReadAttributes":{"Ep":{"01":{"0000":{"TimeStamp":1681842194,"iSQN":{"0004":"06","0000":"06","0001":"06","0005":"06","0007":"06","fffe":"06"},"Attributes":{},"ZigateRequest":{}},"0006":{"TimeStamp":1681843444,"iSQN":{"4000":"03","4001":"03","4002":"03","4003":"03","fffd":"03","fffe":"04","0000":"10"},"Attributes":{"8002":"00"},"ZigateRequest":{}},"0008":{"TimeStamp":1681843444,"iSQN":{"0000":"11"},"Attributes":{"fc03":"00","fc04":"00","f000":"00"},"ZigateRequest":{}}}}},"ReceiveOnIdle":"","Stack Version":"","Stamp":{"time":1681843533.4722004,"Time":"2023-04-18 20:45:33","MsgType":"8102","LastSeen":1681843533},"ZCL Version":3,"Health":"Live","Capability":["Full-Function Device","Main Powered","Receiver during Idle","Standard security","NwkAddr should be allocated"],"ZCLSQN":"11","ZDPSQN":"04","RollingLQI":[200,200,200,204,204,200,184,184,192,196,200],"WriteAttributes":{"Ep":{"01":{"0000":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0006":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0008":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}}}}},"Epv2":{"01":{"ClusterIn":{},"ClusterOut":{},"ProfileID":{},"ZDeviceID":{}},"f2":{"ClusterIn":{},"ClusterOut":{},"ProfileID":{},"ZDeviceID":{}}},"ZCL-IN-SQN":{"0006":"44","0008":"37","000a":"38","0000":"45"},"PairingInProgress":false,"SQN_000a":"38","PairingTime":1681842194.0642672,"BindingTable":{"SQN":0,"Status":"Success","TimeStamp":1681842194.217352,"BindingTableSize":0,"Devices":[]},"ConfigureReporting":{"Ep":{"01":{"0006":{"TimeStamp":1681842194.0695875,"iSQN":{"0000":"01"},"Attributes":{"0000":"00"},"ZigateRequest":{}},"0008":{"TimeStamp":1681842194.0706677,"iSQN":{"0000":"02"},"Attributes":{"0000":"00"},"ZigateRequest":{}}}}},"CertifiedDevice":false,"_rawNodeDescriptor":"1141004200422a00008e424001","Max Buffer Size":"42","Max Rx":"0042","Max Tx":"0042","macapa":"8e","bitfield":"4001","server_mask":"2a00","descriptor_capability":"00","ReadConfigureReporting":{"Ep":{"01":{"0006":{"0000":{"TimeStamp":1681842194.7682455,"Status":"00","DataType":"10","MinInterval":"0001","MaxInterval":"012c"}},"0008":{"0000":{"TimeStamp":1681842194.780119,"Status":"00","DataType":"20","MinInterval":"0005","MaxInterval":"012c","Change":"05"}}}},"TimeStamp":1681842194.7801142}}}`

L'objet créé est bien un control level, mais seul le on/off fonctionne, le dimmer n'a aucun effet.
(Je peux faire les demandes en anglais si vous préférez mais mon niveau est pas terrible et je crois savoir que vous parlez français.)

Merci

pipiche38, 2023-04-19:
If you are on bullseye and you have at least python3.9 , please switch to develop branch

```
git checkout develop
git pull
sudo python3 -m pip install -r requirements.txt --upgrade
```

remove the device from Domoticz
restart the plugin

Make sure the device is not visible in Domotciz nor in the plugin

then redo the pairing

rom289, 2023-04-19:
L'objet passe bien en optimisé, mais le dimmer ne fonctionne pas. Seul le on/off fonctionne.
Comment puis-je vous aider à rendre le dimmer fonctionnel ? Merci

pipiche38, 2023-04-20:
Did you redo the pairing from scratch ? can you provide logs ?
Do you have a way to dim from the device itself ? can you do so I can evaluate what the command is

rom289, 2023-04-20:
J'ai refait un appairage après l'avoir supprimé dans Domoticz et dans le plugin. Je le referai et vous enverrai les logs.
Je peux faire un dim depuis le module, je m'en occupe ce soir.

pipiche38, 2023-04-20:
Yes please do a dim from the device so I can see what are the command used if any

rom289, 2023-04-20:
Faut-il activer des options spécifiques de debug ?

pipiche38, 2023-04-20:
nop

rom289, 2023-04-20:
Log appairage :

` 2023-04-20 22:45:39.800 Status: Zigbee: [-] NEW OBJECT: 3deb Active Endpoint Response Ep: 01 LQI: 208
2023-04-20 22:45:39.801 Status: Zigbee: [-] NEW OBJECT: 3deb Active Endpoint Response Ep: f2 LQI: 208
2023-04-20 22:45:40.220 Error: Zigbee: ReadCluster - Error/unknow b94a/01 Cluster: 0000 Attribute: ff01 Status: 00 DataType: 42 DataSize: 0029 Data: 6410006510006e20006f20ff0121e40c0328120521270008211a360a21abae9923000000009b210000
2023-04-20 22:45:41.939 Zigbee: UpdateDevice - (Zigbee - TS110E-_QS-Zigbee-D02-TRIAC-LN_LvlControl-a4c1381e124b63a3-01) 1:
2023-04-20 22:45:42.002 Zigbee: UpdateDevice - (Zigbee - TS110E-_QS-Zigbee-D02-TRIAC-LN_LvlControl-a4c1381e124b63a3-01) 1:1
2023-04-20 22:45:41.107 Status: Zigbee: [0] NEW OBJECT: 3deb Model Name: TS110E-_QS-Zigbee-D02-TRIAC-LN
2023-04-20 22:45:41.110 Status: Zigbee: [0] NEW OBJECT: 3deb Manufacturer Name: _TZ3210_ngqk6jia
2023-04-20 22:45:41.111 Status: Zigbee: [None] NEW OBJECT: 3deb Request Node Descriptor
2023-04-20 22:45:43.243 Zigbee: Assisted Pairing: Polling: ['3deb']
2023-04-20 22:45:43.243 Zigbee: --> New device: 3deb
2023-04-20 22:45:43.243 Zigbee: --> inDB , removed 3deb from List
2023-04-20 22:45:43.243 Zigbee: ------> New Cluster: {'ClusterId': '0000', 'ClusterDesc': 'Basic'}
2023-04-20 22:45:43.243 Zigbee: ------> New Cluster: {'ClusterId': '0004', 'ClusterDesc': 'Groups'}
2023-04-20 22:45:43.243 Zigbee: ------> New Cluster: {'ClusterId': '0005', 'ClusterDesc': 'Scenes'}
2023-04-20 22:45:43.244 Zigbee: ------> New Cluster: {'ClusterId': '0006', 'ClusterDesc': 'On/Off'}
2023-04-20 22:45:43.244 Zigbee: ------> New Cluster: {'ClusterId': '0008', 'ClusterDesc': 'Level Control'}
2023-04-20 22:45:43.244 Zigbee: ------> New Cluster: {'ClusterId': '000a', 'ClusterDesc': 'Time'}
2023-04-20 22:45:43.244 Zigbee: ------> New Cluster: {'ClusterId': '0019', 'ClusterDesc': 'Over-the-Air Upgrade'}
2023-04-20 22:45:43.244 Zigbee: ----> New Ep: {'Ep': '01', 'Clusters': [{'ClusterId': '0000', 'ClusterDesc': 'Basic'}, {'ClusterId': '0004', 'ClusterDesc': 'Groups'}, {'ClusterId': '0005', 'ClusterDesc': 'Scenes'}, {'ClusterId': '0006', 'ClusterDesc': 'On/Off'}, {'ClusterId': '0008', 'ClusterDesc': 'Level Control'}, {'ClusterId': '000a', 'ClusterDesc': 'Time'}, {'ClusterId': '0019', 'ClusterDesc': 'Over-the-Air Upgrade'}]}
2023-04-20 22:45:43.244 Zigbee: ----> New Ep: {'Ep': 'f2', 'Clusters': []}
2023-04-20 22:45:43.244 Zigbee: --> New Device: {'NwkId': '3deb', 'ProvisionStatus': 'inDB', 'ProvisionStatusDesc': 'inDB', 'IEEE': 'a4c1381e124b63a3', 'ProfileId': '', 'ProfileIdDesc': 'Unknow', 'ZDeviceID': '', 'ZDeviceIDDesc': 'Unknow', 'Model': 'TS110E-_QS-Zigbee-D02-TRIAC-LN', 'PluginCertified': 'yes', 'Ep': [{'Ep': '01', 'Clusters': [{'ClusterId': '0000', 'ClusterDesc': 'Basic'}, {'ClusterId': '0004', 'ClusterDesc': 'Groups'}, {'ClusterId': '0005', 'ClusterDesc': 'Scenes'}, {'ClusterId': '0006', 'ClusterDesc': 'On/Off'}, {'ClusterId': '0008', 'ClusterDesc': 'Level Control'}, {'ClusterId': '000a', 'ClusterDesc': 'Time'}, {'ClusterId': '0019', 'ClusterDesc': 'Over-the-Air Upgrade'}]}, {'Ep': 'f2', 'Clusters': []}]}
2023-04-20 22:45:53.288 Zigbee: Assisted Pairing: Polling: []
`

rom289, 2023-04-20:
Nothing in logs when i dim with local switch.

pipiche38, 2023-04-21:
I need the plugin logs not DomoticZ. Plugin logs are at Logs/ folder under the name of PluginZigbee-xx.logEnvoyé de mon iPhone

pipiche38, 2023-04-22:
Please provides the Plugin logs not the Domoticz logs.
The plugin logs are available under Domoticz-Zigbee/Logs/PluginZigbee-xx.log ( where xx is the plugin hardware Id)

rom289, 2023-04-22:
Oui désolé j'ai pas pris le temps de vous répondre car je voulais chercher un peu plus. Mais j'ai rien de plus concernant ce module dans les logs du plugin.
Je n'ai aucune info quand je dim, ni sur on/off d'ailleurs.
Mais je ne dois pas l'utiliser correctement car je n'ai rien à part les erreurs déjà transmises dans #1585 et des erreurs pour un BP Aqara pour lequel je n'ai pas encore pris le temps de vous faire un retour.

pipiche38, 2023-04-22:
if you cannot provide the file `Domoticz-Zigbee/Logs/PluginZigbee-xx.log` I cannot help

rom289, 2023-04-22:
Je referai un appairage dans l'après midi et vous enverrai le fichier. Mais le Dim et le on/off ne donne rien dans le fichier.
Merci dans tous les cas pour votre travail.

pipiche38, 2023-04-22:
assurez-vous d'activer avant les debug Tuya et Input via le WebUI > Tools > Debug

rom289, 2023-04-23:
Bonsoir,
Voilà le fichier de logs avec les options de debug activées.
Appairage à 21:50:14.

Suivi de plusieurs commandes de dim et de on/off en local.

[PluginZigbee_82.log](https://github.com/zigbeefordomoticz/Domoticz-Zigbee/files/11304785/PluginZigbee_82.log)

Désolé pour l'envoi tardif.

pipiche38, 2023-04-24:
A priroi le debug Tuya n'etait pas activé, du coup pas d'info.
Peux-tu refaire la manip et t'assurer que les debug Tuya et Tuya0601 sont bien actif

rom289, 2023-04-24:
Bizarre, j'avais activé Tuya, Tuya0601 et Input. J'ai validé et redémarré le plugin pour être sûr.
Je vais bien refaire une essai ce soir mais c'est surprenant.

pipiche38, 2023-04-24:
A priori,

seul **Input** est bien activé

rom289, 2023-04-27:
Non désolé ça va être compliqué de refaire des essais dans les prochains jours mais je  vous tiens au courant.

pipiche38, 2023-05-13:
la config a été chargée dans les config de modules certifiés

closed 2023-05-13 by pipiche38