zigbeefordomoticz/Domoticz-Zigbee #1583: [Model Certification] - iluminize 511.201

rom289, 2023-04-18:
Bonjour,

- iluminize 511.201
- https://www.iluminize.com/en/shop/category/smart-home-control-actuators-230v-90
- Il semblerait que la référence ne soit plus disponible.

`{"PluginInfos":{"HardwareID":82,"HomeFolder":"/home/pi/domoticz/plugins/Domoticz-Zigbee/","StartupFolder":"/home/pi/domoticz/","UserDataFolder":"/home/pi/domoticz/","WebRoot":"","Database":"/home/pi/domoticz/domoticz.db","Language":"fr","Version":"6.4","Author":"pipiche38","Name":"Zigbee","Address":"0.0.0.0","Port":"9999","SerialPort":"/dev/ttyUSB-sonoffzigbee2","Key":"Zigate","Mode1":"ZigpyEZSP","Mode2":"USB","Mode3":"False","Mode4":"9440","Mode6":"2","DomoticzVersion":"2023.1","DomoticzHash":"f9b9ac774","DomoticzBuildTime":"2023-02-14 15:06:40","PluginBranch":"beta6","PluginVersion":"6.4.420","CertifiedDbVersion":"2.45","TimeStamp":1681844189,"available":"6.4.418","available-firmMajor":0,"available-firmMinor":0,"FirmwareVersion":"6.10.3.0 build 297","FirmwareUpdate":false,"PluginUpdate":false,"LogLevel":2,"Zigpy":true,"CoordinatorModel":"Unknown Silicon Labs","DisplayFirmwareVersion":"UNK - 6.10.3.0 build 297","CoordinatorIEEE":"dc8e95fffe08a614","NetworkSize":"Total: 24 | Routers: 19 | End Devices: 5","NetworkDevices":{"1037":{"LUMI":["lumi.plug","lumi.remote.b286acn02","lumi.remote.b1acn01","lumi.sensor_switch.aq2"]},"1141":{"_TZ3000_cehuw1lw":["TS011F-plug"],"_TZ3000_typdpbpg":["TS011F-plug"],"_TZ3000_54hjn4vs":["TS0002"]},"1166":{"innr":["SP 220"]},"1286":{"SONOFF":["S26R2ZB"]},"110c":{"OSRAM":["Plug 01"]},"0000":{"SONOFF":["BASICZBR3"]},"117c":{"IKEA of Sweden":["TRADFRI bulb E27 CWS opal 600lm","TRADFRI bulb E27 W opal 1000lm"]}}},"Analytics":1,"Device":{"Version":"3","ZDeviceName":"","Status":"inDB","SQN":"50","Ep":{"01":{"1000":{},"0000":{"4000":"2.4.1_r33","0004":"Sunricher","0005":"ZG9101SAC-HP","0000":3,"0001":0,"0002":0,"0003":0,"0006":"NULL","0007":1,"0008":255,"0009":255,"000a":"3030","000b":"NULL","fffd":1},"0003":{},"0004":{},"0005":{},"0006":{"4000":"01","4001":0,"4002":0,"4003":255,"0000":"00","fffd":1},"0008":{"0000":204},"0b05":{},"0019":{},"ClusterType":{"666":"LvlControl"}},"f2":{}},"Heartbeat":"148","RIA":"4","LQI":156,"Battery":{},"Model":"ZG9101SAC-HP","ForceAckCommands":[],"MacCapa":"8e","IEEE":"000d6f0013f0574d","Type":"Switch/LvlControl","ProfileID":"0104","ZDeviceID":"0101","App Version":0,"Attributes List":{},"DeviceType":"FFD","HW Version":0,"Last Cmds":[],"LogicalType":"Router","Manufacturer":"100b","Manufacturer Name":"Sunricher","NbEp":"2","PowerSource":"Main","ReadAttributes":{"Ep":{"01":{"0000":{"TimeStamp":1681842789,"iSQN":{"4000":"13","8000":"13","0000":"0c","0001":"0c","0002":"0c","0003":"0d","0004":"0d","0005":"0d","0006":"0e","0007":"0e","0008":"0e","0009":"0f","000a":"0f","000b":"0f","000c":"10","000d":"10","000e":"10","0010":"11","0011":"11","0012":"11","0013":"12","0014":"12","0015":"12","fffd":"13","fffe":"14","ffdf":"14","ffe0":"14","ffe1":"15","ffe2":"15","ffe4":"15"},"Attributes":{},"ZigateRequest":{}},"0019":{"TimeStamp":1681842789,"iSQN":{"0000":"03","0001":"03","0002":"03","0003":"04","0004":"04","0005":"04","0006":"05","0007":"05","0008":"05","0009":"06","000a":"06","fffd":"06","fffe":"07"},"Attributes":{},"ZigateRequest":{}},"0006":{"TimeStamp":1681843454,"iSQN":{"4000":"08","4001":"08","4002":"09","4003":"09","fffd":"09","fffe":"0a","0000":"1f"},"Attributes":{},"ZigateRequest":{}},"0008":{"TimeStamp":1681843454,"iSQN":{"0000":"20"},"Attributes":{},"ZigateRequest":{}}},"f2":{"0000":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}}}}},"ReceiveOnIdle":"On","Stack Version":0,"Stamp":{"time":1681844051.0360155,"Time":"2023-04-18 20:54:11","MsgType":"8102","LastSeen":1681844051},"ZCL Version":3,"Health":"Live","Capability":["Full-Function Device","Main Powered","Receiver during Idle","Standard security","NwkAddr should be allocated"],"ZDPSQN":"11","RollingLQI":[160,160,160,160,160,168,160,160,160,156,156],"Epv2":{"01":{"ClusterIn":{"1000":{},"0000":{},"0003":{},"0004":{},"0005":{},"0006":{},"0008":{},"0b05":{}},"ClusterOut":{"1000":{},"0019":{}},"ProfileID":"0104","ZDeviceID":"0101"},"f2":{"ClusterIn":{},"ClusterOut":{},"ProfileID":{},"ZDeviceID":{}}},"ZCLSQN":"20","WriteAttributes":{"Ep":{"01":{"1000":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0000":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0003":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0004":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0005":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0006":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0008":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0b05":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0019":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}}}}},"ZDeviceVersion":"0001","ConfigSource":"8043","PairingInProgress":false,"_rawNodeDescriptor":"100b005200522c00008e524001","Max Buffer Size":"52","Max Rx":"0052","Max Tx":"0052","macapa":"8e","bitfield":"4001","server_mask":"2c00","descriptor_capability":"00","PairingTime":1681842789.2199106,"Bind":{"01":{"1000":{"Target":"0000","Stamp":1681842790,"Phase":"binded","Status":"00","i_sqn":"0b"},"0000":{"Target":"0000","Stamp":1681842789,"Phase":"binded","Status":"00","i_sqn":"04"},"0003":{"Target":"0000","Stamp":1681842789,"Phase":"binded","Status":"00","i_sqn":"05"},"0004":{"Target":"0000","Stamp":1681842789,"Phase":"binded","Status":"00","i_sqn":"06"},"0005":{"Target":"0000","Stamp":1681842789,"Phase":"binded","Status":"00","i_sqn":"07"},"0006":{"Target":"0000","Stamp":1681842789,"Phase":"binded","Status":"00","i_sqn":"08"},"0008":{"Target":"0000","Stamp":1681842790,"Phase":"binded","Status":"00","i_sqn":"09"},"0b05":{"Target":"0000","Stamp":1681842790,"Phase":"binded","Status":"00","i_sqn":"0a"}}},"BindingTable":{"SQN":0,"Status":"Success","TimeStamp":1681842793.1656845,"BindingTableSize":8,"Devices":[{"000d6f0013f0574d":{"sourceEp":"01","Cluster":"0000","targetIEEE":"dc8e95fffe08a614","targetNickName":"Zigbee Coordinator","targetEp":"01"}},{"000d6f0013f0574d":{"sourceEp":"01","Cluster":"0003","targetIEEE":"dc8e95fffe08a614","targetNickName":"Zigbee Coordinator","targetEp":"01"}},{"000d6f0013f0574d":{"sourceEp":"01","Cluster":"0004","targetIEEE":"dc8e95fffe08a614","targetNickName":"Zigbee Coordinator","targetEp":"01"}},{"000d6f0013f0574d":{"sourceEp":"01","Cluster":"0005","targetIEEE":"dc8e95fffe08a614","targetNickName":"Zigbee Coordinator","targetEp":"01"}},{"000d6f0013f0574d":{"sourceEp":"01","Cluster":"0006","targetIEEE":"dc8e95fffe08a614","targetNickName":"Zigbee Coordinator","targetEp":"01"}},{"000d6f0013f0574d":{"sourceEp":"01","Cluster":"0008","targetIEEE":"dc8e95fffe08a614","targetNickName":"Zigbee Coordinator","targetEp":"01"}},{"000d6f0013f0574d":{"sourceEp":"01","Cluster":"0b05","targetIEEE":"dc8e95fffe08a614","targetNickName":"Zigbee Coordinator","targetEp":"01"}},{"000d6f0013f0574d":{"sourceEp":"01","Cluster":"1000","targetIEEE":"dc8e95fffe08a614","targetNickName":"Zigbee Coordinator","targetEp":"01"}}]},"ConfigureReporting":{"Ep":{"01":{"0006":{"TimeStamp":1681842789.2379615,"iSQN":{"0000":"01"},"Attributes":{"0000":"00"},"ZigateRequest":{}},"0008":{"TimeStamp":1681842789.2426615,"iSQN":{"0000":"02"},"Attributes":{"0000":"00"},"ZigateRequest":{}}}}},"CertifiedDevice":false,"ZCL-IN-SQN":{"0006":"4f","0008":"50","0019":"4e"},"SWBUILD_1":"NULL","SWBUILD":"2.4.1_r33","ReadConfigureReporting":{"Ep":{"01":{"0006":{"0000":{"TimeStamp":1681842793.05908,"Status":"00","DataType":"10","MinInterval":"0001","MaxInterval":"012c"}},"0008":{"0000":{"TimeStamp":1681842793.132886,"Status":"00","DataType":"20","MinInterval":"0005","MaxInterval":"012c","Change":"05"}}}},"TimeStamp":1681842793.132881},"OTAClient":{"ManufacturerCode":"100b","ImageType":"27ed","CurrentImageVersion":"00000021"}}}`

Fonctionne correctement mais apparait en non-optimisé.

Merci

pipiche38, 2023-04-19:
If you are on bullseye and you have at least python3.9 , please switch to develop branch

```
git checkout develop
git pull
sudo python3 -m pip install -r requirements.txt --upgrade
```

restart the plugin

The device should then be seen as optimized/certified.

For test purposes, I would appreciate if you can remove /delete the device from Domoticz and the plugin
restart the plugin
redo the pairing
check that it is still functional

rom289, 2023-04-19:
Objet optimisé. Parfait. Merci

closed 2023-04-19 by rom289