zigbeefordomoticz/Domoticz-Zigbee #1585: [Model Certification] - Aqara QBKG21LM

rom289, 2023-04-18:
Aqara QBKG21LM

`{"PluginInfos":{"HardwareID":82,"HomeFolder":"/home/pi/domoticz/plugins/Domoticz-Zigbee/","StartupFolder":"/home/pi/domoticz/","UserDataFolder":"/home/pi/domoticz/","WebRoot":"","Database":"/home/pi/domoticz/domoticz.db","Language":"fr","Version":"6.4","Author":"pipiche38","Name":"Zigbee","Address":"0.0.0.0","Port":"9999","SerialPort":"/dev/ttyUSB-sonoffzigbee2","Key":"Zigate","Mode1":"ZigpyEZSP","Mode2":"USB","Mode3":"False","Mode4":"9440","Mode6":"2","DomoticzVersion":"2023.1","DomoticzHash":"f9b9ac774","DomoticzBuildTime":"2023-02-14 15:06:40","PluginBranch":"beta6","PluginVersion":"6.4.420","CertifiedDbVersion":"2.45","TimeStamp":1681844969,"available":"6.4.418","available-firmMajor":0,"available-firmMinor":0,"FirmwareVersion":"6.10.3.0 build 297","FirmwareUpdate":false,"PluginUpdate":false,"LogLevel":2,"Zigpy":true,"CoordinatorModel":"Unknown Silicon Labs","DisplayFirmwareVersion":"UNK - 6.10.3.0 build 297","CoordinatorIEEE":"dc8e95fffe08a614","NetworkSize":"Total: 24 | Routers: 19 | End Devices: 5","NetworkDevices":{"1037":{"LUMI":["lumi.plug","lumi.remote.b286acn02","lumi.remote.b1acn01","lumi.sensor_switch.aq2"]},"1141":{"_TZ3000_cehuw1lw":["TS011F-plug"],"_TZ3000_typdpbpg":["TS011F-plug"],"_TZ3000_54hjn4vs":["TS0002"]},"1166":{"innr":["SP 220"]},"1286":{"SONOFF":["S26R2ZB"]},"110c":{"OSRAM":["Plug 01"]},"0000":{"SONOFF":["BASICZBR3"]},"117c":{"IKEA of Sweden":["TRADFRI bulb E27 CWS opal 600lm","TRADFRI bulb E27 W opal 1000lm"]}}},"Analytics":1,"Device":{"Version":"3","ZDeviceName":"","Status":"inDB","SQN":"23","Ep":{"29":{"0012":{"0055":"0001"}},"01":{"fcc0":{"00f7":"641000032814052101009a20100a2125370b20000d230b0e00000e2300000000","00ee":"00000e0b"},"0006":{"4000":"01","0000":"00","00f5":"58238208"},"0000":{"0005":"lumi.switch.l1aeu1","0000":3,"0001":11,"0002":2,"0003":1,"0004":"LUMI","0006":"01-12-2021","0007":1,"fffd":1},"ClusterType":{"668":"Switch"}},"f2":{}},"Heartbeat":"272","RIA":"4","LQI":148,"Battery":{},"Model":"lumi.switch.l1aeu1","ForceAckCommands":[],"MacCapa":"80","IEEE":"54ef441000225527","Type":"Switch","ProfileID":"0104","ZDeviceID":"0100","App Version":11,"Attributes List":{},"DeviceType":"RFD","HW Version":1,"Last Cmds":[],"LogicalType":"End Device","Manufacturer":"115f","Manufacturer Name":"LUMI","NbEp":"3","PowerSource":"Battery","ReadAttributes":{"Ep":{"29":{"0012":{"TimeStamp":1681844271,"iSQN":{},"Attributes":{"0055":"00"},"ZigateRequest":{}}},"01":{"0006":{"TimeStamp":1681844271,"iSQN":{"4000":"01","4001":"01","4002":"02","4003":"02","0000":"01","fffd":"02","fffe":"03"},"Attributes":{"00f5":"00"},"ZigateRequest":{}},"0000":{"TimeStamp":1681843609,"iSQN":{"4000":"0b","8000":"0b","0000":"04","0001":"04","0002":"04","0003":"05","0004":"05","0005":"05","0006":"06","0007":"06","0008":"06","0009":"07","000a":"07","000b":"07","000c":"08","000d":"08","000e":"08","0010":"09","0011":"09","0012":"09","0013":"0a","0014":"0a","0015":"0a","fffd":"0b","fffe":"0c","ffdf":"0c","ffe0":"0c","ffe1":"0d","ffe2":"0d","ffe4":"0d","ff01":"0e","ff02":"0e"},"Attributes":{},"ZigateRequest":{}}}}},"ReceiveOnIdle":"Off","Stack Version":2,"Stamp":{"time":1681844946.97107,"Time":"2023-04-18 21:09:06","MsgType":"8011","LastSeen":1681844946},"ZCL Version":3,"Health":"Live","Capability":["Reduced-Function Device","Standard security","NwkAddr should be allocated"],"ZDPSQN":"06","PairingInProgress":false,"RollingLQI":[148,148,148,148,148,148,152,152,148,148,148],"LUMI":{"DeviceTemperature":51.2,"RSSI dB":-255},"SQN_000a":"23","Epv2":{"29":{"ClusterIn":{"0012":{}},"ClusterOut":{},"ProfileID":"0104","ZDeviceID":"0100"},"01":{"ClusterIn":{},"ClusterOut":{},"ProfileID":{},"ZDeviceID":{}},"f2":{"ClusterIn":{},"ClusterOut":{},"ProfileID":{},"ZDeviceID":{}}},"ZDeviceVersion":"0001","ConfigSource":"8043","_rawNodeDescriptor":"115f006400642c0000807f4002","Max Buffer Size":"7f","Max Rx":"0064","Max Tx":"0064","macapa":"80","bitfield":"4002","server_mask":"2c00","descriptor_capability":"00","PairingTime":1681843609.0976753,"Bind":{"29":{"0012":{"Target":"0000","Stamp":1681843609,"Phase":"binded","Status":"00","i_sqn":"04"}}},"BindingTable":{"SQN":0,"Status":"Success","TimeStamp":1681843609.6246502,"BindingTableSize":1,"Devices":[{"54ef441000225527":{"sourceEp":"29","Cluster":"0012","targetIEEE":"dc8e95fffe08a614","targetNickName":"Zigbee Coordinator","targetEp":"01"}}]},"ConfigureReporting":{"Ep":{"01":{"0006":{"TimeStamp":1681843609.1047566,"iSQN":{"0000":"00"},"Attributes":{"0000":"00"},"ZigateRequest":{}}}}},"ZCLSQN":"0f","CertifiedDevice":false,"SWBUILD_1":"01-12-2021","WriteAttributes":{"Ep":{"29":{"0012":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}}},"01":{"fcc0":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0006":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0000":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}}}}},"ReadConfigureReporting":{"Ep":{"01":{"0006":{"0000":{"TimeStamp":1681843613.6723325,"Status":"00","DataType":"10","MinInterval":"0000","MaxInterval":"ffff"}}}},"TimeStamp":1681843613.6723213}}}`

Fonctionne correctement mais apparait en non-optimisé.

Merci

pipiche38, 2023-04-19:
If you are on bullseye and you have at least python3.9 , please switch to develop branch

```
git checkout develop
git pull
sudo python3 -m pip install -r requirements.txt --upgrade
```

restart the plugin

The device should then be seen as optimized/certified.

For test purposes, I would appreciate if you can remove /delete the device from Domoticz and the plugin
restart the plugin
redo the pairing
check that it is still functional

rom289, 2023-04-19:
Erreur de ma part : le fichier de log envoyé dans mon message initial correspond à un Aqara WS-EUK02.

Le fichier de log du Aqara QBKG21LM est le suivant : 
 
`{"PluginInfos":{"HardwareID":82,"HomeFolder":"/home/pi/domoticz/plugins/Domoticz-Zigbee/","StartupFolder":"/home/pi/domoticz/","UserDataFolder":"/home/pi/domoticz/","WebRoot":"","Database":"/home/pi/domoticz/domoticz.db","Language":"fr","Version":"6.4","Author":"pipiche38","Name":"Zigbee","Address":"0.0.0.0","Port":"9999","SerialPort":"/dev/ttyUSB-sonoffzigbee2","Key":"Zigate","Mode1":"ZigpyEZSP","Mode2":"USB","Mode3":"False","Mode4":"9440","Mode6":"2","DomoticzVersion":"2023.1","DomoticzHash":"f9b9ac774","DomoticzBuildTime":"2023-02-14 15:06:40","PluginBranch":"beta6","PluginVersion":"6.4.423","CertifiedDbVersion":"2.46","TimeStamp":1681933962,"available":"6.4.423","available-firmMajor":0,"available-firmMinor":0,"FirmwareVersion":"6.10.3.0 build 297","FirmwareUpdate":false,"PluginUpdate":false,"LogLevel":2,"Zigpy":true,"CoordinatorModel":"Unknown Silicon Labs","DisplayFirmwareVersion":"UNK - 6.10.3.0 build 297","CoordinatorIEEE":"dc8e95fffe08a614","NetworkSize":"Total: 25 | Routers: 20 | End Devices: 5","NetworkDevices":{"1037":{"LUMI":["lumi.plug","lumi.remote.b286acn02","lumi.remote.b1acn01","lumi.sensor_switch.aq2","lumi.sensor_motion.aq2"]},"1141":{"_TZ3000_cehuw1lw":["TS011F-plug"],"_TZ3000_typdpbpg":["TS011F-plug"],"_TZ3000_54hjn4vs":["TS0002"]},"1166":{"innr":["SP 220"]},"1286":{"SONOFF":["S26R2ZB"]},"110c":{"OSRAM":["Plug 01"]},"0000":{"SONOFF":["BASICZBR3"]},"117c":{"IKEA of Sweden":["TRADFRI bulb E27 CWS opal 600lm","TRADFRI bulb E27 W opal 1000lm"]}}},"Analytics":1,"Device":{"Version":"3","ZDeviceName":"Centre_Cuisine_Ete","Status":"inDB","SQN":"55","Ep":{"01":{"0000":{"0005":"lumi.switch.b1lacn02","0001":1,"ff01":"6410006510006e20006f20ff0121e40c0328160521220008211a360a21f05f9923000000009b210000"}},"02":{"0006":{"0000":"00","f000":117440575},"ClusterType":{"668":"Switch"}},"03":{"0010":{},"0006":{"0000":"00"},"0004":{},"0005":{},"ClusterType":{}},"04":{"0012":{},"0006":{"0000":"01"},"ClusterType":{}},"05":{"0012":{},"0006":{"0000":"00"},"ClusterType":{}},"06":{"0012":{},"0006":{"0000":"00"},"ClusterType":{}},"08":{"000c":{},"ClusterType":{}}},"Heartbeat":"140","RIA":"4","LQI":200,"Battery":{},"Model":"lumi.switch.b1lacn02","ForceAckCommands":[],"MacCapa":"84","IEEE":"00158d0006fe41fa","Type":"Switch/Analog","ProfileID":"0104","ZDeviceID":"0053","App Version":1,"Attributes List":{},"DeviceType":"RFD","HW Version":"","Last Cmds":[],"LogicalType":"Router","Manufacturer":"1037","Manufacturer Name":"","NbEp":"7","PowerSource":"Main","ReadAttributes":{"Ep":{"01":{"0000":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}}},"02":{"0006":{"TimeStamp":1681933260,"iSQN":{"4000":"05","4001":"05","4002":"06","4003":"06","fffd":"06","fffe":"07","0000":"3e"},"Attributes":{"f000":"00"},"ZigateRequest":{}}},"03":{"0006":{"TimeStamp":1681930612,"iSQN":{"4000":"08","4001":"08","4002":"09","4003":"09","fffd":"09","fffe":"0a","0000":"2b"},"Attributes":{},"ZigateRequest":{}}},"04":{"0006":{"TimeStamp":1681930612,"iSQN":{"4000":"0b","4001":"0b","4002":"0c","4003":"0c","fffd":"0c","fffe":"0d","0000":"2c"},"Attributes":{},"ZigateRequest":{}}},"05":{"0006":{"TimeStamp":1681930612,"iSQN":{"4000":"0e","4001":"0e","4002":"0f","4003":"0f","fffd":"0f","fffe":"10","0000":"2d"},"Attributes":{},"ZigateRequest":{}}},"06":{"0006":{"TimeStamp":1681930612,"iSQN":{"4000":"11","4001":"11","4002":"12","4003":"12","fffd":"12","fffe":"13","0000":"2e"},"Attributes":{},"ZigateRequest":{}}}}},"ReceiveOnIdle":"Off","Stack Version":"","Stamp":{"time":1681933909.2422917,"Time":"2023-04-19 21:51:49","MsgType":"8011","LastSeen":1681933909},"ZCL Version":"","Health":"Live","Capability":["Reduced-Function Device","Main Powered","Standard security","NwkAddr should be allocated"],"ZDPSQN":"16","RollingLQI":[200,200,196,196,196,200,200,196,196,200,200],"WriteAttributes":{"Ep":{"01":{"0000":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}}},"02":{"0006":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}}},"03":{"0010":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0006":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0004":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0005":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}}},"04":{"0012":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0006":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}}},"05":{"0012":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0006":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}}},"06":{"0012":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0006":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}}},"08":{"000c":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}}}}},"Epv2":{"01":{"ClusterIn":{},"ClusterOut":{},"ProfileID":{},"ZDeviceID":{}},"02":{"ClusterIn":{},"ClusterOut":{},"ProfileID":{},"ZDeviceID":{}},"03":{"ClusterIn":{"0010":{},"0006":{},"0004":{},"0005":{}},"ClusterOut":{},"ProfileID":"0104","ZDeviceID":"0100"},"04":{"ClusterIn":{"0012":{},"0006":{}},"ClusterOut":{},"ProfileID":"0104","ZDeviceID":"0000"},"05":{"ClusterIn":{"0012":{},"0006":{}},"ClusterOut":{},"ProfileID":"0104","ZDeviceID":"0000"},"06":{"ClusterIn":{"0012":{},"0006":{}},"ClusterOut":{},"ProfileID":"0104","ZDeviceID":"0000"},"08":{"ClusterIn":{"000c":{}},"ClusterOut":{}," [... cut ...] Request":{}}},"04":{"0006":{"TimeStamp":1681930527.2060263,"iSQN":{"0000":"02"},"Attributes":{"0000":"8c"},"ZigateRequest":{}}},"05":{"0006":{"TimeStamp":1681930527.2100623,"iSQN":{"0000":"03"},"Attributes":{"0000":"8c"},"ZigateRequest":{}}},"06":{"0006":{"TimeStamp":1681930527.2140877,"iSQN":{"0000":"04"},"Attributes":{"0000":"8c"},"ZigateRequest":{}}}}},"ZCLSQN":"3f","CertifiedDevice":false,"ZCL-IN-SQN":{"0006":"2e"},"SQN_000a":"55","Param":{},"CheckParam":false}}`

Désolé pour le temps perdu.

pipiche38, 2023-04-20:
If you are on bullseye and you have at least python3.9 , please switch to develop branch
If you are already on that branch, please redo those commands

```
git checkout develop
git pull
sudo python3 -m pip install -r requirements.txt --upgrade
```

restart the plugin

The device should then be seen as optimized/certified.

For test purposes, I would appreciate if you can remove /delete the device from Domoticz and the plugin
restart the plugin
redo the pairing
check that it is still functional

rom289, 2023-04-20:
Both are ok. Thank you.

rom289, 2023-04-20:
J'ai des erreurs en fait. Mais ça fonctionne bien il me semble.

` 2023-04-20 22:42:52.112 Error: Zigbee: ReadCluster - Error/unknow 26f1/01 Cluster: fcc0 Attribute: 00ee Status: 00 DataType: 23 DataSize: 0004 Data: 00000e0b
2023-04-20 22:45:40.220 Error: Zigbee: ReadCluster - Error/unknow b94a/01 Cluster: 0000 Attribute: ff01 Status: 00 DataType: 42 DataSize: 0029 Data: 6410006510006e20006f20ff0121e40c0328120521270008211a360a21abae9923000000009b210000 `

26f1 : Aqara WS-EUK02
b94a : Aqara QBKG21LM

pipiche38, 2023-04-22:
will be fixed in 2.50

closed 2023-04-22 by pipiche38