zigbeefordomoticz/Domoticz-Zigbee #1597: Problems wiszb-120

kuipers250, 2023-04-30:
Frient SIWZB 120
Worked fine, but now not seen more than 24h. Battery level is 100 , signallevel 9
[image]

[wisz120.json.txt](https://github.com/zigbeefordomoticz/Domoticz-Zigbee/files/11361708/wisz120.json.txt)

pipiche38, 2023-04-30:
From the LQI standpoint I'm not sure that is signalevel of 9. But in short the device didn't send any messages in the last 24h.
Reason could be multiple:
- device is out of the network 
- device is reaching the coordinator via a router, and this is bad router ( like OSRAM and some other brands)

As regards to the fact that it was not optimized, I have created a config file.

If you are on bullseye (buster is not supported) you need to move to the develop branch

```
git checkout develop
git pull
sudo python3 -m pip install -r requirements.txt --upgrade
```

kuipers250, 2023-04-30:
Hi

Whats a good router brand?
The other things i will try tomorrow.

Best regards
Gert

Verzonden vanaf Outlook voor Android<https://aka.ms/AAb9ysg>

pipiche38, 2023-05-02:
> Better results can usually be achieved by using mains-powered devices IKEA and Nue/3A Home or dedicated DIY routing devices based on Texas Instruments CC253x/CC26x2 and XBee Series 2/3 Zigbee radios.

extracted from 
https://github.com/zigbeefordomoticz/Domoticz-Zigbee/discussions/1042

You can also find some more informations (with a lot of additional links): https://zigbeefordomoticz.github.io/wiki/en-eng/HowTo_Build-a-ZigBee-network.html

closed 2023-05-02 by pipiche38