zigbeefordomoticz/Domoticz-Zigbee #1598: Siren r7051 to optimized

olive6707, 2023-05-01:
The device is paired correctly and identify as a TS0129 zigbee device, a selector device is created in domoticz and all seems to work armed, disarmed...) . The only thing is the device is not optimised and in the plugin web interface (and the domoticz widget is yellow)

* Product Name : siren R7051
* Brand : WOOX
* URL on a web site: https://www.domadoo.fr/fr/peripheriques/5682-woox-sirene-interieure-intelligente-zigbee-30-8435606701105.html
* What do you expect : Plugin certifcation
* The logs from the pairing process
* export.json file as described here above:
{"PluginInfos":{"HardwareID":28,"HomeFolder":"/home/pi/domoticz/plugins/Domoticz-Zigbee/","StartupFolder":"/home/pi/domoticz/","UserDataFolder":"/home/pi/domoticz/","WebRoot":"","Database":"/home/pi/domoticz/domoticz.db","Language":"fr","Version":"6.4","Author":"pipiche38","Name":"Zigbee","Address":"0.0.0.0","Port":"9999","SerialPort":"/dev/ttyAMA0","Key":"Zigate","Mode1":"ZigpyEZSP","Mode2":"PI","Mode3":"False","Mode4":"9440","Mode6":"2","DomoticzVersion":"2023.1","DomoticzHash":"f9b9ac774","DomoticzBuildTime":"2023-02-14 15:06:40","PluginBranch":"beta6","PluginVersion":"6.4.423","CertifiedDbVersion":"2.46","TimeStamp":1682961190,"available":"6.4.427","available-firmMajor":0,"available-firmMinor":0,"FirmwareVersion":"6.10.0.0 build 169","FirmwareUpdate":false,"PluginUpdate":true,"LogLevel":7,"Zigpy":true,"CoordinatorModel":"Elelabs, ELR02x","CoordinatorFirmwareVersion":"6.10.0.0 build 169","DisplayFirmwareVersion":"Ezsp - 6.10.0.0 build 169","CoordinatorIEEE":"8cf681fffec7d318","NetworkSize":"Total: 15 | Routers: 4 | End Devices: 11","NetworkDevices":{"1002":{"_TZ3000_wamqdr3f":["TS011F"],"_TZ3000_49qchf10":["TS0502A"],"_TZ3000_lbtpiody":["TS0201"],"_TYZB01_ynsiasng":["TS0219"]},"1037":{"unknown":["lumi.sensor_motion.aq2"],"LUMI":["lumi.sensor_motion.aq2"]},"1141":{"_TZ3000_0zrccfgx":["TS0215A-secure-remote"],"_TZE200_nlrfgpny":["TS0601-Solar-Siren"]},"0000":{"eWeLink":["DS01"]}}},"Analytics":0,"Device":{"Version":"3","ZDeviceName":"","Status":"inDB","SQN":"14","Ep":{"01":{"0000":{"0004":"_TYZB01_ynsiasng","0000":"3","0001":"80","0005":"TS0219","0007":2},"0001":{"0020":35,"0021":1},"0502":{"0000":"240","0001":"50","0002":"50","0003":"00"},"0500":{"0002":"alarm1: 0, alarm2: 0, tamper: 0, battery: 0, Support Reporting: 1, restore Reporting: 0, trouble: 0, acmain: 0, test: 0, battdef: 0","0000":"01","0001":"0225","0010":"8cf681fffec7d318"},"ClusterType":{"311":"Voltage","312":"AlarmWD"}}},"Heartbeat":"9","RIA":"4","LQI":68,"Battery":1,"Model":"TS0219","ForceAckCommands":[],"MacCapa":"8e","IEEE":"847127fffe096c12","Type":"Voltage/AlarmWD/Door","ProfileID":{},"ZDeviceID":{},"App Version":"80","Attributes List":{},"DeviceType":"FFD","HW Version":"","Last Cmds":[],"LogicalType":"Router","Manufacturer":"1002","Manufacturer Name":"_TYZB01_ynsiasng","NbEp":"1","PowerSource":"Main","ReadAttributes":{"Ep":{"01":{"0000":{"TimeStamp":1681753381,"iSQN":{"fffe":"00","0004":"09","0000":"09","0001":"09","0005":"09","0007":"09"},"Attributes":{"fffe":"86","0004":"00","0000":"00","0001":"00","0005":"00","0007":"00"},"ZigateRequest":{}},"0001":{"TimeStamp":1681803289,"iSQN":{"0000":"06","0001":"06","0003":"06","0020":"06","0021":"06","0033":"06","0035":"07"},"Attributes":{"0000":"86","0001":"86","0003":"86","0020":"00","0021":"00","0033":"86","0035":"86"},"ZigateRequest":{}},"0502":{"TimeStamp":1682960731,"iSQN":{"0000":"08"},"Attributes":{"0001":"00","0002":"00","0003":"00","0000":"00"},"ZigateRequest":{}},"0500":{"TimeStamp":1682961141,"iSQN":{},"Attributes":{"0002":"00","0000":"00","0001":"00","0010":"00"},"ZigateRequest":{}}}}},"ReceiveOnIdle":"","Stack Version":"","Stamp":{"time":1682961143.4767988,"Time":"2023-05-01 19:12:23","MsgType":"8011","LastSeen":1682961143},"ZCL Version":"3","Health":"Live","IAS":{"01":{"EnrolledStatus":1,"ZoneType":549,"ZoneTypeName":"standard_warning","ZoneStatus":{"alarm1":1,"alarm2":0,"tamper":0,"battery":0,"Support Reporting":1,"Restore Reporting":0,"trouble":0,"acmain":0,"test":0,"battdef":0,"doorbell":0,"GlobalInfos":"1;0;0;0;1;0;0;0;0;0;0","TimeStamp":1682961141}},"Auto-Enrollment":{"Status":"Enrolled","Ep":{"01":{"Status":"Enrolled","TimeStamp":1681753092.123154}}}},"bitfield":"4001","_rawNodeDescriptor":"0000005200522c00008e524001","macapa":"8e","BatteryUpdateTime":1682961035,"BindingTable":{"SQN":0,"Status":"Success","TimeStamp":1681753092.5920558,"BindingTableSize":1,"Devices":[{"847127fffe096c12":{"sourceEp":"01","Cluster":"0500","targetIEEE":"8cf681fffec7d318","targetNickName":"Zigbee Coordinator","targetEp":"01"}}]},"Capability":["Full-Function Device","Main Powered","Receiver during Idle","Standard security","NwkAddr should be allocated"],"Max Tx":"0052","server_mask":"2c00","ReadConfigureReporting":{"Ep":{"01":{"0500":{"0000":{"TimeStamp":1681753093.6409621,"Status":"00","DataType":"30","MinInterval":"003c","MaxInterval":"0384"},"0001":{"TimeStamp":1681753093.641432,"Status":"00","DataType":"31","MinInterval":"003c","MaxInterval":"0384"},"0002":{"TimeStamp":1681753093.6418982,"Status":"00","DataType":"19","MinInterval":"003c","MaxInterval":"0384"}},"0502":{"0000":{"TimeStamp":1681753093.8802574,"Status":"00","DataType":"21","MinInterval":"003c","MaxInterval":"0384","Change":"0001"}}}},"TimeStamp":1681753093.8802443},"descriptor_capability":"00","WriteAttributes":{"Ep":{"01":{"0000":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0500":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0001":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0502":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}}}}},"Max Rx":"0052","Param":{},"Log_UnknowDeviceFlag":1682005471.2363334,"Epv2":{"01":{"ClusterIn":{},"ClusterOut":{},"ProfileID":{},"ZDeviceID":{}}},"CertifiedDevice":false,"ConfigureReporting":{"Ep":{"01":{"0500":{"TimeStamp":1681753092.142894,"iSQN":{"0000":"03","0001":"03","0002":"03"},"Attributes":{"0000":"00","0001":"00","0002":"00"},"ZigateRequest":{}},"0502":{"TimeStamp":1681753092.1469536,"iSQN":{"0000":"04"},"Attributes":{"0000":"00"},"ZigateRequest":{}},"0001":{"TimeStamp":1681753092.1621337,"iSQN":{"0000":"05","0020":"05","0021":"05"},"Attributes":{"0000":"86","0020":"8d","0021":"8d"},"ZigateRequest":{}}}}},"Max Buffer Size":"52","ConsistencyCheck":"ok","RollingLQI":[64,64,64,64,64,64,64,68,68,68,68],"CheckParam":false,"IAS WD":{"MaxDuration":"240"},"ZCLSQN":"10","ZCL-IN-SQN":{"0006":"08"},"IASBattery":100}}

pipiche38, 2023-05-02:
please switch to develop branch ( you need to be on bullseye - buster is not supported anymore )

```
git checkout develop
git pull
sudo python3 -m pip install -r requirements.txt --upgrade
```

olive6707, 2023-05-06:
initialy no change
I deleated the device and try to pair again but paring doesnt't work even if the SIREN go out of pairing mode after 5 sec
[_PluginZigbee_28 (12).log](https://github.com/zigbeefordomoticz/Domoticz-Zigbee/files/11412294/_PluginZigbee_28.12.log)

pipiche38, 2023-05-06:
from the logs it looks like the device was not properly removed.
Make sure that you don't have any widget left on Domoticz and the device is not listed in the plugin. 
Then restart the plugin and do the pairing

olive6707, 2023-05-06:
My mistake I forgot to enable new device in domoticz
Device is green in web plugin but still yellow in domoticz

pipiche38, 2023-05-06:
What do you mean yellow in domoticz ? Is that related to battery ?

can you give me again the Device Information -as you did in the initial post - ?

olive6707, 2023-05-06:
{"PluginInfos":{"HardwareID":28,"HomeFolder":"/home/pi/domoticz/plugins/Domoticz-Zigbee/","StartupFolder":"/home/pi/domoticz/","UserDataFolder":"/home/pi/domoticz/","WebRoot":"","Database":"/home/pi/domoticz/domoticz.db","Language":"fr","Version":"6.4","Author":"pipiche38","Name":"Zigbee","Address":"0.0.0.0","Port":"9999","SerialPort":"/dev/ttyAMA0","Key":"Zigate","Mode1":"ZigpyEZSP","Mode2":"PI","Mode3":"False","Mode4":"9440","Mode6":"2","DomoticzVersion":"2023.1","DomoticzHash":"f9b9ac774","DomoticzBuildTime":"2023-02-14 15:06:40","PluginBranch":"beta6","PluginVersion":"6.4.431","CertifiedDbVersion":"2.56","TimeStamp":1683394110,"available":0,"available-firmMajor":0,"available-firmMinor":0,"FirmwareVersion":"6.10.0.0 build 169","FirmwareUpdate":false,"PluginUpdate":false,"LogLevel":7,"Zigpy":true,"CoordinatorModel":"Elelabs, ELR02x","CoordinatorFirmwareVersion":"6.10.0.0 build 169","DisplayFirmwareVersion":"Ezsp - 6.10.0.0 build 169","CoordinatorIEEE":"8cf681fffec7d318","NetworkSize":"Total: 15 | Routers: 4 | End Devices: 11","NetworkDevices":{"1002":{"_TZ3000_wamqdr3f":["TS011F"],"_TZ3000_49qchf10":["TS0502A"],"_TZ3000_lbtpiody":["TS0201"]},"1037":{"unknown":["lumi.sensor_motion.aq2"],"LUMI":["lumi.sensor_motion.aq2"]},"1141":{"_TZ3000_0zrccfgx":["TS0215A-secure-remote"],"_TZE200_nlrfgpny":["TS0601-Solar-Siren"]},"0000":{"eWeLink":["DS01"]}}},"Analytics":-1,"Device":{"Version":"3","ZDeviceName":"","Status":"inDB","SQN":"01","Heartbeat":"2925","RIA":"0","LQI":80,"Battery":200,"Model":"TS0219","ForceAckCommands":[],"MacCapa":"8e","IEEE":"847127fffe096c12","Type":"AlarmWD","ProfileID":"0104","ZDeviceID":"0403","App Version":80,"Attributes List":{},"DeviceType":"FFD","HW Version":"","Last Cmds":[],"LogicalType":"Router","Manufacturer":"0000","Manufacturer Name":"_TYZB01_ynsiasng","NbEp":"1","PowerSource":"Main","ReadAttributes":{"Ep":{"01":{"0000":{"TimeStamp":1683379370,"iSQN":{"0004":"08","0000":"08","0001":"08","0005":"08","0007":"08","fffe":"08"},"Attributes":{},"ZigateRequest":{}},"0502":{"TimeStamp":1683386373,"iSQN":{"0000":"07"},"Attributes":{"0001":"00","0002":"00","0003":"00","0000":"00"},"ZigateRequest":{}},"0500":{"TimeStamp":1683386554,"iSQN":{},"Attributes":{"0010":"00","0000":"00","0001":"00","0002":"00"},"ZigateRequest":{}},"0001":{"TimeStamp":1683379370,"iSQN":{"0020":"06","0021":"06"},"Attributes":{},"ZigateRequest":{}}}}},"ReceiveOnIdle":"","Stack Version":"","Stamp":{"time":1683386654.8376231,"Time":"2023-05-06 17:24:14","MsgType":"8102","LastSeen":1683386654},"ZCL Version":3,"Health":"Not Reachable","Capability":["Full-Function Device","Main Powered","Receiver during Idle","Standard security","NwkAddr should be allocated"],"ZCLSQN":"fd","ZDPSQN":"f7","RollingLQI":[80,80,80,80,80,80,80,80,80,80,80],"Param":{"BatteryDevice":1,"VoltageConverter":10,"BatteryPercentageConverter":2},"ConfigSource":"DeviceConf","Ep":{"01":{"0000":{"0004":"_TYZB01_ynsiasng","0000":3,"0001":80,"0005":"TS0219","0007":2},"0003":{},"0004":{},"0009":{},"0019":{},"0500":{"0010":"8cf681fffec7d318","0000":"01","0001":"0225","0002":"alarm1: 0, alarm2: 0, tamper: 0, batter: 0, srepor: 1, rrepor: 0, troubl: 0, acmain: 0, test: 0, batdef: 0, doorbell: 0"},"0502":{"0000":"240","0001":"50","0002":"50","0003":"00"},"Type":"AlarmWD","0001":{"0020":50,"0021":200},"ClusterType":{"314":"AlarmWD"}}},"WriteAttributes":{"Ep":{"01":{"0000":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0003":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0004":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0009":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0019":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0500":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0502":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0001":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}}}}},"Epv2":{"01":{"ProfileID":"0104","ZDeviceID":"0403","ClusterIn":{"0000":{},"0001":{},"0003":{},"0004":{},"0005":{},"0500":{},"0502":{},"eeff":{}},"ClusterOut":{"0003":{},"0019":{}}}},"ZDeviceVersion":"0000","BatteryUpdateTime":1683386654,"IAS":{"Auto-Enrollment":{"Status":"Enrolled","Ep":{"01":{"Status":"Enrolled","ZoneId":"00"}}},"01":{"EnrolledStatus":1,"ZoneType":549,"ZoneTypeName":"standard_warning","ZoneStatus":{"alarm1":0,"alarm2":0,"tamper":0,"battery":0,"Support Reporting":1,"Restore Reporting":0,"trouble":0,"acmain":0,"test":0,"battdef":0,"doorbell":0,"GlobalInfos":"0;0;0;0;1;0;0;0;0;0;0","TimeStamp":1683386554}}},"IAS WD":{"MaxDuration":"240"},"PairingInProgress":false,"PairingTime":1683379370.0937493,"BindingTable":{"SQN":0,"Status":"Success","TimeStamp":1683379370.3586962,"BindingTableSize":1,"Devices":[{"847127fffe096c12":{"sourceEp":"01","Cluster":"0500","targetIEEE":"8cf681fffec7d318","targetNickName":"Zigbee Coordinator","targetEp":"01"}}]},"ConfigureReporting":{"Ep":{"01":{"0001":{"TimeStamp":1683379370.0966551,"iSQN":{"0020":"fd","0021":"fd"},"Attributes":{"0021":"8d"},"ZigateRequest":{}}}}},"CheckParam":false,"CertifiedDevice":true,"_rawNodeDescriptor":"0000005200522c00008e524001","Max Buffer Size":"52","Max Rx":"0052","Max Tx":"0052","macapa":"8e","bitfield":"4001","server_mask":"2c00","descriptor_capability":"00","ReadConfigureReporting":{"Ep":{"01":{"0001":{"0021":{"TimeStamp":1683379430.3039024,"Status":"8b"}}}},"TimeStamp":1683379430.303883},"Bind":{"01":{"0001":{"Target":"0000","Stamp":1683386685,"Phase":"requested","Status":"","i_sqn":"f7"}}},"pingDeviceRetry":{"Retry":0,"TimeStamp":1683390255}}}

the widget was yellow but now it is ok

olive6707, 2023-05-08:
on my side all is fine
could be close

pipiche38, 2023-05-08:
Thanks for your help and tests.

commit 83850027 (2023-05-09): Move to certified repo ( https://github.com/zigbeefordomoticz/z4d-certified-devices/commit/fe295a45464e664baa96f2eccb2a64a686b91c6f ) and fix #1598
closed 2023-05-08 by pipiche38