zigbeefordomoticz/Domoticz-Zigbee #1599: [Model Certification] - QT06 Intelligent watering timer

francouai, 2023-05-03:
"QT06 Intelligent watering timer"

Water timer and sensors are not detected (or wrongly) by Domoticz4zigbee.
My  Plugin: 6.4.431 [2.56] 

https://www.aliexpress.com/item/1005004222098040.html

`{"PluginInfos":{"HardwareID":3,"HomeFolder":"/home/pi/domoticz/plugins/Domoticz-Zigbee/","StartupFolder":"/home/pi/domoticz/","UserDataFolder":"/home/pi/domoticz/","WebRoot":"","Database":"/home/pi/domoticz/domoticz.db","Language":"fr","Version":"6.4","Author":"pipiche38","Name":"Zigbee4domoticz","Address":"0.0.0.0","Port":"9999","SerialPort":"/dev/ttyUSB0","Key":"Zigate","Mode1":"ZigpyZNP","Mode2":"USB","Mode3":"False","Mode4":"9440","Mode6":"2","DomoticzVersion":"2023.1","DomoticzHash":"f9b9ac774","DomoticzBuildTime":"2023-02-14 15:06:40","PluginBranch":"beta6","PluginVersion":"6.4.431","CertifiedDbVersion":"2.56","TimeStamp":1683139720,"available":"6.4.427","available-firmMajor":0,"available-firmMinor":0,"FirmwareVersion":"Z-Stack 3.30+ (build 20210708)","FirmwareUpdate":false,"PluginUpdate":false,"LogLevel":7,"Zigpy":true,"CoordinatorModel":"CC1352/CC2652, Z-Stack 3.30+ (znp)","CoordinatorFirmwareVersion":"Z-Stack 3.30+ (build 20210708)","DisplayFirmwareVersion":"Znp - Z-Stack 3.30+ (build 20210708)","CoordinatorIEEE":"00124b002a2e963d","NetworkSize":"Total: 20 | Routers: 11 | End Devices: 9","NetworkDevices":{"1002":{"_TZ3000_n2egfsli":["TS0203"],"_TZ3000_xr3htd96":["TS0201"],"_TZ3000_nkkl7uzv":["TS0207-extender"],"SONOFF":["01MINIZB"]},"1141":{"_TZ3000_xr3htd96":["TS0201"],"_TZ3000_2putqrmw":["TS011F-plug"],"_TZ3040_bb6xaihh":["TS0202"]},"1310":{"AL001":["WG001-Z01"]},"0000":{"eWeLink":["TH01"]}}},"Analytics":1,"Device":{"Version":"3","ZDeviceName":"","Status":"UNKNOW","SQN":"06","Ep":{"01":{"0000":{"0004":"_TZE200_a7sghmms","0000":3,"0001":72,"0005":"TS0601","0007":3,"fffe":0,"ffe2":56,"ffe4":0,"ffdf":"+f+\u0012+e"},"0004":{},"0005":{},"ef00":{},"0019":{},"000a":{}}},"Heartbeat":"6","RIA":"4","LQI":147,"Battery":{},"Model":"TS0601","ForceAckCommands":[],"MacCapa":"80","IEEE":"a4c1384bf62b6b3a","Type":{},"ProfileID":"0104","ZDeviceID":"0051","App Version":72,"Attributes List":{},"DeviceType":"RFD","HW Version":"","Last Cmds":[],"LogicalType":"End Device","Manufacturer":"1141","Manufacturer Name":"_TZE200_a7sghmms","NbEp":"1","PowerSource":"Battery","ReadAttributes":{"Ep":{"01":{"0000":{"TimeStamp":1683139687,"iSQN":{"0004":"00","0000":"00","0001":"00","0005":"00","0007":"00","fffe":"00"},"Attributes":{},"ZigateRequest":{}}}}},"ReceiveOnIdle":"Off","Stack Version":"","Stamp":{"time":1683139710.5738811,"Time":"2023-05-03 20:48:30","MsgType":"8011","LastSeen":1683139710},"ZCL Version":3,"Health":"Live","Capability":["Reduced-Function Device","Standard security","NwkAddr should be allocated"],"ZCLSQN":"00","ZDPSQN":"03","RollingLQI":[147,147,147,147,147,147,147,147,147,147,147],"PairingInProgress":true,"Tuya":{"Valve 0x6b":"00","Valve 0x01":"00","Valve 0x67":"00000000","Valve 0x68":"00000000","Valve 0x69":"00000000","TUYA_MCU_VERSION_RSP":"40"},"Epv2":{"01":{"ProfileID":"0104","ZDeviceID":"0051","ClusterIn":{"0004":{},"0005":{},"ef00":{},"0000":{}},"ClusterOut":{"0019":{},"000a":{}}}},"ReqEpv2":{"01":1683139693.4548557},"ZDeviceVersion":"0001","ConfigSource":"8043","_rawNodeDescriptor":"1141004200422a000080424002","Max Buffer Size":"42","Max Rx":"0042","Max Tx":"0042","macapa":"80","bitfield":"4002","server_mask":"2a00","descriptor_capability":"00","SQN_000a":"04"}}`

pipiche38, 2023-05-04:
Work on going on #1596

Can you switch to the specific branch ?

```
git checkout TS0601-Irrigation-Valve
git pull
sudo python3 -m pip install -r requirements.txt --upgrade
```

restart the plugin and redo the pairing

pipiche38, 2023-05-16:
Available on beta6-6.4.438 [2.62]

closed 2023-05-16 by pipiche38