zigbeefordomoticz/Domoticz-Zigbee #1603: Please add support!  Candeo wall dimmer (mains, EU, 1-gang, zigbee, no neutral, rotary)

gregharewood, 2023-05-09:
* Product Name:  C202 - Candeo Smart Switch LED Dimmer (rotary)
* Brand: Candeo
* URL on a web site: https://www.amazon.co.uk/dp/B091HP42S8?psc=1&ref=ppx_yo2ov_dt_b_product_details  and https://candeo.io/
* What do you expect:  It's mostly good except for "unoptimized" except this - very slow fade when setting a level from Domoticz... lots of little fades to get where it needs to go.  I'm struggling to find or guess parameters that might help
* The logs from the pairing process
* export.json file as described here above

[export.json.zip](https://github.com/zigbeefordomoticz/Domoticz-Zigbee/files/11436545/export.json.zip)

THANKS!!

gregharewood, 2023-05-09:
https://candeo.io/store/zigbee-dimmer-rotary-switch-c202/

pipiche38, 2023-05-10:
if your are on 'develop' branch please update  using the the WebUI plugin update feature

if you are not on develop, but on stable6 , you need to be on bullseye in order to the 'develop' branch working. 

To switch

```
git checkout develop
git pull
sudo python3 -m pip install -r requirements.txt --upgrade
```

after that just restart the plugin

pipiche38, 2023-05-10:
> very slow fade when setting a level from Domoticz... lots of little fades to get where it needs to go. I'm struggling to find or guess parameters that might help

There is no much parameter, it looks like it is a basic dimmer. When searching on the net I found that sometime the Candeo dimmer can produce a lot of Zigbee traffic and that could explain the slow fading.

let see what's happen with the config that I pushed.

gregharewood, 2023-05-11:
Well that killed it....

root@lux:/home/pi/domoticz/plugins/Domoticz-Zigbee# python3 -m pip install -r requirements.txt --upgrade
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting bellows==0.35.1 (from -r requirements.txt (line 1))
  Using cached https://files.pythonhosted.org/packages/7b/90/dba4e28f45f384ad598708e0a5363b2d6fa0c6f6952a11ee6471a13498e8/bellows-0.35.1-py3-none-any.whl
Collecting dnspython==2.3.0 (from -r requirements.txt (line 2))
  Using cached https://files.pythonhosted.org/packages/12/86/d305e87555430ff4630d729420d97dece3b16efcbf2b7d7e974d11b0d86c/dnspython-2.3.0-py3-none-any.whl
Requirement already up-to-date: pyserial>=3.5 in /usr/local/lib/python3.7/dist-packages (from -r requirements.txt (line 3)) (3.5)
Collecting z4d-certified-devices (from -r requirements.txt (line 4))
  Using cached https://files.pythonhosted.org/packages/fc/31/aec8c004f05f0f4d516ab07457331c270d7bde3c12db4d05d6603c5085f9/z4d_certified_devices-2.58-py3-none-any.whl
Collecting zigpy==0.54.1 (from -r requirements.txt (line 5))
  Could not find a version that satisfies the requirement zigpy==0.54.1 (from -r requirements.txt (line 5)) (from versions: 0.0.1, 0.0.2, 0.0.3, 0.1.0, 0.2.0, 0.20.0a1, 0.20a2, 0.20.0, 0.20.1a1, 0.20.1a2, 0.20.1a3, 0.20.1, 0.20.2, 0.20.3, 0.20.4, 0.21.0, 0.22.0, 0.22.1, 0.22.2, 0.23.0, 0.23.1, 0.23.2, 0.24.0, 0.24.1, 0.24.2, 0.24.3, 0.25.0, 0.26.0, 0.27.0, 0.27.1, 0.28.0, 0.28.1, 0.28.2, 0.29.0, 0.30.0, 0.30.1, 0.31.0, 0.32.0, 0.33.0, 0.34.0, 0.35.0, 0.35.1, 0.35.2, 0.36.0, 0.36.1, 0.36.2, 0.37.0, 0.37.1, 0.37.2, 0.38.0, 0.39.0, 0.40.0, 0.41.0, 0.42.0, 0.43.0, 0.44.1, 0.44.2, 0.45.0.dev0, 0.45.0, 0.45.1, 0.46.0, 0.47.0, 0.47.1, 0.47.2, 0.50.1, 0.50.2, 0.50.3, 0.51.0, 0.51.1, 0.51.2, 0.51.3, 0.51.4, 0.51.5, 0.51.6)
No matching distribution found for zigpy==0.54.1 (from -r requirements.txt (line 5))

I'll spend a little to try to work this out myself.... could be the older python version on Raspberry Pi? 

root@lux:/home/pi/domoticz/plugins/Domoticz-Zigbee# python3 -V
Python 3.7.3

pipiche38, 2023-05-11:
I believe you are not on bullseye, so you cannot switch to develop

pipiche38, 2023-05-13:
Closing the issue in between

closed 2023-05-13 by pipiche38