zigbeefordomoticz/Domoticz-Zigbee #1604: Platform.dist() removed since Python3.7

CocoRid, 2023-05-13:
**Describe the bug**
The following error message appeared in Domoticz log, every restart : 
"Error: Zigate: switchPiZigate_mode - unable to find distribution: Assuming debian, Error: module 'platform' has no attribute 'dist'"

Why still using "platform.dist()" or even "platform.linux_distribution()" in "Modules/piZigate.py", it's removed since Python 3.7 ([Platform Unix Platforms](https://docs.python.org/3.5/library/platform.html#unix-platforms)) ?
Can you check to use another function, that is available and not already deprecated (as "distro.linux_distribution()" 😄)?

**To Reproduce**
=> Just restart Domoticz

**Expected behavior**
Not have this error message

**Plugin (please complete the following information):**
 - OS : Raspbian Bullseye
 - Plugin 6.3
 - Plugin stable6

pipiche38, 2023-05-14:
Thanks for that. Would you mind creating a PR ?
I'mnot using any PiZigate nor the Raspbian distrib

pipiche38, 2023-05-19:
@CocoRid is that something you can help and provide a PR ?

CocoRid, 2023-05-20:
Hello
I will do a pull request soon about this.
It's not blocking.

Just : what is expected at lines 55/56 ? Linux distribution will return "raspbian" and not "debian". Actually forced as Debian, i have to see if the fonction called at Line 56 really works.

closed 2023-06-05 by pipiche38