zigbeefordomoticz/Domoticz-Zigbee #1607: [Model Certification] - Aqara Roller Shade Driver E1

cizetta, 2023-05-16:
Product name: Roller Shade Driver E1  -  lumi.curtain.acn002"
Brand: Aqara
URL: https://www.aqara.com/en/product/roller-shade-driver-e1
What do you expect: To be able to configure and control the roller - up and down end to end.

Export json:

`"Version":"3","ZDeviceName":"","Status":"inDB","SQN":"3e","Ep":{"01":{"0000":{"0004":"LUMI","0005":"lumi.curtain.acn002","0000":"3","0001":"30","0002":"2","0003":"1","0006":"10-09-2022","0007":"0"},"0002":{},"0003":{},"0004":{},"0005":{},"0006":{},"0009":{},"000d":{},"0013":{},"0102":{"0000":4,"0007":43,"0008":0,"0010":"0","0011":"65535","0017":8,"fffd":"1"},"000a":{},"0019":{},"ClusterType":{"3":"Switch","4":"Alarm"}},"f2":{}},"Heartbeat":"3496","RIA":"4","LQI":180,"Battery":{},"Model":"lumi.curtain.acn002","ForceAckCommands":[],"MacCapa":"80","IEEE":"54ef44100033b989","Type":"Switch/Alarm/WindowCovering","ProfileID":"0104","ZDeviceID":"0100","App Version":"30","Attributes List":{},"DeviceType":"RFD","HW Version":"1","Last Cmds":[],"LogicalType":"End Device","Manufacturer":"115f","Manufacturer Name":"LUMI","NbEp":"2","PowerSource":"Battery","ReadAttributes":{"Ep":{"01":{"0000":{"TimeStamp":"16/05/2023 14:30:05","iSQN":{"4000":"3a","0004":"36","0005":"36","0000":"36","0001":"37","0002":"37","0003":"37","0006":"38","0007":"38","000a":"38","000f":"39","0010":"39","0015":"39","f000":"3a","ff01":"3a","ff02":"3b"},"Attributes":{"4000":"86","0004":"00","0005":"00","0000":"00","0001":"00","0002":"00","0003":"00","0006":"00","0007":"00","000a":"86","000f":"86","0010":"86","f000":"86","ff02":"86"},"ZigateRequest":{}},"0019":{"TimeStamp":"16/05/2023 14:29:46","iSQN":{"0001":"04","0002":"04","0003":"04","0004":"05","0005":"05","0006":"05","0007":"06","0008":"06","0009":"06","000a":"07"},"Attributes":{},"ZigateRequest":{}},"0102":{"TimeStamp":"16/05/2023 14:30:03","iSQN":{"0000":"30","0001":"30","0002":"30","0003":"31","0004":"31","0007":"31","0008":"32","0009":"32","000a":"32","000b":"33","0010":"33","0011":"33","0014":"34","0017":"34","fffd":"34"},"Attributes":{"0000":"00","0001":"86","0003":"86","0004":"86","0007":"00","0008":"00","0009":"86","000b":"86","0010":"00","0011":"00","0014":"86","0017":"00","fffd":"00"},"ZigateRequest":{}},"0006":{"TimeStamp":"16/05/2023 14:29:46","iSQN":{"4001":"35","4002":"35","0000":"35"},"Attributes":{},"ZigateRequest":{}}},"f2":{"0000":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}}}}},"ReceiveOnIdle":"Off","Stack Version":"2","Stamp":{"time":1684236683.9522934,"Time":"2023-05-16 14:31:23","MsgType":"8002","LastSeen":"16/05/2023 14:31:23"},"ZCL Version":"3","Health":"Live","Capability":["Reduced-Function Device","Standard security","NwkAddr should be allocated"],"ZDPSQN":"0f","Epv2":{"01":{"ClusterIn":{"0000":{},"0002":{},"0003":{},"0004":{},"0005":{},"0006":{},"0009":{},"000d":{},"0013":{},"0102":{}},"ClusterOut":{"000a":{},"0019":{}},"ProfileID":"0104","ZDeviceID":"0100"},"f2":{"ClusterIn":{},"ClusterOut":{},"ProfileID":{},"ZDeviceID":{}}},"ZCLSQN":"3e","RollingLQI":[174,174,174,177,177,177,177,177,177,180,180],"WriteAttributes":{"Ep":{"01":{"0000":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0002":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0003":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0004":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0005":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0006":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0009":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"000d":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0013":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0102":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"000a":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0019":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}}}}},"ZDeviceVersion":"0001","ConfigSource":"8043","PairingInProgress":false,"_rawNodeDescriptor":"115f006400642c0000807f4002","Max Buffer Size":"7f","Max Rx":"0064","Max Tx":"0064","macapa":"80","bitfield":"4002","server_mask":"2c00","descriptor_capability":"00","ZCL-IN-SQN":{"0019":"d4"},"OTAClient":{"ManufacturerCode":"115f","ImageType":"3a80","CurrentImageVersion":"00000f1e"},"PairingTime":1684236586.132312,"Bind":{"01":{"0000":{"Target":"0000","Stamp":"16/05/2023 14:29:46","Phase":"binded","Status":"00","i_sqn":"04"},"0002":{"Target":"0000","Stamp":"16/05/2023 14:29:46","Phase":"binded","Status":"00","i_sqn":"05"},"0003":{"Target":"0000","Stamp":"16/05/2023 14:29:46","Phase":"binded","Status":"00","i_sqn":"06"},"0004":{"Target":"0000","Stamp":"16/05/2023 14:29:46","Phase":"requested","Status":"","i_sqn":"07"},"0005":{"Target":"0000","Stamp":"16/05/2023 14:29:46","Phase":"requested","Status":"","i_sqn":"08"},"0006":{"Target":"0000","Stamp":"16/05/2023 14:29:46","Phase":"requested","Status":"","i_sqn":"09"},"0009":{"Target":"0000","Stamp":"16/05/2023 14:29:46","Phase":"requested","Status":"","i_sqn":"0a"},"000d":{"Target":"0000","Stamp":"16/05/2023 14:29:46","Phase":"requested","Status":"","i_sqn":"0b"},"0013":{"Target":"0000","Stamp":"16/05/2023 14:29:46","Phase":"requested","Status":"","i_sqn":"0c"},"0102":{"Target":"0000","Stamp":"16/05/2023 14:29:46","Phase":"requested","Status":"","i_sqn":"0d"}}},"BindingTable":[{"Devices":[],"SQN":0,"Status":"","TimeStamp":"16/05/2023 14:29:46","Time":"16/05/2023 14:29:46"}],"ConfigureReporting":{"Ep":{"01":{"0006":{"TimeStamp":"16/05/2023 14:29:46","iSQN":{"0000":"01"},"Attributes":{},"ZigateRequest":{}},"0102":{"TimeStamp":"16/05/2023 14:29:46","iSQN":{"0003":"02","0004":"02","0008":"02","0009":"03"},"Attributes":{"0003":"86","0e10":"86","0004":"86","0009":"86"},"ZigateRequest":{}}}}},"CertifiedDevice":false,"SWBUILD_1":"10-09-2022","MapLQI":{"0000":172},"NwkID":"5f06"}`

Thank you in advance!

pipiche38, 2023-05-19:
Unfortunately the device informations are not enough, because Xiaomi/Aqara is using some propitiatory clusters we have to reverse-engineer and understand what they do.

It would be interesting if you could enable some debug parameters from the WebUI > Tools > Debug
- Input
- Lumi
- Clusters

Then when the device is paired with the plugin, do some actions on wit the curtain remote and see if that generate some logs ( Log/PluginZigbee-xx.log )
It would be good if you could like the logs with the remote command actions

Then please drop here the outcome of your findings

cizetta, 2023-05-20:
Hello,

Thank you for your answer. I set the logging settings like that:

[image]

And device is active:

[image]

But I do not get any log entries when pushing its hardware buttons and moving the motor:

`2023-05-20 18:00:52,688 DEBUG   : [ZigpyForwarder_4 1077] ZigateRead - MsgType: 8043,  Data: 00000000010101040400000500000006000a00190501040001002005000502, LQI: 0
2023-05-20 18:00:52,688 DEBUG   : [ZigpyForwarder_4 1077] Decode8043 - Received SQN: 00 Addr: 0000 Len: 01 Status: 00 Data: 00000000010101040400000500000006000a00190501040001002005000502
2023-05-20 18:00:52,690 DEBUG   : [ZigpyForwarder_4 1077] ZigateRead - MsgType: 0302,  Data: , LQI: 0
2023-05-20 18:00:52,690 DEBUG   : [ZigpyForwarder_4 1077] Decode0302 - PDM Load
2023-05-20 18:00:52,692 DEBUG   : [ZigpyForwarder_4 1077] ZigateRead - MsgType: 8002,  Data: 0000008002000002000002000001890000, LQI: 0
2023-05-20 18:00:52,694 DEBUG   : [ZigpyForwarder_4 1077] Decode8042 - Reception of Node Descriptor for 0000 with status 89
2023-05-20 18:00:52,696 DEBUG   : [ZigpyForwarder_4 1077] ZigateRead - MsgType: 8002,  Data: 0000008002000002000002000001890000, LQI: 0
2023-05-20 18:00:52,698 DEBUG   : [ZigpyForwarder_4 1077] Decode8042 - Reception of Node Descriptor for 0000 with status 89
2023-05-20 18:00:59,108 INFO    : [  MainThread 1059] Request Disabling Accepting new Hardware
2023-05-20 18:00:59,169 INFO    : [  ZigpyCom_4 1076] PERMIT-TO-JOIN: {'cmd': 'PERMIT-TO-JOIN', 'datas': {'Duration': 0, 'targetRouter': 'FFFC'}, 'NwkId': None, 'TimeStamp': 1684594859.106921, 'ACKIsDisable': False, 'Sqn': None}
2023-05-20 18:00:59,171 INFO    : [  ZigpyCom_4 1076] PERMIT-TO-JOIN: duration: 0 for Radio: znp for node: None
2023-05-20 18:00:59,172 INFO    : [  ZigpyCom_4 1076] Calling self.app.permit(time_s=0, node=None )
2023-05-20 18:00:59,198 DEBUG   : [ZigpyForwarder_4 1077] ZigateRead - MsgType: 8014,  Data: 00, LQI: 0
2023-05-20 18:00:59,199 DEBUG   : [ZigpyForwarder_4 1077] Decode8014 - Permit Join status: False
2023-05-20 18:00:59,201 INFO    : [ZigpyForwarder_4 1077] Accepting new Hardware: Disable (Off)
2023-05-20 18:00:59,202 DEBUG   : [ZigpyForwarder_4 1077] ---> self.permitTojoin['Starttime']: 1684594859
2023-05-20 18:00:59,207 DEBUG   : [ZigpyForwarder_4 1077] ---> self.permitTojoin['Duration']: 0
2023-05-20 18:00:59,220 DEBUG   : [ZigpyForwarder_4 1077] ---> Current time                  : 1684594859
2023-05-20 18:00:59,224 DEBUG   : [ZigpyForwarder_4 1077] ---> self.Ping['Permit']  (prev)   : None
2023-05-20 18:00:59,227 DEBUG   : [ZigpyForwarder_4 1077] ---> self.Ping['Permit']  (new )   : Off
2023-05-20 18:00:59,229 DEBUG   : [ZigpyForwarder_4 1077] ---> _to_notify                    : True
2023-05-20 18:00:59,230 DEBUG   : [ZigpyForwarder_4 1077] Ping - received
2023-05-20 18:00:59,232 DEBUG   : [ZigpyForwarder_4 1077] ZigateRead - MsgType: 8014,  Data: 00, LQI: 0
2023-05-20 18:00:59,233 DEBUG   : [ZigpyForwarder_4 1077] Decode8014 - Permit Join status: False
2023-05-20 18:00:59,234 DEBUG   : [ZigpyForwarder_4 1077] ---> self.permitTojoin['Starttime']: 1684594859
2023-05-20 18:00:59,235 DEBUG   : [ZigpyForwarder_4 1077] ---> self.permitTojoin['Duration']: 0
2023-05-20 18:00:59,236 DEBUG   : [ZigpyForwarder_4 1077] ---> Current time                  : 1684594859
2023-05-20 18:00:59,237 DEBUG   : [ZigpyForwarder_4 1077] ---> self.Ping['Permit']  (prev)   : Off
2023-05-20 18:00:59,237 DEBUG   : [ZigpyForwarder_4 1077] ---> self.Ping['Permit']  (new )   : Off
2023-05-20 18:00:59,241 DEBUG   : [ZigpyForwarder_4 1077] ---> _to_notify                    : True
2023-05-20 18:00:59,242 DEBUG   : [ZigpyForwarder_4 1077] Ping - received
2023-05-20 18:00:59,243 INFO    : [  ZigpyCom_4 1076] returning from the self.app.permit(time_s=0, node=None )
2023-05-20 18:01:04,071 INFO    : [  MainThread 1059] Switch Blue Led On
2023-05-20 18:01:04,075 INFO    : [  MainThread 1059] coordinator set to Certification : 1/CE -> CE
2023-05-20 18:01:05,214 INFO    : [  MainThread 1059] ota_scan_folder Following Firmware have been loaded 
2023-05-20 18:01:05,217 INFO    : [  MainThread 1059] Plugin with Zigpy, Coordinator CC1352/CC2652, Z-Stack 3.30+ (znp) firmware Znp - Z-Stack 3.30+ (build 20210708) correctly initialized
`

Should I do some additional configuration in order to get debug logs?

Thank you.

cizetta, 2023-06-05:
Hello,

I played a bit with the controller again - deleted it and added it several times. This is the adding log with the debug options enabled as you suggested:

`2023-06-05 08:45:58,715 INFO    : [ MainThread 29248] Request Accepting new Hardware for 240 seconds 
2023-06-05 08:45:58,765 INFO    : [ ZigpyCom_4 29274] PERMIT-TO-JOIN: {'cmd': 'PERMIT-TO-JOIN', 'datas': {'Duration': 240, 'targetRouter': 'FFFC'}, 'NwkId': None, 'TimeStamp': 1685943958.711093, 'ACKIsDisable': False, 'Sqn': None}
2023-06-05 08:45:58,768 INFO    : [ ZigpyCom_4 29274] PERMIT-TO-JOIN: duration: 240 for Radio: znp for node: None
2023-06-05 08:45:58,774 INFO    : [ ZigpyCom_4 29274] Calling self.app.permit(time_s=240, node=None )
2023-06-05 08:45:58,807 DEBUG   : [ZigpyForwarder_4 29275] ZigateRead - MsgType: 8014,  Data: 01, LQI: 0
2023-06-05 08:45:58,808 DEBUG   : [ZigpyForwarder_4 29275] Decode8014 - Permit Join status: True
2023-06-05 08:45:58,810 INFO    : [ZigpyForwarder_4 29275] Accepting new Hardware: Enable (On)
2023-06-05 08:45:58,812 DEBUG   : [ZigpyForwarder_4 29275] ---> self.permitTojoin['Starttime']: 1685943958
2023-06-05 08:45:58,813 DEBUG   : [ZigpyForwarder_4 29275] ---> self.permitTojoin['Duration']: 240
2023-06-05 08:45:58,815 DEBUG   : [ZigpyForwarder_4 29275] ---> Current time                  : 1685943958
2023-06-05 08:45:58,817 DEBUG   : [ZigpyForwarder_4 29275] ---> self.Ping['Permit']  (prev)   : Off
2023-06-05 08:45:58,818 DEBUG   : [ZigpyForwarder_4 29275] ---> self.Ping['Permit']  (new )   : On
2023-06-05 08:45:58,820 DEBUG   : [ZigpyForwarder_4 29275] ---> _to_notify                    : True
2023-06-05 08:45:58,821 DEBUG   : [ZigpyForwarder_4 29275] Ping - received
2023-06-05 08:45:58,837 INFO    : [ ZigpyCom_4 29274] returning from the self.app.permit(time_s=240, node=None )
2023-06-05 08:46:00,097 DEBUG   : [ZigpyForwarder_4 29275] ZigateRead - MsgType: 8002,  Data: 0001040008010102202702000008640a00002024, LQI: 87
2023-06-05 08:46:00,105 DEBUG   : [ZigpyForwarder_4 29275] Decode8102 - Attribute Reports: [2027:01] MsgSQN: 64 ClusterID: 0008 AttributeID: 0000 Status: 00 Type: 20 Size: 0001 ClusterData: >24<
2023-06-05 08:46:00,111 DEBUG   : [ZigpyForwarder_4 29275] scan_attribute_reponse - Sqn: 64 i_sqn: 64 Nwkid: 2027 Ep: 01 Cluster: 0008 MsgData: 64202701000800000020000124 Type: 8102
2023-06-05 08:46:00,113 DEBUG   : [ZigpyForwarder_4 29275] scan_attribute_reponse - 8102 idx: 26 Read Attribute Response: [2027:01] ClusterID: 0008 MsgSQN: 64, i_sqn: 64, AttributeID: 0000 Status: 00 Type: 20 Size: 0001 ClusterData: >24<
2023-06-05 08:46:00,115 DEBUG   : [ZigpyForwarder_4 29275] Decode8102 - LQI:  87 Received Cluster:0008 Attribute: 0000 Value:   24 from (2027/01)
2023-06-05 08:46:00,117 DEBUG   : [ZigpyForwarder_4 29275] ReadCluster - 0008 - 2027/01 AttrId: 0000 AttrType: 20 Attsize: 0001 Status: 00 AttrValue: 24
2023-06-05 08:46:00,120 DEBUG   : [ZigpyForwarder_4 29275] ReadCluster - ClusterID: 0008 Addr: 2027 MsgAttrID: 0000 MsgAttType: 20 MsgAttSize: 0001 MsgClusterData: 24
2023-06-05 08:46:00,124 DEBUG   : [ZigpyForwarder_4 29275] ReadCluster - ClusterId=0008 - 2027/01 Level Control: 24
2023-06-05 08:46:02,421 DEBUG   : [ZigpyForwarder_4 29275] ZigateRead - MsgType: 8002,  Data: 0000000013000002096f020000d36f0989b933001044ef5480, LQI: 0
2023-06-05 08:46:02,429 INFO    : [ZigpyForwarder_4 29275] --> Adding device 096f in self.DevicesInPairingMode
2023-06-05 08:46:02,431 INFO    : [ZigpyForwarder_4 29275] --> ['096f']
2023-06-05 08:46:02,436 INFO    : [ZigpyForwarder_4 29275] [None] NEW OBJECT: 096f None
2023-06-05 08:46:02,550 DEBUG   : [ZigpyForwarder_4 29275] ZigateRead - MsgType: 8011,  Data: 00096f, LQI: 0
2023-06-05 08:46:02,567 DEBUG   : [ZigpyForwarder_4 29275] ZigateRead - MsgType: 8011,  Data: 00096f, LQI: 0
2023-06-05 08:46:02,828 DEBUG   : [ZigpyForwarder_4 29275] ZigateRead - MsgType: 8002,  Data: 0000008006000002096f02000000006f0900, LQI: 0
2023-06-05 08:46:02,832 INFO    : [ZigpyForwarder_4 29275] Decode8046 - Match Descriptor response: SQN: 00 Status: 00 Nwkid: 096f Lenght: 00 List: 
2023-06-05 08:46:03,060 DEBUG   : [ZigpyForwarder_4 29275] ZigateRead - MsgType: 8002,  Data: 0000008005000002096f02000001006f090201f2, LQI: 0
2023-06-05 08:46:03,071 INFO    : [ZigpyForwarder_4 29275] [-] NEW OBJECT: 096f Active Endpoint Response Ep: 01 LQI: 0
2023-06-05 08:46:03,072 INFO    : [ZigpyForwarder_4 29275] [-] NEW OBJECT: 096f Active Endpoint Response Ep: f2 LQI: 0
2023-06-05 08:46:03,076 INFO    : [ZigpyForwarder_4 29275] --> Not full scope
2023-06-05 08:46:03,077 INFO    : [ZigpyForwarder_4 29275] --> Build list of Attributes
2023-06-05 08:46:03,079 INFO    : [ZigpyForwarder_4 29275] --> ListOfEp with 0x0000: []
2023-06-05 08:46:03,080 INFO    : [ZigpyForwarder_4 29275] --> Build list Eps for Cluster Basic ['01', 'f2']
2023-06-05 08:46:03,082 INFO    : [ZigpyForwarder_4 29275] Request Basic  Manufacturer via Read Attribute request: 0004
2023-06-05 08:46:03,083 INFO    : [ZigpyForwarder_4 29275] Request Basic  Model Name via Read Attribute request: 0005
2023-06-05 08:46:03,085 INFO    : [ZigpyForwarder_4 29275] EP: {'01': {}, 'f2': {}} Attributes: [4, 5]
2023-06-05 08:46:03,087 DEBUG   : [ZigpyForwarder_4 29275] get_device_config_param: 096f Config: ReadAttributeChunk
2023-06-05 08:46:03,092 DEBUG   : [ZigpyForwarder_4 29275] get_device_config_param: 096f Config: ReadAttributeChunk
[... cut ...]
2023-06-05 08:46:24,163 DEBUG   : [ZigpyForwarder_4 29275] ReadCluster - 0102 - 096f/01 AttrId: 0007 AttrType: 18 Attsize: 0001 Status: 00 AttrValue: 2b
2023-06-05 08:46:24,164 DEBUG   : [ZigpyForwarder_4 29275] decodeAttribute( 18, 2b) 
2023-06-05 08:46:24,165 DEBUG   : [ZigpyForwarder_4 29275] ReadCluster - 0102 - 096f/01 - Attribute: 0007, Type: 18, Size: 0001 Data: 2b-43
`

However no additional logs are generated when I'm moving the motor with the hardware buttons :(
According to https://zigbee.blakadder.com/Aqara_ZNJLBL01LM.html it is working with some implementations somehow.

pipiche38, 2023-06-10:
This sound better.
I'll work to make a 1st step (reporting device status), but not commendable yet

pipiche38, 2023-06-10:
I just pushed a update on 'develop' branch. As mentioned you must be on python3.8 (bullseye). If you have python3.8 on buster this will not work.

1) remove the device from domoticz and plugin if you have any remaining widgets left

2) To switch to develop

```
git checkout develop
git pull
sudo python3 -m pip install -r requirements.txt --upgrade
```

3) restart the plugin

4) redo the pairing

cizetta, 2023-06-11:
Hello,

Thank you for your feedback!

I'm testing on a RPI with Bullseye and python 3.9. I can confirm that after pulling the development branch and started using it two devices are created - a status/control one and a temperature one:

[image]

[image]

Cheers!

pipiche38, 2023-06-11:
If you command the device do you get some update on DomoticZ?

cizetta, 2023-06-12:
Hello,

I do not see any updates in Domoticz and the logs. However since the curtain roller is not configured there are no set points for fully open and fully closed. 
I see this as a tricky implementation unfortunately.

pipiche38, 2023-06-12:
Aqara are not the easiest one to get integrated.
Not sure how I can help you ?

cizetta, 2023-06-14:
Then just leave it like this :)  I'll play more when I have time in future.

closed 2023-06-14 by cizetta