zigbeefordomoticz/Domoticz-Zigbee #1609: pairing new temperature sensor failed on develop branch

Achterhoeker, 2023-05-23:
Hi,

Pairing a new temperature sensor failed with the following log (accept new devices on in domoticz settings) device is not created in domoticz, but is present in teh plugin

Log from pairing:
2023-05-23 21:36:52.827 Status: zigbee: Accepting new Hardware: Enable (On)
2023-05-23 21:36:58.716 Status: zigbee: [None] NEW OBJECT: 1c31 None
2023-05-23 21:36:59.364 Status: zigbee: [-] NEW OBJECT: 1c31 Active Endpoint Response Ep: 01 LQI: 0
2023-05-23 21:36:59.365 Status: zigbee: [-] NEW OBJECT: 1c31 Request Simple Descriptor for Ep: 01
2023-05-23 21:36:59.842 Status: zigbee: [-] NEW OBJECT: 1c31 Simple Descriptor Response EP: 0x01 LQI: 0
2023-05-23 21:36:59.843 Status: zigbee: [-] NEW OBJECT: 1c31 ProfileID 0104
2023-05-23 21:36:59.844 Status: zigbee: [-] NEW OBJECT: 1c31 ZDeviceID 0051
2023-05-23 21:36:59.844 Status: zigbee: [-] NEW OBJECT: 1c31 Application Version 0001
2023-05-23 21:36:59.845 Status: zigbee: [-] NEW OBJECT: 1c31 Cluster IN Count: 04
2023-05-23 21:36:59.846 Status: zigbee: [-] NEW OBJECT: 1c31 Cluster In 1: 0004 (Groups)
2023-05-23 21:36:59.847 Status: zigbee: [-] NEW OBJECT: 1c31 Cluster In 2: 0005 (Scenes)
2023-05-23 21:36:59.848 Status: zigbee: [-] NEW OBJECT: 1c31 Cluster In 3: ef00
2023-05-23 21:36:59.849 Status: zigbee: [-] NEW OBJECT: 1c31 Cluster In 4: 0000 (Basic)
2023-05-23 21:36:59.849 Status: zigbee: [-] NEW OBJECT: 1c31 Cluster OUT Count: 02
2023-05-23 21:36:59.850 Status: zigbee: [-] NEW OBJECT: 1c31 Cluster Out 1: 0019 (Over-the-Air Upgrade)
2023-05-23 21:36:59.850 Status: zigbee: [-] NEW OBJECT: 1c31 Cluster Out 2: 000a (Time)
2023-05-23 21:37:02.149 Status: zigbee: [0] NEW OBJECT: 1c31 Model Name: TS0601
2023-05-23 21:37:02.155 Status: zigbee: [0] NEW OBJECT: 1c31 Manufacturer Name: _TZE200_yjjdcqsq
2023-05-23 21:37:02.177 Status: zigbee: [None] NEW OBJECT: 1c31 Request Node Descriptor
2023-05-23 21:37:03.597 zigbee: Assisted Pairing: Polling: ['1c31']
2023-05-23 21:37:03.598 zigbee: --> New device: 1c31
2023-05-23 21:37:07.156 Status: zigbee: [1] NEW OBJECT: 1c31 Model Name: TS0601
2023-05-23 21:37:07.162 Status: zigbee: [1] NEW OBJECT: 1c31 Manufacturer Name: _TZE200_yjjdcqsq
2023-05-23 21:37:12.138 Status: zigbee: [2] NEW OBJECT: 1c31 Model Name: TS0601
2023-05-23 21:37:12.144 Status: zigbee: [2] NEW OBJECT: 1c31 Manufacturer Name: _TZE200_yjjdcqsq
2023-05-23 21:37:13.576 zigbee: Assisted Pairing: Polling: ['1c31']
2023-05-23 21:37:13.576 zigbee: --> New device: 1c31
2023-05-23 21:37:17.115 Status: zigbee: [3] NEW OBJECT: 1c31 Model Name: TS0601
2023-05-23 21:37:17.121 Status: zigbee: [3] NEW OBJECT: 1c31 Manufacturer Name: _TZE200_yjjdcqsq
2023-05-23 21:37:17.120 Error: zigbee: processNotinDBDevices - Creat Domo Device Failed !!! for 1c31 status: 8043
2023-05-23 21:37:23.602 zigbee: Assisted Pairing: Polling: ['1c31']
2023-05-23 21:37:23.602 zigbee: --> New device: 1c31
2023-05-23 21:37:23.602 zigbee: --> UNKNOW , removed 1c31 from List
2023-05-23 21:37:23.602 zigbee: ------> New Cluster: {'ClusterId': '0000', 'ClusterDesc': 'Basic'}
2023-05-23 21:37:23.602 zigbee: ------> New Cluster: {'ClusterId': '0004', 'ClusterDesc': 'Groups'}
2023-05-23 21:37:23.602 zigbee: ------> New Cluster: {'ClusterId': '0005', 'ClusterDesc': 'Scenes'}
2023-05-23 21:37:23.602 zigbee: ------> New Cluster: {'ClusterId': 'ef00', 'ClusterDesc': 'Unknown'}
2023-05-23 21:37:23.602 zigbee: ------> New Cluster: {'ClusterId': '0019', 'ClusterDesc': 'Over-the-Air Upgrade'}
2023-05-23 21:37:23.602 zigbee: ------> New Cluster: {'ClusterId': '000a', 'ClusterDesc': 'Time'}
2023-05-23 21:37:23.602 zigbee: ----> New Ep: {'Ep': '01', 'Clusters': [{'ClusterId': '0000', 'ClusterDesc': 'Basic'}, {'ClusterId': '0004', 'ClusterDesc': 'Groups'}, {'ClusterId': '0005', 'ClusterDesc': 'Scenes'}, {'ClusterId': 'ef00', 'ClusterDesc': 'Unknown'}, {'ClusterId': '0019', 'ClusterDesc': 'Over-the-Air Upgrade'}, {'ClusterId': '000a', 'ClusterDesc': 'Time'}]}
2023-05-23 21:37:23.603 zigbee: --> New Device: {'NwkId': '1c31', 'ProvisionStatus': 'Failed', 'ProvisionStatusDesc': 'Failed', 'IEEE': 'a4c138960822e653', 'ProfileId': '0104', 'ProfileIdDesc': 'ZHA', 'ZDeviceID': '0051', 'ZDeviceIDDesc': 'Unknow', 'Model': 'TS0601', 'PluginCertified': 'no', 'Ep': [{'Ep': '01', 'Clusters': [{'ClusterId': '0000', 'ClusterDesc': 'Basic'}, {'ClusterId': '0004', 'ClusterDesc': 'Groups'}, {'ClusterId': '0005', 'ClusterDesc': 'Scenes'}, {'ClusterId': 'ef00', 'ClusterDesc': 'Unknown'}, {'ClusterId': '0019', 'ClusterDesc': 'Over-the-Air Upgrade'}, {'ClusterId': '000a', 'ClusterDesc': 'Time'}]}]}
2023-05-23 21:37:33.605 zigbee: Assisted Pairing: Polling: []
2023-05-23 21:37:33.605 zigbee: --> Empty queue
2023-05-23 21:37:43.608 zigbee: Assisted Pairing: Polling: []
2023-05-23 21:37:43.608 zigbee: --> Empty queue
2023-05-23 21:37:50.792 zigbee: Disable Assisted pairing

db info from device in the plugin:

{"PluginInfos":{"HardwareID":18,"HomeFolder":"/home/ronald/domoticz/plugins/Domoticz-Zigbee/","StartupFolder":"/home/ronald/domoticz/","UserDataFolder":"/home/ronald/domoticz/","WebRoot":"","Database":"/home/ronald/domoticz/domoticz.db","Language":"nl","Version":"6.4","Author":"pipiche38","Name":"zigbee","Address":"0.0.0.0","Port":"9999","SerialPort":"/dev/serial/by-id/usb-1a86_USB_Serial-if00-port0","Key":"Zigate","Mode1":"ZigpyZNP","Mode2":"USB","Mode3":"False","Mode4":"9440","Mode6":"2","DomoticzVersion":"2023.1","DomoticzHash":"f9b9ac774","DomoticzBuildTime":"2023-02-14 15:06:40","PluginBranch":"beta6","PluginVersion":"6.4.438","CertifiedDbVersion":"2.61","TimeStamp":1684871307,"available":"6.4.434","available-firmMajor":0,"available-firmMinor":0,"FirmwareVersion":"Z-Stack 3.30+ (build 20220219)","FirmwareUpdate":false,"PluginUpdate":false,"LogLevel":7,"Zigpy":true,"CoordinatorModel":"CC1352/CC2652, Z-Stack 3.30+ (znp)","CoordinatorFirmwareVersion":"Z-Stack 3.30+ (build 20220219)","DisplayFirmwareVersion":"Znp - Z-Stack 3.30+ (build 20220219)","CoordinatorIEEE":"00124b001caa84ec","NetworkSize":"Total: 49 | Routers: 23 | End Devices: 26","NetworkDevices":{"1002":{"_TZ1800_fcdjzz3s":["TY0202"],"_TZ3000_bguser20":["TS0201"],"_TZ3000_rco1yzb1":["TS004F"],"_TZ3000_tiwq83wk":["TS0202"],"_TZE200_ntcy3xu1":["TS0601-smoke"],"_TZE200_t1blo2bj":["TS0601-_TZE200_t1blo2bj"],"_TZE200_cpmgn2cf":["TS0601-eTRV2"],"_TZ3000_rmjr4ufz":["TS0001"],"_TZ3000_el5kt5im":["TS0502A"]},"1037":{"ADUROLIGHT":["CSW_ADUROLIGHT"],"LUMI":["lumi.remote.b28ac1"]},"1141":{"_TZ3000_au1rjicn":["TS0203"]},"1168":{"LDS":["ZBT-CCTLight-GU100001","ZBT-CCTLight-A0001"]},"1224":{"ROBB smarrt":["ROB_200-004-0"]},"1286":{"SONOFF":["ZBMINI-L","01MINIZB"]},"0000":{"eWeLink":["TH01","MS01"]},"122d":{"AduroSmart Eria":["BDP3001"]},"117c":{"IKEA of Sweden":["TRADFRI onoff switch","TRADFRIbulbE14WSglobeopal470lm","TRADFRI control outlet"]},"126a":{"EcoDim BV":["EcoDim-Zigbee 3.0"]},"124f":{"GLEDOPTO":["GL-B-004P"]}}},"Analytics":1,"Device":{"Version":"3","ZDeviceName":"temp_wwonkamer_zig","Status":"CreateDB","SQN":"d7","Ep":{"01":{"0000":{"0004":"_TZE200_yjjdcqsq","0000":3,"0001":72,"0005":"TS0601","0007":3,"fffe":0,"ffe2":56,"ffe4":1,"ffdf":"+\u0019+\u0007\u0011"},"0004":{},"0005":{},"ef00":{},"0019":{},"000a":{}}},"Heartbeat":"138","RIA":"4","LQI":142,"Battery":{},"Model":"TS0601","ForceAckCommands":[],"MacCapa":"80","IEEE":"a4c138960822e653","Type":{},"ProfileID":"0104","ZDeviceID":"0051","App Version":72,"Attributes List":{},"DeviceType":"RFD","HW Version":"","Last Cmds":[],"LogicalType":"End Device","Manufacturer":"1141","Manufacturer Name":"_TZE200_yjjdcqsq","NbEp":"1","PowerSource":"Battery","ReadAttributes":{"Ep":{"01":{"0000":{"TimeStamp":1684870618,"iSQN":{"0004":"00","0000":"00","0001":"00","0005":"00","0007":"00","fffe":"00"},"Attributes":{},"ZigateRequest":{}}}}},"ReceiveOnIdle":"Off","Stack Version":"","Stamp":{"time":1684871186.729636,"Time":"2023-05-23 21:46:26","MsgType":"8011","LastSeen":1684871186},"ZCL Version":3,"Health":"Live","Capability":["Reduced-Function Device","Standard security","NwkAddr should be allocated"],"ZCLSQN":"00","ZDPSQN":"03","RollingLQI":[145,145,142,142,142,142,142,142,142,142,142],"WriteAttributes":{"Ep":{"01":{"0000":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}}}}},"Epv2":{"01":{"ProfileID":"0104","ZDeviceID":"0051","ClusterIn":{"0004":{},"0005":{},"ef00":{},"0000":{}},"ClusterOut":{"0019":{},"000a":{}}}},"ReqEpv2":{"01":1684870619.3603528},"ZDeviceVersion":"0001","ConfigSource":"8043","PairingInProgress":true,"_rawNodeDescriptor":"1141004200422a000080424002","Max Buffer Size":"42","Max Rx":"0042","Max Tx":"0042","macapa":"80","bitfield":"4002","server_mask":"2a00","descriptor_capability":"00","Tuya":{"TUYA_MCU_VERSION_RSP":"40","Valve 0x01":"00000108"},"SQN_000a":"cc","Param":{},"CheckParam":true}}

pipiche38, 2023-05-24:
Please do upgrade to  Plugin: 6.4.439 [2.65]

Achterhoeker, 2023-05-24:
i've upgraded to 6.4.439, still the same issue with pairing.
I think 2023-05-24 10:54:23.125 Error: zigbee: processNotinDBDevices - Creat Domo Device Failed !!! for 6634 status: 8043
is causing the troubles.

2023-05-24 10:54:01.633 zigbee: Assisted Pairing: Polling: []
2023-05-24 10:54:01.633 zigbee: --> Empty queue
2023-05-24 10:54:05.015 Status: zigbee: [None] NEW OBJECT: 6634 None
2023-05-24 10:54:05.663 Status: zigbee: [-] NEW OBJECT: 6634 Active Endpoint Response Ep: 01 LQI: 0
2023-05-24 10:54:05.664 Status: zigbee: [-] NEW OBJECT: 6634 Request Simple Descriptor for Ep: 01
2023-05-24 10:54:05.946 Status: zigbee: [-] NEW OBJECT: 6634 Simple Descriptor Response EP: 0x01 LQI: 0
2023-05-24 10:54:05.946 Status: zigbee: [-] NEW OBJECT: 6634 ProfileID 0104
2023-05-24 10:54:05.947 Status: zigbee: [-] NEW OBJECT: 6634 ZDeviceID 0051
2023-05-24 10:54:05.948 Status: zigbee: [-] NEW OBJECT: 6634 Application Version 0001
2023-05-24 10:54:05.948 Status: zigbee: [-] NEW OBJECT: 6634 Cluster IN Count: 04
2023-05-24 10:54:05.949 Status: zigbee: [-] NEW OBJECT: 6634 Cluster In 1: 0004 (Groups)
2023-05-24 10:54:05.950 Status: zigbee: [-] NEW OBJECT: 6634 Cluster In 2: 0005 (Scenes)
2023-05-24 10:54:05.951 Status: zigbee: [-] NEW OBJECT: 6634 Cluster In 3: ef00
2023-05-24 10:54:05.952 Status: zigbee: [-] NEW OBJECT: 6634 Cluster In 4: 0000 (Basic)
2023-05-24 10:54:05.952 Status: zigbee: [-] NEW OBJECT: 6634 Cluster OUT Count: 02
2023-05-24 10:54:05.953 Status: zigbee: [-] NEW OBJECT: 6634 Cluster Out 1: 0019 (Over-the-Air Upgrade)
2023-05-24 10:54:05.953 Status: zigbee: [-] NEW OBJECT: 6634 Cluster Out 2: 000a (Time)
2023-05-24 10:54:08.151 Status: zigbee: [0] NEW OBJECT: 6634 Model Name: TS0601
2023-05-24 10:54:08.158 Status: zigbee: [0] NEW OBJECT: 6634 Manufacturer Name: _TZE200_yjjdcqsq
2023-05-24 10:54:08.164 Status: zigbee: [None] NEW OBJECT: 6634 Request Node Descriptor
2023-05-24 10:54:12.215 zigbee: Assisted Pairing: Polling: ['6634']
2023-05-24 10:54:12.215 zigbee: --> New device: 6634
2023-05-24 10:54:13.126 Status: zigbee: [1] NEW OBJECT: 6634 Model Name: TS0601
2023-05-24 10:54:13.189 Status: zigbee: [1] NEW OBJECT: 6634 Manufacturer Name: _TZE200_yjjdcqsq
2023-05-24 10:54:18.140 Status: zigbee: [2] NEW OBJECT: 6634 Model Name: TS0601
2023-05-24 10:54:18.142 Status: zigbee: [2] NEW OBJECT: 6634 Manufacturer Name: _TZE200_yjjdcqsq
2023-05-24 10:54:22.234 zigbee: Assisted Pairing: Polling: ['6634']
2023-05-24 10:54:22.234 zigbee: --> New device: 6634
2023-05-24 10:54:23.146 Status: zigbee: [3] NEW OBJECT: 6634 Model Name: TS0601
2023-05-24 10:54:23.152 Status: zigbee: [3] NEW OBJECT: 6634 Manufacturer Name: _TZE200_yjjdcqsq
2023-05-24 10:54:23.125 Error: zigbee: processNotinDBDevices - Creat Domo Device Failed !!! for 6634 status: 8043
2023-05-24 10:54:32.254 zigbee: Assisted Pairing: Polling: ['6634']
2023-05-24 10:54:32.254 zigbee: --> New device: 6634
2023-05-24 10:54:32.254 zigbee: --> UNKNOW , removed 6634 from List
2023-05-24 10:54:32.254 zigbee: ------> New Cluster: {'ClusterId': '0000', 'ClusterDesc': 'Basic'}
2023-05-24 10:54:32.254 zigbee: ------> New Cluster: {'ClusterId': '0004', 'ClusterDesc': 'Groups'}
2023-05-24 10:54:32.254 zigbee: ------> New Cluster: {'ClusterId': '0005', 'ClusterDesc': 'Scenes'}
2023-05-24 10:54:32.254 zigbee: ------> New Cluster: {'ClusterId': 'ef00', 'ClusterDesc': 'Unknown'}
2023-05-24 10:54:32.254 zigbee: ------> New Cluster: {'ClusterId': '0019', 'ClusterDesc': 'Over-the-Air Upgrade'}
2023-05-24 10:54:32.255 zigbee: ------> New Cluster: {'ClusterId': '000a', 'ClusterDesc': 'Time'}
2023-05-24 10:54:32.255 zigbee: ----> New Ep: {'Ep': '01', 'Clusters': [{'ClusterId': '0000', 'ClusterDesc': 'Basic'}, {'ClusterId': '0004', 'ClusterDesc': 'Groups'}, {'ClusterId': '0005', 'ClusterDesc': 'Scenes'}, {'ClusterId': 'ef00', 'ClusterDesc': 'Unknown'}, {'ClusterId': '0019', 'ClusterDesc': 'Over-the-Air Upgrade'}, {'ClusterId': '000a', 'ClusterDesc': 'Time'}]}
2023-05-24 10:54:32.255 zigbee: --> New Device: {'NwkId': '6634', 'ProvisionStatus': 'Failed', 'ProvisionStatusDesc': 'Failed', 'IEEE': 'a4c138960822e653', 'ProfileId': '0104', 'ProfileIdDesc': 'ZHA', 'ZDeviceID': '0051', 'ZDeviceIDDesc': 'Unknow', 'Model': 'TS0601', 'PluginCertified': 'no', 'Ep': [{'Ep': '01', 'Clusters': [{'ClusterId': '0000', 'ClusterDesc': 'Basic'}, {'ClusterId': '0004', 'ClusterDesc': 'Groups'}, {'ClusterId': '0005', 'ClusterDesc': 'Scenes'}, {'ClusterId': 'ef00', 'ClusterDesc': 'Unknown'}, {'ClusterId': '0019', 'ClusterDesc': 'Over-the-Air Upgrade'}, {'ClusterId': '000a', 'ClusterDesc': 'Time'}]}]}

Achterhoeker, 2023-05-24:
I updated to the mentioned version and also updated the requirement.txt dependencies. Pairing is now working, thanks for your effort.

closed 2023-05-24 by Achterhoeker