zigbeefordomoticz/Domoticz-Zigbee #1610: [Model Certification] - robb-smarrt-zigbee-dimmer-400w

Achterhoeker, 2023-05-23:
1. Please do provide device information such as Brand and Model name.
     own model from robbshop (nl) ROB_200-004-0
1. Provide the URL where you have purchase the device
     https://www.robbshop.nl/robb-smarrt-zigbee-dimmer-400w
1. Finaly copy-paste all information provided in the log

{"PluginInfos":{"HardwareID":18,"HomeFolder":"/home/ronald/domoticz/plugins/Domoticz-Zigbee/","StartupFolder":"/home/ronald/domoticz/","UserDataFolder":"/home/ronald/domoticz/","WebRoot":"","Database":"/home/ronald/domoticz/domoticz.db","Language":"nl","Version":"6.4","Author":"pipiche38","Name":"zigbee","Address":"0.0.0.0","Port":"9999","SerialPort":"/dev/serial/by-id/usb-1a86_USB_Serial-if00-port0","Key":"Zigate","Mode1":"ZigpyZNP","Mode2":"USB","Mode3":"False","Mode4":"9440","Mode6":"2","DomoticzVersion":"2023.1","DomoticzHash":"f9b9ac774","DomoticzBuildTime":"2023-02-14 15:06:40","PluginBranch":"beta6","PluginVersion":"6.4.438","CertifiedDbVersion":"2.63","TimeStamp":1684873783,"available":"6.4.434","available-firmMajor":0,"available-firmMinor":0,"FirmwareVersion":"Z-Stack 3.30+ (build 20220219)","FirmwareUpdate":false,"PluginUpdate":false,"LogLevel":7,"Zigpy":true,"CoordinatorModel":"CC1352/CC2652, Z-Stack 3.30+ (znp)","CoordinatorFirmwareVersion":"Z-Stack 3.30+ (build 20220219)","DisplayFirmwareVersion":"Znp - Z-Stack 3.30+ (build 20220219)","CoordinatorIEEE":"00124b001caa84ec","NetworkSize":"Total: 50 | Routers: 23 | End Devices: 27","NetworkDevices":{"1002":{"_TZ1800_fcdjzz3s":["TY0202"],"_TZ3000_bguser20":["TS0201"],"_TZ3000_rco1yzb1":["TS004F"],"_TZ3000_tiwq83wk":["TS0202"],"_TZE200_ntcy3xu1":["TS0601-smoke"],"_TZE200_t1blo2bj":["TS0601-_TZE200_t1blo2bj"],"_TZE200_cpmgn2cf":["TS0601-eTRV2"],"_TZ3000_rmjr4ufz":["TS0001"],"_TZ3000_au1rjicn":["TS0203"],"_TZ3000_el5kt5im":["TS0502A"],"_TZE200_yjjdcqsq":["TS0601"]},"1037":{"ADUROLIGHT":["CSW_ADUROLIGHT"],"LUMI":["lumi.remote.b28ac1"]},"1168":{"LDS":["ZBT-CCTLight-GU100001","ZBT-CCTLight-A0001"]},"1224":{"ROBB smarrt":["ROB_200-004-0"]},"1286":{"SONOFF":["ZBMINI-L","01MINIZB"]},"0000":{"eWeLink":["TH01","MS01"]},"122d":{"AduroSmart Eria":["BDP3001"]},"117c":{"IKEA of Sweden":["TRADFRI onoff switch","TRADFRIbulbE14WSglobeopal470lm","TRADFRI control outlet"]},"126a":{"EcoDim BV":["EcoDim-Zigbee 3.0"]},"124f":{"GLEDOPTO":["GL-B-004P"]}}},"Analytics":1,"Device":{"Version":"3","ZDeviceName":"KeukenHoek","Status":"inDB","SQN":"3d","Ep":{"01":{"1000":{},"0000":{"4000":"2.9.2_r56","0004":"ROBB smarrt","0005":"ROB_200-004-0","0000":3,"0001":0,"0002":0,"0003":1,"0006":"NULL","0007":1,"0008":255,"0009":255,"000a":"3030","000b":"NULL","fffd":3},"0003":{},"0004":{},"0005":{},"0006":{"4000":"01","4001":0,"4002":0,"4003":255,"0000":"01","fffd":2},"0008":{"0000":31},"0b05":{},"0019":{"0002":57},"ClusterType":{"2063":"LvlControl"}},"f2":{}},"Heartbeat":"674","RIA":"4","LQI":40,"Battery":{},"Model":"ROB_200-004-0","ForceAckCommands":[],"MacCapa":"8e","IEEE":"385b44fffe2e0bac","Type":"Switch/LvlControl","ProfileID":"0104","ZDeviceID":"0101","App Version":0,"Attributes List":{"Request":{"01":{"1000":1681767203.1322606,"0000":1681767203.1238685,"0003":1681767203.1252677,"0004":1681767203.1268413,"0005":1681767203.1283655,"0006":1681767203.1294732,"0008":1681767203.1303897,"0b05":1681767203.1312962,"0019":1681767203.13318},"f2":{}}},"DeviceType":"FFD","HW Version":1,"Last Cmds":[],"LogicalType":"Router","Manufacturer":"1224","Manufacturer Name":"ROBB smarrt","NbEp":"2","PowerSource":"Main","ReadAttributes":{"Ep":{"01":{"0000":{"TimeStamp":1681754902,"iSQN":{"4000":"13","8000":"13","0000":"0c","0001":"0c","0002":"0c","0003":"0d","0004":"0d","0005":"0d","0006":"0e","0007":"0e","0008":"0e","0009":"0f","000a":"0f","000b":"0f","000c":"10","000d":"10","000e":"10","0010":"11","0011":"11","0012":"11","0013":"12","0014":"12","0015":"12","fffd":"13","fffe":"14","ffdf":"14","ffe0":"14","ffe1":"15","ffe2":"15","ffe4":"15"},"Attributes":{},"ZigateRequest":{}},"0019":{"TimeStamp":1681754902,"iSQN":{"0000":"03","0001":"03","0002":"03","0003":"04","0004":"04","0005":"04","0006":"05","0007":"05","0008":"05","0009":"06","000a":"06","fffd":"06","fffe":"07"},"Attributes":{},"ZigateRequest":{}},"0006":{"TimeStamp":1684870117,"iSQN":{"4000":"08","4001":"08","4002":"09","4003":"09","fffd":"09","fffe":"0a","0000":"01"},"Attributes":{},"ZigateRequest":{}},"0008":{"TimeStamp":1684870117,"iSQN":{"0000":"02"},"Attributes":{},"ZigateRequest":{}}},"f2":{"0000":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}}}}},"ReceiveOnIdle":"On","Stack Version":0,"Stamp":{"time":1684873647.5578113,"Time":"2023-05-23 22:27:27","MsgType":"8102","LastSeen":1684873647},"ZCL Version":3,"Health":"Live","ConfigSource":"8043","NeighbourTableSize":20,"GroupMemberShip":{"01":{}},"SWBUILD_1":"NULL","Capability":["Full-Function Device","Main Powered","Receiver during Idle","Standard security","NwkAddr should be allocated"],"ConfigureReporting":{"Ep":{"01":{"0006":{"TimeStamp":1681754902.178198,"iSQN":{"0000":"01"},"Attributes":{"0000":"00"},"ZigateRequest":{}},"0008":{"TimeStamp":1681754902.184959,"iSQN":{"0000":"02"},"Attributes":{"0000":"00"},"ZigateRequest":{}}}}},"WriteAttributes":{"Ep":{"01":{"1000":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0000":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0003":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0004":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0005":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0006":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0008":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0b05":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0019":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}}}}},"Param":{},"AssociatedDevices":[{"Devices":["7489","31ce","82dd","9bfc"],"SQN":0,"Status":"","TimeStamp":1684871931,"Time":1684871931}],"Log_UnknowDeviceFlag":1682276558.3140337,"Neighbours":[{"SQN":0,"Status":"","TimeStamp":1684871931,"Time":1684871931,"Devices":[{"0000":{"_relationshp":"Parent","_devicetype":"Coordinator","_depth":0,"_lnkqty":91,"_rxonwhenidl":"Rx-On","_IEEE":"00124b001caa84ec","_permitjnt":"??"}},{"1760":{"_relationshp":"Sibling","_devicetype":"Router","_depth":15,"_lnkqty":127,"_rxonwhenidl":"Rx-On","_IEEE":"00124b0026b6b52e","_permitjnt":"??"}},{"1c04":{"_relationshp":"Sibling","_devicetype":"Router","_depth":15,"_lnkqty":178,"_rxonwhenidl":"Rx-On","_IEEE":"00158d0006e15f6c","_permitjnt":"??"}},{"31e3":{"_relationshp":"Sibling","_devicetype":"Router","_depth":15,"_lnkqty":156,"_rxonwhenidl":"Rx-On","_IEEE":"040d84fffe88c9c8","_permitjnt":"??"}},{"3fff":{"_relationshp":"Sibling","_devicetype":"Router","_depth":15,"_lnkqty":193,"_rxonwhenidl":"Rx-On","_IEEE":"a4c1383077632b3e","_permitjnt":"??"}},{"47dd":{"_relationshp":"Sibling","_devicetype":"Router","_depth":15,"_lnkqty":196,"_rxonwhenidl":"Rx-On","_IEEE":"a4c138666548c123","_permitjnt":"??"}},{"486a":{"_relationshp":"Sibling","_devicetype":"Router","_depth":15,"_lnkqty":188,"_rxonwhenidl":"Rx-On","_IEEE":"a4c138dfae401f1b","_permitjnt":"??"}},{"50f2":{"_relationshp":"Sibling","_devicetype":"Router","_depth":15,"_lnkqty":171,"_rxonwhenidl":"Rx-On","_IEEE":"385b44fffe2e14c7","_permitjnt":"??"}},{"5830":{"_relationshp":"Sibling","_devicetype":"Router","_depth":15,"_lnkqty":171,"_rxonwhenidl":"Rx-On","_IEEE":"847127fffe7f0b1f","_permitjnt":"??"}},{"5a5e":{"_relationshp":"Sibling","_devicetype":"Router","_depth":15,"_lnkqty":204,"_rxonwhenidl":"Rx-On","_IEEE":"847127fffe99dc2d","_permitjnt":"??"}},{"5bc9":{"_relationshp":"Sibling","_devicetype":"Router","_depth":15,"_lnkqty":201,"_rxonwhenidl":"Rx-On","_IEEE":"847127fffe7f076f","_permitjnt":"??"}},{"a156":{"_relationshp":"Sibling","_devicetype":"Router","_depth":15,"_lnkqty":207,"_rxonwhenidl":"Rx-On","_IEEE":"847127fffe8cb2e8","_permitjnt":"??"}},{"bb0f":{"_relationshp":"Sibling","_devicetype":"Router","_depth":15,"_lnkqty":174,"_rxonwhenidl":"Rx-On","_IEEE":"385b44fffe2e0be8","_permitjnt":"??"}},{"c8ec":{"_relationshp":"Sibling","_devicetype":"Router","_depth":15,"_lnkqty":191,"_rxonwhenidl":"Rx-On","_IEEE":"84b4dbfffec3be87","_permitjnt":"??"}},{"ec2b":{"_relationshp":"Sibling","_devicetype":"Router","_depth":15,"_lnkqty":141,"_rxonwhenidl":"Rx-On","_IEEE":"00158d0006e12561","_permitjnt":"??"}},{"7489":{"_relationshp":"Child","_devicetype":"End Device","_depth":2,"_lnkqty":208,"_rxonwhenidl":"Rx-Off","_IEEE":"a4c1382a3e530d41","_permitjnt":"Off"}} [... cut ...] uired":0,"NextHopNwkId":"a156"}},{"7110":{"Status":"Active (0)","MemoryConstrained":0,"ManyToOne":0,"RouteRecordRequired":0,"NextHopNwkId":"1760"}}]}],"CertifiedDevice":false,"ConsistencyCheck":"ok","RollingLQI":[105,47,47,47,51,51,105,105,105,40,40],"ZDPSQN":"11","MapLQI":{"1760":63,"5830":168,"7110":34,"0000":51,"a44c":123,"31e3":164,"50f2":172,"486a":123,"5bc9":146,"5a5e":139,"bb0f":177,"297f":62,"a156":170,"c8ec":191,"3fff":109,"47dd":121,"5f53":188,"1c04":96,"ffbb":160},"CheckParam":false}}

pipiche38, 2023-05-24:
Please do upgrade to  Plugin: 6.4.439 [2.65]

Achterhoeker, 2023-05-24:
i've updated to 6.4.439 (and also did the requirements.txt rerun)
The devices are now seen as validated, but in the log after startup i saw now:
2023-05-24 11:03:32.161 Status: zigbee: ------ We have detected a miss configuration reports for device 3ad3 on ep 01 and cluster 0006
2023-05-24 11:03:32.166 Status: zigbee: - Attribut 0000 request to force a Configure Reporting due to field MinInterval '0001' != '0005'
2023-05-24 11:03:32.175 Status: zigbee: - Attribut 0000 request to force a Configure Reporting due to field MaxInterval '012c' != '0e10'

Device 3ad3 is one of the robb-smart dimmers.

pipiche38, 2023-05-24:
I don't see as an issue, except if that is looping in the logs.
The plugin detected a miss-match between its expectations and what was in the device, and it should have correct it.

closed 2023-05-24 by Achterhoeker