zigbeefordomoticz/Domoticz-Zigbee #1612: BlitzWolf  BW-SHP15

agebos, 2023-06-03:
https://www.banggood.com/BlitzWolf-BW-SHP15-Zigbee-3_0-16A-3680W-Smart-Plug-Wireless-Power-Socket-Outlet-EU-Plug-APP-Remote-Control-or-Voice-Control-or-Multiple-Timer-Modes-Compatible-With-Amazon-Alexa-or-Google-Assistant-p-1856492.html?p=CM27171011078201412U&custlinkid=759721&cur_warehouse=CN&ID=557527

The Blitzwolf plug BW-SHP13 works excellently and displays the power status correctly.
The BW-SHP15 is also supported by the ZigbeeForDomoticz Plugin but does not display any power status.
There is a Difference Blitzwolf BW-SHP13 & BW-SHP15
BW-SHP13 doesn’t push status changes, they get pulled. 
BW-SHP15 pushes status changes.

[Desktop.zip](https://github.com/zigbeefordomoticz/Domoticz-Zigbee/files/11643342/Desktop.zip)

pipiche38, 2023-06-10:
I do not understand your issue, I just received a BW-SHP15 and it works as expected.

There is a polling mechanism to get the Instant Power, Current. by default  {'PowerPollingFreq': 300} every 5 minutes.
The current summation is then send by the device when a certain level of energy has been consumed.

I have a Tuya gateway, and can confirm that BW-SHP15 do not push Instant power, but only the summation.

closed 2023-06-10 by pipiche38

agebos, 2023-06-11:
Pipiche38,
I have removed the Device from Domoticz and the Zigbee plugin.
Then updated the (beta)plugin and added the BW-SHP15 again.
Unfortunately no success again. Only the Switch works, and no Power values.
[BWSHP15.zip](https://github.com/zigbeefordomoticz/Domoticz-Zigbee/files/11714240/BWSHP15.zip)

agebos, 2023-06-11:
I also tested the Plug with a SilverCrest Gateway (Lidl), but this also does not show any values. I can, however, switch the Indicate Led on and off.

pipiche38, 2023-06-11:
As mentioned I bought the same plug as I was not able to understand your issue.And mine works as the others on Tuya gateway with smart life app and with a sonoff dongle with the plugin.If in your case you have the same behaviour   , this is probably that the plug is buggy. No ?Envoyé de mon iPhone

agebos, 2023-06-12:
pipiche38,
Thank you very much for exploring the BlitzWolf plug. 
After all your work, I am convinced that my copy of SHP15 is defective and can only be used as a Switch.
For Power measurement I will have to buy a new one.
Thanks again.

From: Pipiche 
Sent: Sunday, June 11, 2023 8:11 PM
To: zigbeefordomoticz/Domoticz-Zigbee 
Cc: agebos ; Author 
Subject: Re: [zigbeefordomoticz/Domoticz-Zigbee] BlitzWolf BW-SHP15 (Issue #1612)

As mentioned I bought the same plug as I was not able to understand your issue.And mine works as the others on Tuya gateway with smart life app and with a sonoff dongle with the plugin.If in your case you have the same behaviour   , this is probably that the plug is buggy. No ?Envoyé de mon iPhone