zigbeefordomoticz/Domoticz-Zigbee #1615: Aqara Smart Outlet - Power meter not working correctly

deve87, 2023-06-10:
- Aqara Smart Outlet
- [https://www.aqara.com/eu/smart_outlet.html](url)

Domoticz: 2023.1 (build 15255)
Zigate USB firmware: 3.23
Zigate plugin: 4.11.017

After paring, I got 3 devices:
- Light/Switch
- Power
- Power/kWh

Power do not update until enabling "pollingLumiPower" in plugin.
kWh is OK, but power is one decimal off.
Domoticz device: 6573W actual 657,3W

For json export file (device 21)
[https://drive.google.com/file/d/11w45OnIwuyViO9Qninq7JjurTY3yhl_J/view?usp=sharing](url)

pipiche38, 2023-06-10:
Plugin 4.11 is not supported anymore.

Can I recommend you to switch to 6.3 version
https://zigbeefordomoticz.github.io/wiki/en-eng/Plugin_Version-6.html

closed 2023-06-10 by pipiche38

deve87, 2023-06-11:
That's embarrassing.. Sorry about that. I just did a "git pull" and did not double check actual newest version and just assumed it was.

Switch to 6.3 now and plug works fine