zigbeefordomoticz/Domoticz-Zigbee #1619: LEXMAN ENKI Shutters

grobihh, 2023-06-20:
* Product Name: Modulo Zigbee conectado para persianas LEXMAN ENKI (Lexman Enki Zigbee for Shutters)
* Brand: Lexman (Adeo)
* URL on a web site: https://www.leroymerlin.es/productos/electricidad-y-domotica/videoporteros-y-motores-para-puertas/motores-para-persianas-y-toldos/modulo-zigbee-conectado-para-persianas-lexman-enki-83424577.html
* Manual: https://media.adeo.com/marketplace/LMES/83424577/3284420.pdf
* What do you expect: Modul should open and close the shutters entirely or report a proper status back. Currently it does one step per activation. In Domoticz the device toggles between Open and Close (0%/100%) per activation although the motor did only one step.
* The logs from the pairing process: [pairing.log](https://github.com/zigbeefordomoticz/Domoticz-Zigbee/files/11804042/pairing.log)
* export.json file as described here above: [export.json.txt](https://github.com/zigbeefordomoticz/Domoticz-Zigbee/files/11804053/export.json.txt)

pipiche38, 2023-06-21:
Don't know on which version of the plugin you are, but in all cases make sure that you are on the latest 'development' branch - Supported only with Python3.8 and above ( requires bullseye on PI).

Once you are on the latest version of the develop branch, you should be running 7.0.007 [68]

remove the existing pairing (remove the WindowsCovering widget from DOmoticz and then make sure the device is not existing on the plugin)
restart the plugin
finally redo the pairing

grobihh, 2023-06-21:
Plugin version is "Version 6.3.013".
OS version is "Raspbian GNU/Linux 11 (bullseye)".
Python version is "Python 3.9.2".

I will check the development branch.

grobihh, 2023-06-21:
Plugin version is now "Version 7.0.007".
The device is shown as "optimized" after a new pairing.
The device problem is still unchanged. The motor does only one step and stops.
[_PluginZigbee_13(1).log](https://github.com/zigbeefordomoticz/Domoticz-Zigbee/files/11825963/_PluginZigbee_13.1.log)

pipiche38, 2023-06-22:
Could you enable the following debug parameters

ZclCluster
ZclCommand
ZclDecoder
Command

then can you do some manual action on your blind (via the module, or the wired switch) to open or close ) ?

finally do a Open or Close from the Domoticz widget

and the provide the logs.

This module is an OEM of [Nodon](https://zigbee.blakadder.com/NodOn_SIN-4-RS-20_PRO.html) which work perfectly.

grobihh, 2023-06-26:
Interesting. Thanks for the Nodon link.
My manual do not have the calibration section.
When running calibration my Lexman device did the same thing. Seems it cannot determine the motor operation.
So I will also check the wiring. Maybe that device is not suitable for my motor.

pipiche38, 2023-06-26:
let us know

grobihh, 2023-07-05:
I installed LoraTap TS130F instead.
Works out of the box without problems.
I guess we can close this issue.

closed 2023-08-08 by pipiche38