zigbeefordomoticz/Domoticz-Zigbee #1624: Philips HUE switch RWL022

leosmiers, 2023-07-01:
[image]

1. Got to the Tools Menu
1. Select Plugin Raw Devices (json)
1. Export

This will create an export.json file that you can save.
   
Finally you create an issue on GitHub with the following details

* Product Name 
* Brand
* URL on a web site
* What do you expect
* The logs from the pairing process
* export.json file as described here above

[export_json.txt](https://github.com/zigbeefordomoticz/Domoticz-Zigbee/files/11925851/export_json.txt)

pipiche38, 2023-07-03:
Please try version  beta7-7.0.013 [2.70]

leosmiers, 2023-07-03:
Thanks for the fast reply

At the moment I am at stable6. When I switch to develop (sudo git checkout develop) I get the following error when I restart domoticz:

`
2023-07-03 19:40:32.122  Error: Zigate: (Zigate) failed to load 'plugin.py', Python Path used was '/home/pi/domoticz/plugins/Domoticz-Zigate/:/usr/lib/python38.zip:/usr/lib/python3.8:/usr/lib/python3.8/lib-dynload:/usr/lib/python3.8/site-packages'.
2023-07-03 19:40:32.132  Error: Zigate: Traceback (most recent call last):
2023-07-03 19:40:32.132  Error: Zigate:   File "/home/pi/domoticz/plugins/Domoticz-Zigate/plugin.py", line 112, in <module>
2023-07-03 19:40:32.132  Error: Zigate:     from Classes.WebServer.WebServer import WebServer
2023-07-03 19:40:32.132  Error: Zigate:   File "/home/pi/domoticz/plugins/Domoticz-Zigate/Classes/WebServer/WebServer.py", line 55, in <module>
2023-07-03 19:40:32.133  Error: Zigate:     class WebServer(object):
2023-07-03 19:40:32.133  Error: Zigate:   File "/home/pi/domoticz/plugins/Domoticz-Zigate/Classes/WebServer/WebServer.py", line 86, in WebServer
2023-07-03 19:40:32.133  Error: Zigate:     from Classes.WebServer.rest_PluginUpgrade import (
2023-07-03 19:40:32.133  Error: Zigate:   File "/home/pi/domoticz/plugins/Domoticz-Zigate/Classes/WebServer/rest_PluginUpgrade.py", line 6, in <module>
2023-07-03 19:40:32.133  Error: Zigate:     import z4d_certified_devices
2023-07-03 19:40:32.134  Error: Zigate: ModuleNotFoundError: No module named 'z4d_certified_devices'
`

Any suggestions?

leosmiers, 2023-07-03:
It seems that I have to rerun 'pip install -r requirements.txt'   8-(

pipiche38, 2023-07-03:
better to use the update feature from teh WebUI > Plugin > Admin this will do all for you

otherwise you must do
`
sudo python3 -m pip install -r requirements.txt --upgrade`

pipiche38, 2023-07-06:
any update @leosmiers

leosmiers, 2023-07-06:
I have not made much progress. As said before I aam currently on stable6 and to test this I have switch to develop.
When I switch to develop and start domoticz in the log file the following appears:

`2023-07-06 17:08:11.534  Error: Zigate: Call to function 'onStart' failed, exception details:
2023-07-06 17:08:11.544  Error: Zigate: Traceback (most recent call last):
2023-07-06 17:08:11.544  Error: Zigate:   File "/home/pi/domoticz/plugins/Domoticz-Zigate/plugin.py", line 1562, in onStart
2023-07-06 17:08:11.544  Error: Zigate:     _plugin.onStart()
2023-07-06 17:08:11.544  Error: Zigate:   File "/home/pi/domoticz/plugins/Domoticz-Zigate/plugin.py", line 391, in onStart
2023-07-06 17:08:11.544  Error: Zigate:     self.log.openLogFile()
2023-07-06 17:08:11.544  Error: Zigate:   File "/home/pi/domoticz/plugins/Domoticz-Zigate/Classes/LoggingManagement.py", line 142, in openLogFile
2023-07-06 17:08:11.544  Error: Zigate:     self.open_logging_mode()
2023-07-06 17:08:11.544  Error: Zigate:   File "/home/pi/domoticz/plugins/Domoticz-Zigate/Classes/LoggingManagement.py", line 169, in open_logging_mode
2023-07-06 17:08:11.544  Error: Zigate:     logging.basicConfig(
2023-07-06 17:08:11.544  Error: Zigate:   File "/usr/lib/python3.8/logging/__init__.py", line 1997, in basicConfig
2023-07-06 17:08:11.544  Error: Zigate:     raise ValueError('Unrecognised argument(s): %s' % keys)
2023-07-06 17:08:11.544  Error: Zigate: ValueError: Unrecognised argument(s): encoding`

I have attached the commands I gave with output (commands.txt) and the relevant part from the domoticz and PluginZigbee_10 logs.

Thanks for looking into it

[commands.txt](https://github.com/zigbeefordomoticz/Domoticz-Zigbee/files/11969916/commands.txt)
[domoticz.log](https://github.com/zigbeefordomoticz/Domoticz-Zigbee/files/11969918/domoticz.log)
[PluginZigbee_10.log](https://github.com/zigbeefordomoticz/Domoticz-Zigbee/files/11969919/PluginZigbee_10.log)

pipiche38, 2023-07-06:
can you share the Conf/PluginConf-xx.json file ?

pipiche38, 2023-07-06:
Can you do a git pull to retreive the latest develop, and check if you still have the issue ?

leosmiers, 2023-07-06:
Just te be clear. When I switch from stable6 to develop I do the following commands:
sudo systemctl stop domoticz
sudo git checkout develop
sudo git pull
sudo python3 -m pip install -r requirements.txt --upgrade
sudo systemctl start domoticz

leosmiers, 2023-07-06:
I have done another pull. And the development version is now running.

Now see it the switch is ok 8-)

pipiche38, 2023-07-06:
> 

Good news. It looks like we where using something not supported on that version of python3 you are running.

leosmiers, 2023-07-06:
Good that you found that.

Now when I try to pair the switch I get the following error:

`NXP_Extended_Error_Code - Extended Error Code: [81] No free APDUs (resource error) - the number of APDUs is set in the Instances property of the appropriate APDU child of the PDU Manager section of the ZPS Configuration Editor`

leosmiers, 2023-07-06:
Here is the errors file (it is the last error)
[errors-json.txt](https://github.com/zigbeefordomoticz/Domoticz-Zigbee/files/11970851/errors-json.txt)

pipiche38, 2023-07-06:
That is a Zigate error. I cannot do much against that oneEnvoyé de mon iPhone

leosmiers, 2023-07-06:
Ok. thanks for your help. I will look further for this new problem

leosmiers, 2023-07-07:
Hello

I have updated the firmware in my Zigate+ with the following command:

C:\NXP\DK6ProductionFlashProgrammer\DK6Programmer.exe -s COM5 -P 500000 -e FLASH -Y -p .\ZiGatev2-322.bin 

Now the plugin complaines that the corrector is not correctly initialized. I have added the log file.

[_PluginZigbee_10(3).log](https://github.com/zigbeefordomoticz/Domoticz-Zigbee/files/11982673/_PluginZigbee_10.3.log)

leosmiers, 2023-07-07:
I seem to have fixed it. In another post I saw that you asked if they set ' Initialize Coordinator: to True ' I did this and now it works again

leosmiers, 2023-07-09:
I seem to have the develop version of the ZigbeePlugin now running. I have tried to pair the RWL022 device. The result is that I see 1 device in the devices list.

'<html><body>

Zigate | 001788010bddf6a4 | 10 | Zigate - Hue_Remote_LvlControl-001788010bddf6a4-02 | Light/Switch | Switch | Off
-- | -- | -- | -- | -- | -- | --

</body>
</html>'

I would have expected a few more. The device also does not react to any button presses.

I have attached a file with the relevant data from the PluginZigbee_10.log file.

[PluginZigbee.log](https://github.com/zigbeefordomoticz/Domoticz-Zigbee/files/11995718/PluginZigbee.log)

pipiche38, 2023-07-09:
What do you expect more ?

As you are on Zigate, I need to look how to get the Philips Hue bounded to the Zigate.

Can you try to enable 
debug Philips 
debug Inputs

then press few time on the remote and drop the PluginZigbee.log here

leosmiers, 2023-07-09:
There are 4 buttons: on/off; brighter; dimmer and hue. When I first paired the device (plugin stable6, zigate 3.1f) I got 3 (or 4 I do not exactly remember) devices in domoticz. Now I only get 1 device.

I now also have that zigate error: NXP_Extended_Error_Code - Extended Error Code: [81] No free APDUs (resource error) - the number of APDUs is set in the Instances property of the appropriate APDU child of the PDU Manager section of the ZPS Configuration Editor

Maybe I should get that solved first.

Any way when I press any of the 4 buttons I do not see any loggings in the PlugibZigbee.log (I have set the asked debug values).

Can you tell me what the best place is to report the zigate error?

I do appriciate all the time and effort that you put in to this, thanks

pipiche38, 2023-07-10:
For the Zigate error you need to report to the Zigate vendor.

leosmiers, 2023-07-10:
Hello

I found an answer to the Zigate problem. It advised to install version 3.a0. (https://zigate.fr/forum/topic/error-zigate-nxp_extended_error_code-extended-error-code-81-no-free-apdus/)

However if start the plugin I get the following errors

'2023-07-10 19:32:18,057 INFO    :b'_coordinator_ready transport: V2-USB PDMready: True'
2023-07-10 19:32:18,066 INFO    :b"onHeartbeat - busy = False, Health: {'Flag': 4, 'Txt': 'Startup'}, startZigateNeeded: False/13, InitPhase1: True InitPhase2: False, InitPhase3: False PDM_LOCK: False ErasePDMinProgress: False ErasePDMDone: False"
2023-07-10 19:32:18,067 INFO    :b'zigate_get_nwk_state'
2023-07-10 19:32:18,068 ERROR   : [       MainThread] We are having difficulties to start coordinator. Basically we are not receiving what we expect from CIE
2023-07-10 19:32:18,084 ERROR   : [       MainThread] Plugin is not started ...
2023-07-10 19:32:18,140 INFO    :b'ZigateRead - MsgType: 8000,  Data: 0000000900ee0000, LQI: 0'
2023-07-10 19:32:18,141 INFO    :b'ZigateRead - MsgType: 8009,  Data: ffff00158d000500737ad7c900000000000000000b, LQI: 0'
2023-07-10 19:32:18,142 INFO    :b'Decode8009: Network state - Address:ffff extaddr:00158d000500737a PanID: d7c9 Channel: 11'
2023-07-10 19:32:18,143 ERROR   : [ZiGateForwarder_10] Coordinator not correctly initialized'

I have added the plugin logging.

[PluginZigbee_10_copy.log](https://github.com/zigbeefordomoticz/Domoticz-Zigbee/files/12004890/PluginZigbee_10_copy.log)

leosmiers, 2023-07-10:
Wait a moment. After I did a soft reset of the Coordinator (from the GUI) the coordinator is initialized correctly.

leosmiers, 2023-07-10:
And after initializing the coordinator from the Domomicz hardware screen. It seems to work all fine. I can add the RWL022 and see the on/off button. May be I do not understand the hue RWL022 device but I would expect that if I press the 'on/off' button on the hue device I would see this in the Domoticz device screen also change. But I do not see anything.

Do you know if a RWL021 does the same?

Any way I now run the develop version of the plugin with the Zigate firware 3.a0 and so far it seems ok.

thanks again

leosmiers, 2023-07-12:
I have decided to go back to stable6 and zigate version 3.22. The develop version and zigate 3.a0 does not seem stable enough for me.

pipiche38, 2023-07-13:
What are the stability issues ? We are closed to push to stable .
can you share the file Conf/PluginConf-xx.json ?

leosmiers, 2023-07-13:
To help you I have switched back to the develop version. I still use Zigate+ frimware version 3.22. So far (5 minutes) it seems fine. 

One of the things I noticed the the topology reports does not show all end devices while the table in the logs does contain them.

leosmiers, 2023-07-14:
A status update:
After 17 hours the systemseems to ok (develop + zigate+ v3.22) Because I use Zigate v3.22 I still can  not use the Hue RWL022 because of the error metioned before, but Ik think the Domitcz-Zigbee plugin functions fine.

pipiche38, 2023-07-14:
I'm sorry for you.
Unfortunately the Zigate firmware is not at the level where it should be. I have stop development around zigate a year ago, because this was not going in the right direction. Zigate is based on #NXP chipset and the stack provided by NXP is not really open and level of support is nearly none.
So now I'm using Texas Instrument and SiliconLabs based Zigbee coordinator and life is much much simple.
From a plugin development we get a lot of functionalities like the possibility to get backup/restore (even from one key to an other one).

leosmiers, 2023-07-14:
Thank you for your clarification. Is there a coordinator that you would like to advise?

leosmiers, 2023-07-19:
I have installed the sonoff and are now running the develop version without any warnings.

pipiche38, 2023-07-19:
Excellent, great news.
You can switch to stable7 which is the latest stable version, develop will be deprecated soon

git pull
git checkout stable7
sudo python3 -m pip install -r requirements.txt --upgrade

and restart of Domoticz

closed 2023-07-19 by pipiche38