zigbeefordomoticz/Domoticz-Zigbee #1633: New option: WebUI IP

jf67-07, 2023-08-04:
Hi,

There is already an option "WebUI Port" to set the port of the z4d dashboard.

Would it be possible to add an option "WebUI IP" or" WebUI Interface" ?
The default value would be 0.0.0.0, so the server would listen on all interfaces (as today), but it would be possible to set 127.0.0.1 to restrict the interface and to secure the access to the dashboard (by putting an auth proxy in front of z4d or by using a ssh redirection).
Thanks.

pipiche38, 2023-08-04:
I like the idea.
Adding an extra parameters would be challenging, but we could probably change parameters WebUI to be
<port> or <ip:port>
So if you have only a port number then we will bind on 0.0.0.0 and if the pare meter has the format x.y.z.a:pppp then we will bind port pppp on ip x.y.z.a

If that makes sense I'll see to implement it

pipiche38, 2023-08-05:
Unfortunatly not so easy with Domoticz 

https://www.domoticz.com/forum/viewtopic.php?t=40629

jf67-07, 2023-08-05:
Hi,
Sure, it makes sense to use the current WebUI Port.
Let's wait for a solution from the Domoticz team.
Thanks

pipiche38, 2023-09-05:
looks like there is not much willingness on the Domoticz side. 
Sorry for that

closed 2023-09-05 by pipiche38