zigbeefordomoticz/Domoticz-Zigbee #1634: [Model Certification] - Sonoff SNZB-03 Motion Sensor

rcrocus, 2023-08-04:
_Please do provide device information such as Brand and Model name =>_ Sonoff SNZB-03 Motion Sensor
_Provide the URL where you have purchase the device_ =>https://www.domotique-store.fr/domotique/usages/domotique-alarme-securite/detection-intrusion-sans-fil-domotique/detection-mouvements-sans-fil-domotique/1481-sonoff-snzb-03-detecteur-de-mouvements-sans-fil-zigbee-compatible-zigate-conbee2-et-homey.html?search_query=%09SNZB-03&results=179
_Finaly copy-paste all information provided in the log_ 
[pairing.log.txt](https://github.com/zigbeefordomoticz/Domoticz-Zigbee/files/12264142/pairing.log.txt)
[export_non_optimisé.json.txt](https://github.com/zigbeefordomoticz/Domoticz-Zigbee/files/12264143/export_non_optimise.json.txt)
[export.json.txt](https://github.com/zigbeefordomoticz/Domoticz-Zigbee/files/12264144/export.json.txt)

Je suis sur la version du plugin: 7.0.004 [2.80]

pipiche38, 2023-08-08:
Something missing in the logs and in the pairing information. I cannot do much here.

Can you try to do much pairing, may be pair a router before, and pair the motion via the router

rcrocus, 2023-08-09:
Hi,
I've paired several times and now it detects the product but with the wrong model : MS01 (Moisture Sensor) instead of SNZB-03 (Motion Sensor)
Which infirmation ar missing during the pairing information ? If I can help or try something let me know.

[export.json.txt](https://github.com/zigbeefordomoticz/Domoticz-Zigbee/files/12301983/export.json.txt)
[pairing.log.txt](https://github.com/zigbeefordomoticz/Domoticz-Zigbee/files/12301984/pairing.log.txt)

pipiche38, 2023-08-09:
MS01 is a motion sensor from Develco/Frient, so that should be correct. what is wrong ?

rcrocus, 2023-08-09:
Sorry for that. I think the model was wrong but it appears to work well.

rcrocus, 2023-08-09:
The battery level is bad displayed in domoticz and not displayed in the WEB UI of domoticz-zigbee :
[image]

[image]

pipiche38, 2023-08-09:
should be fixed with [2.86] of Certified modules. You need to leave time to the device to report a battery level and get it analysed by the plugin

closed 2023-08-09 by pipiche38