zigbeefordomoticz/Domoticz-Zigbee #1641: Error while trying to get Adjusted Value for Temp Temp+Hum 'result'

CocoRid, 2023-09-03:
**Describe the bug**
Des erreurs apparaissent lors de la remontée de température depuis des capteur de température Xiaomi AQARA, voir le log attaché.
[errors.json.txt](https://github.com/zigbeefordomoticz/Domoticz-Zigbee/files/12506525/errors.json.txt)

**To Reproduce**
Il suffit d'attendre la remontée de données.

**Expected behavior**
Pas d'erreur, qui n'étaient pas présentes dans la version stable6 et Domoticz 2023.1

**Plugin (please complete the following information):**
 - OS: Debian 11
 - Plugin version : 7.1.002
 - Plugin branch stable7

pipiche38, 2023-09-03:
Please could you provide the PluginLogs which can be find in Logs/PluginZigbee-xx.log 
Please provide also the Domoticz logs

CocoRid, 2023-09-03:
Log Domoticz-Zigbee : 
[PluginZigbee_14.log.txt](https://github.com/zigbeefordomoticz/Domoticz-Zigbee/files/12507420/PluginZigbee_14.log.txt)

Log Domoticz : 
[domoticz.txt](https://github.com/zigbeefordomoticz/Domoticz-Zigbee/files/12507470/domoticz.txt)

pipiche38, 2023-09-03:
could you please enable debug parameter DZDB via WebUI > Tools > Debug

[image]

then restart the plugin and after a while drop the PluginZigbee_14.log here

CocoRid, 2023-09-05:
Hello

Full day log available, with specific DZDB log activated => [PluginZigbee_14.log.2023-09-04.txt](https://github.com/zigbeefordomoticz/Domoticz-Zigbee/files/12522743/PluginZigbee_14.log.2023-09-04.txt)

pipiche38, 2023-09-05:
I do not see the restart. I guess that the log have been rotated and you just gave me the latest on.

pipiche38, 2023-09-05:
Can you share what value you put in the Domoticz widget for the Temperature Adjustement ?

pipiche38, 2023-09-05:
can you try to open your browser and query directly the Domotciz API

`http://127.0.0.1/json.htm?type=command&param=getdevices&rid=190'`

seems to respond

`{'ActTime': 1693778551, 'AstrTwilightEnd': '21:50', 'AstrTwilightStart': '05:24', 'CivTwilightEnd': '20:38', 'CivTwilightStart': '06:36', 'DayLength': '13:04', 'NautTwilightEnd': '21:13', 'NautTwilightStart': '06:01', 'ServerTime': '2023-09-04 00:02:31', 'SunAtSouth': '13:37', 'Sunrise': '07:05', 'Sunset': '20:09', 'app_version': '2023.2', 'status': 'OK', 'title': 'Devices'}"`

which seems to be uncomplete !

pipiche38, 2023-09-05:
Please use the forum to continue, this is not a plugin issue

closed 2023-09-05 by pipiche38