zigbeefordomoticz/Domoticz-Zigbee #1720: Motion Sensors Cooldown

atmfrenki, 2024-03-27:
**Describe the bug**

My problem is simular to this: https://forum.phoscon.de/t/silvercrest-lidl-motion-sensor-usage-and-control-logic-problems/759

I have 2 types of motion sensors, Philips Hue Outdoor and Lidl Indoor.
Both sensors have the same problem.

I use them to turn on a light that has to stay on while movement is detected.
At some point both motion sensors send an OFF signal while movement is still going on (philips +-30 seconds and Lidl +- 60 seconds) 
That is not a problem because I made a timer to keep the light on for about 5 minutes after the last OFF signal(Virtual switch).
The problem is, the sensors don't turn ON again while motion keeps happening in the room.
They only turn ON when there has been no motion in the room voor 60 of 70 seconds (have to leave the room).
In domoticz this is a problem, because the ON and OFF signal is a trigger. The senser however does not turn back ON or OFF while it keeps seeing motion. Thats why my light turns off while there is still someone in the room/movement after 5 minutes when the timer runs out.

Somehow the motion sensor "talks" with the zigbee network on another level and negotiates that there is still movement going on. Meanwhile the cooldown periode of 30 seconds or 60 seconds has triggered the OFF switch in ZFD/domoticz, but the sensor still communicates that there is motion going on and won't send the ON signal anymore. (I tried it for 5 minutes 10 minutes 15 minutes 20 minutes....... it always acts the same, until i leave the room for 60/70 seconds, come back in, then the motion sensors sends an ON and OFF signal again.

**Expected behavior**
I expect the motion sensors to keep sending notifications "On and OFF" to domoticz, while movement is still detected when the cooldown of 30 or 60 seconds have past.

Here you see 2 motion activations, while there was constant motion, after 7 minutes I left the room, and after 1 minute/70 seconds I came back, then the new motion was triggered:
[image]

**Desktop (please complete the following information):**
 - OS: Windows 11
 - Browser EDGE

**Plugin (please complete the following information):**
 - Raspberri Pi (bookworm)
 - Plugin: 7.1.011 [4.245]
 - Branch Stable
 - Firmware: Znp - Z-Stack 3.30+ (build 20230507)

**Additional context**
Add any other context about the problem here.

pipiche38, 2024-03-27:
Did you set any specific parameters to the Device ? Did you set any specific settings like the Motion should turn off by 30s ?

resetMotiondelay

On Philips Hue there is a Device parameter 

PIROccupiedToUnoccupiedDelay, which define the time when the PIR must go Off, by default it is set to 30 seconds

atmfrenki, 2024-03-27:
No I left those parameters default, the sensors do turn OFF in domoticz, so I think domoticz receives an OFF signal from the sensor.
Or do you think the OFF signal is not coming from the sensors but Domoticz uses the default value to force an OFF state?

That would explain why the sensors never send a new ON signal while motion is happening.

atmfrenki, 2024-03-27:
I checked the logs in domoticz an see this ON signal:
2024-03-27 12:08:34.290 ZigbeeForDomoticz: UpdateDevice - ( PIR_Badkamer) 1:On
[image]

But in the logs there is no OFF signal, while the sensor has turned OFF

atmfrenki, 2024-03-27:
I migrated a few weeks ago from zigbee2mqtt to zigbeefordomoticz, also updated domoticz to 2024 version.
Before, i did not have this problem. Don't know if it is because of the zigbee plugin or the newer domoticz version.

pipiche38, 2024-03-27:
You need to look in the plugin WebUI under Device Management

see here after

[image]

atmfrenki, 2024-03-28:
Ok, thanks for the right direction! I did'nt look there so it's default now i guess..
[image]

atmfrenki, 2024-04-05:
I've been testing, the lidl PIR did not have this value by default, but the Hue already default did had the value.
PIROccupiedToUnoccupiedDelay': 30 (I guess that is value you pointed me at?)

I have'set this to "10" , I expect it to be seconds.
I did not see any changes in behaviour for both the PIR's
They both need a "Cool down" time with no motion happening, before they respond again.

[image]

[image]

atmfrenki, 2024-04-14:
I tested now by removing all the pre-defined parameters but it did nothing for the lidl motion sensor.

atmfrenki, 2024-04-16:
This is the info about the Lidl motion sensor i found:

Version: "3"
ZDeviceName: "Bewegingsmelder Badkamer"
Status: "inDB"
SQN: "6d"
Ep: Object {"01":{"0000":{"0004":"_TZ1800_fcdjzz3s","0005":"TY0202","0000":1,"0001":67,"0007":3},"0001":{"0021":200,"0020":31},"0003":{},"0500":{"0000":"01","0001":"000d","0002":"alarm1: 1, alarm2: 0, tamper: 0, battery: 0, Support Reporting: 0, restore Reporting: 1, trouble: 0, acmain: 0, test: 0, battdef: 0"},"0b05":{},"0019":{},"Type":"Motion/Tamper","ClusterType":{"10197":"Motion","10198":"Tamper"}}}
Heartbeat: "5036"
RIA: "1"
LQI: 47
Battery: 100
Model: "TY0202"
ForceAckCommands: Array[0] []
MacCapa: "80"
IEEE: "cc86ecfffef02b52"
Type: "Motion/Tamper"
ProfileID: "0104"
ZDeviceID: "0402"
App Version: 67
Attributes List: Object {}
DeviceType: "RFD"
HW Version: ""
Last Cmds: Array[0] []
LogicalType: "End Device"
Manufacturer: "1002"
Manufacturer Name: "_TZ1800_fcdjzz3s"
NbEp: "1"
PowerSource: "Battery"
ReadAttributes: Object {"Ep":{"01":{"0000":{"TimeStamp":"08/04/2024 12:51:00","iSQN":{"0004":"08","0000":"08","0001":"08","0005":"09","0007":"09","fffe":"09"},"Attributes":{"0004":"00","0000":"00","0001":"00","0005":"00","0007":"00","fffe":"86"},"ZigateRequest":{}},"0500":{"TimeStamp":"08/04/2024 12:50:54","iSQN":{},"Attributes":{"0000":"00","0001":"00","0002":"00"},"ZigateRequest":{}},"0001":{"TimeStamp":"16/04/2024 09:19:42","iSQN":{},"Attributes":{"0021":"00","0020":"00"},"ZigateRequest":{}}}}}
ReceiveOnIdle: "Off"
Stack Version: ""
Stamp: Object {"time":1713256334.4425168,"Time":"2024-04-16 10:32:14","MsgType":"8011","LastSeen":"16/04/2024 10:32:14"}
ZCL Version: 1
Health: "Live"
Epv2: Object {"01":{"ProfileID":"0104","ZDeviceID":"0402","ClusterIn":{"0000":{},"0001":{},"0003":{},"0500":{},"0b05":{}},"ClusterOut":{"0019":{}}}}
Max Tx: "0052"
IAS: Object {"Auto-Enrollment":{"Status":"Enrolled","Ep":{"01":{"Status":"Enrolled","ZoneId":"00"}}},"01":{"EnrolledStatus":1,"ZoneType":13,"ZoneTypeName":"motion","ZoneStatus":{"alarm1":1,"alarm2":0,"tamper":0,"battery":0,"Support Reporting":0,"Restore Reporting":1,"trouble":0,"acmain":0,"test":0,"battdef":0,"doorbell":0,"GlobalInfos":"1;0;0;0;0;1;0;0;0;0;0","TimeStamp":"08/04/2024 12:50:54"}}}
WriteAttributes: Object {"Ep":{"01":{"0500":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0000":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0001":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0003":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0b05":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}},"0019":{"TimeStamp":0,"iSQN":{},"Attributes":{},"ZigateRequest":{}}}}}
Capability: Array[3] ["Reduced-Function Device","Standard security","NwkAddr should be allocated"]
BindingTable: Object {"SQN":0,"Status":"Success","TimeStamp":"08/04/2024 12:50:58","BindingTableSize":2,"Devices":[{"cc86ecfffef02b52":{"sourceEp":"01","Cluster":"0500","targetIEEE":"00124b0023933851","targetNickName":"Zigbee Coordinator","targetEp":"01"}},{"cc86ecfffef02b52":{"sourceEp":"01","Cluster":"0001","targetIEEE":"00124b0023933851","targetNickName":"Zigbee Coordinator","targetEp":"01"}}]}
descriptor_capability: "00"
bitfield: "4002"
OTAClient: Object {"ManufacturerCode":"120b","ImageType":"4c04","CurrentImageVersion":"00000043"}
ConfigSource: "DeviceConf"
ReadConfigureReporting: Object {"Ep":{"01":{"0001":{"0020":{"TimeStamp":"08/04/2024 12:51:02","Status":"00","DataType":"20","MinInterval":"0005","MaxInterval":"3840","Change":"01"},"0021":{"TimeStamp":"08/04/2024 12:51:02","Status":"00","DataType":"20","MinInterval":"0005","MaxInterval":"3840","Change":"01"}}}},"TimeStamp":"08/04/2024 12:51:02"}
Max Rx: "0052"
IASBattery: 100
server_mask: "2c00"
CertifiedDevice: true
ConfigureReporting: Object {"Ep":{"01":{"0001":{"TimeStamp":"08/04/2024 12:50:56","iSQN":{"0020":"07","0021":"07"},"Attributes":{"0020":"00","0021":"00"},"ZigateRequest":{}}}}}
Bind: Object {"01":{"0001":{"Target":"0000","Stamp":"08/04/2024 12:50:58","Phase":"binded","Status":"00","i_sqn":"03"}}}
macapa: "80"
Max Buffer Size: "52"
_rawNodeDescriptor: "120b005200522c000080524002"
BatteryUpdateTime: "16/04/2024 09:19:42"
Param: Object {"ReadAttributeChunk":5,"ForceIASRegistrationEp":"01","MotionViaIASAlarm1":1}
CheckParam: false
ConsistencyCheck: "ok"
ZCL-IN-SQN: Object {"0500":"6d"}
RollingLQI: Array[1] [47]
NwkID: "ac4b"

atmfrenki, 2024-04-19:
Ok thank you for the info, than i am going to look for a sensor that works better for this solution. Thanks again!

atmfrenki, 2024-04-19:
The Lidl sensor can not be modified to resolve the problem.
Thanks

closed 2024-04-19 by atmfrenki