zigbeefordomoticz/Domoticz-Zigbee #1761: Model ZTH01 manufactured by Tuya

timmpo, 2024-07-25:
Temperature and Humidity Sensor
Model ZTH01 manufactured by Tuya 
https://zigbee.blakadder.com/Tuya_ZTH01.html

good thing is that all the data is in place, but not in the right place.
logs..

`INFO:Device 0xfe99 (a4:c1:38:ad:a1:86:01:e8) joined the network`
`INFO:b'tuyaReadRawAPS - fe99/01 fcf: 09 sqn: 67 cmd: 02 Payload: 09670201bd01020004000000e4'`
`INFO:b'tuya_responsee - Model: TS0601 Nwkid: fe99/01 dp: 01 dt: 02 data: 000000e4'	<------Temperature in celsius`
`INFO:b'tuya_responsee - Model: TS0601 Nwkid: fe99/01 dp: 02 dt: 02 data: 0000004b'	<------ Hum in %`
`INFO:b'tuya_responsee - Model: TS0601 Nwkid: fe99/01 dp: 03 dt: 04 data: 01'	<------ Unknown`
`INFO:b'tuya_responsee - Model: TS0601 Nwkid: fe99/01 dp: 09 dt: 04 data: 00'	<------ Unknown`

it missidentifies as a wathertimer
`INFO:b'tuya_watertimer_response - Model: TS0601 Nwkid: fe99/01 dp: 03 data type: 04 data: 01'` 

the sensor data ends here: 
tuya.py in function tuya_response line 308
`    elif _ModelName in ("TS0601-Parkside-Watering-Timer"):
        tuya_watertimer_response(
            self, Devices, _ModelName, NwkId, srcEp, ClusterID, dstNWKID, dstEP, dp, datatype, data )
`
[export(5).json](https://github.com/user-attachments/files/16375455/export.5.json)

timmpo, 2024-07-25:
i have done some investigating....
added `_TZE204_yjjdcqsq` in line 59 in tuyaConst.py
changed line 1302 in tuya.py to:
`and self.ListOfDevices[ NwkId ][ 'Manufacturer Name' ] not in ( '_TZE200_qoy0ekbd', '_TZE200_whkgqxse','_TZE204_yjjdcqsq')`
so it dont trigger the if statment (humi /= 10) the value needs not be divided.
added `"TS0601-_TZE200_nklqjk62",` in line 685 in pairingProcess.py now the sensor works but bat state is not working yet

i did this only for test the sensor, this changes must be confirmed or redone the right whay.

pipiche38, 2024-07-26:
can you try to upgrade to [5.286] as for the Certified Db.

you can do it by doing
`
sudo python3 -m pip install -r requirements.txt --upgrade`

and then restart the plugin, this is a much flexible way to integrate this device that changing some of the plugin code

As you can see they are several Tuya Temp/Humi config. I guess one should work with yours (for now I have added your config to TS0601-temphumi, but they are already 2 others possibles one).

https://github.com/zigbeefordomoticz/z4d-certified-devices/blob/main/z4d_certified_devices/Certified/Tuya/TS0601-temphumi.json
https://github.com/zigbeefordomoticz/z4d-certified-devices/blob/main/z4d_certified_devices/Certified/Tuya/TS0601-temphumi-2.json
https://github.com/zigbeefordomoticz/z4d-certified-devices/blob/main/z4d_certified_devices/Certified/Tuya/TS0601-temphumix10.json

If you want to play around you can simply copy the TS0601-temphumix10.json file into the plugin Conf/Local-Devices and it will overwrite the standard config.

PS/
I tried to document as much possible, but they are still work to do
https://github.com/zigbeefordomoticz/wiki/blob/master/en-eng/HowTo_Device-Customization-TS0601.md
https://github.com/zigbeefordomoticz/wiki/blob/master/en-eng/HowTo_Device-Customization.md

timmpo, 2024-07-26:
i upgrade to  Plugin: 7.1.012 [5.287]  now the sensor works without messing around

pipiche38, 2024-07-26:
Super ...