zigbeefordomoticz/Domoticz-Zigbee #1809: 4-channel switch with Temp probe

wvries, 2024-12-04:
* Product Name: 4-channel switch with temperature probe

* Brand: Mumubiz (many different manufacturers on AliExpress)

*URL: https://nl.aliexpress.com/item/1005006822806514.html?spm=a2g0o.productlist.main.55.1976aXqRaXqRrV&algo_pvid=41804ede-33ad-445c-beb9-db6aa2f4cec5&aem_p4p_detail=202412040511331580265526368620000878833&algo_exp_id=41804ede-33ad-445c-beb9-db6aa2f4cec5-27&pdp_npi=4%40dis%21EUR%2129.27%2120.49%21%21%2130.03%2121.02%21%402101c71a17333178935146064e6c1b%2112000038416332343%21sea%21ES%21742000869%21X&curPageLogUid=rY2Oo1xevrrh&utparam-url=scene%3Asearch%7Cquery_from%3A&search_p4p_id=202412040511331580265526368620000878833_8

* When pairing, device is recognised and temp sensor is created and is updated from the device in Domoticz (Working as expected.
4 switches are recognised and created in Domoticz. When switching them on, on the device itself it updates the switches in Domoticz.
When switching in Domoticz, noting happens.
Expectine the switches to switch on / Off.

* The logs from the pairing process: Pairing was successfull.
* export.json file attached.
[export (1).json](https://github.com/user-attachments/files/18008860/export.1.json)

pipiche38, 2024-12-04:
which version of the plugin you must be on 7.016

wvries, 2024-12-05:
Indeed, upgraded and working.
Thanks.

closed 2024-12-05 by wvries