zigbeefordomoticz/Domoticz-Zigbee #1815: Granular REST API call for updating device parameters

wvries, 2024-12-13:
Several zigbee devices have multiple parameters that can be used. Many of them are already exposed to Domotica through a device that is created.

But still many devices have parameters that can only be set through the web interface of Zigbee4Domoticz.

Management -> Device Management

There you have the devices with the green "wheel" at the end of the line. There you can change the parameters of the device and validate them.

[image]

This will send the change to the device. However, the entire database of all devices will be send to all devices, therefore unnecessary creating load on the plugin (as mentioned by pipiche in my conversations with him).

Request is to develop a separate API to change the parameters of only this device.

rest-z4d/1/update-device-param with the appropriate json construct.

Maybe through sending raw zigbee messages ?

pipiche38, 2024-12-13:
I have created a specific branch [restapi-deviceparam](https://github.com/zigbeefordomoticz/Domoticz-Zigbee/tree/restapi-deviceparam) to develop this new api

GET will be used to retrieve the parameters
`wget http://127.0.0.1:9441/rest-z4d/1/device-param/1234` would be providing the parameters of device 0x1234

PUT will be used to update the parameters (the all list needs to be provided, not only the updated one)

```
  curl -X PUT -d '{
       "Param": "{'TuyaAlarmDuration': 300, 'TuyaAlarmLevel': 3, 'TuyaAlarmMelody': 1}",
       "NWKID": "1234"
   }' http://127.0.0.1:9441/rest-z4d/1/device-param
```

I have done the coding, but didn't tested and don't have such possibilities between early next week. So kindly stay tuned

pipiche38, 2024-12-16:
@wvries kindly test it

```
git pull
git checkout restapi-deviceparam
git pull
```

`sudo python3 -m pip install -r requirements.txt --upgrade`

then restart the plugin

This following command will give you the parameters of device 275a
`curl http://127.0.0.1:9440/rest-z4d/1/device-param/275a`

This following command will allow you to add/update the parameter of device 275a. Here we will update "PowerOnAfterOffOn" to 0

```
curl -X PUT -d ' {
"Param": {"PowerOnAfterOffOn": 0},
"NWKID": "275a"
}' http://127.0.0.1:9440/rest-z4d/1/device-param
```

pipiche38, 2024-12-17:
@wvries have you been able to test ?

wvries, 2024-12-17:
Hello Pipiche,

Yes, I have tested it in my test environment.

Both retrieving and setting the parameters work as requested.

It can be merged into the production branch.

Thanks for the changes.

pipiche38, 2024-12-17:
I juste pushed an update, you can use either NWKID or IEEE for pushing the update.

For the GET, you can put ether a nwkid ( 4 characters string) or an ieee ( 16 characters length)

wvries, 2024-12-17:
Tested and working.
Thanks.

I will work on the dzVents script (try to make it as generic as possible to use for other devices also).
Will share it with you when finished (maybe a add-on for future releases / or as comment in the help section somewhere).

pipiche38, 2024-12-17:
Super.
For now, I'll keep that branch ,and will merge into the future release in development.
So you can stay there safe.

And indeed would really appreciate your contribution. You can use the Wiki ( https://github.com/zigbeefordomoticz/wiki ) where there is also a contribution folder

wvries, 2024-12-17:
Hi pipiche,

Although the api call works, it does change the description in the z4d interface of the parameters.
[image]

[image]

Not sure if this is only the display of the parameters or actually the changing of the number of possible parameters in the system.

pipiche38, 2024-12-17:
Not sure I understand.
The Webui display the parameters as they are. So if you change them with the APi you should get them updated in the webui

pipiche38, 2024-12-19:
@wvries please make sure to keep the 3 parameters, if you are pushing only one, at the next plugin restart the missing one will be added with their default, which I guess is not what you want.

```
 {
        "TuyaAlarmDuration": 10,
        "TuyaAlarmLevel": 0,
        "TuyaAlarmMelody": 1
    }
```

pipiche38, 2024-12-24:
Closing the Issue, I'm planning to release this feature as part of 7.1.018

closed 2024-12-24 by pipiche38

pipiche38, 2024-12-28:
release 7.1.018 (on branch stable7) has been done. so if you want you can switch back

```
git checkout stable7
git pull
```

then you go to the WebUI and you do a Plugin Upgrade