zigbeefordomoticz/Domoticz-Zigbee #1843: After upgrade to version stable7-7.1.021 i get the message "DNS resolution timed out for zigate_plugin.pipiche.net after 2 second"

mealgi, 2025-05-11:
Context:
Domoticz Version: 2025.1 (build 16682)
Build Hash: 66acb85e9
Compile Date: 2025-05-10 12:24:54
dzVents Version: 3.1.8
Python Version: 3.9.2 (default, Mar 12 2021, 04:06:34) [GCC 10.2.1 20210110]

Issue:
After the upgrade from version stable7-7.1.016 to version stable7-7.1.021 i get the message "DNS resolution timed out for zigate_plugin.pipiche.net after 2 second"

Many thanks for your help

pipiche38, 2025-05-11:
No issue, it just report that in the allocated time of 2 seconds the plugin was not able to get an internet access.
This could be related to several things. (your DNS server not too responding).

In all case it is a known and a none-issue, the plugin is not impacted except to be aware of a new version .

closed 2025-05-11 by pipiche38

mealgi, 2025-05-11:
Hello

I thank you for your prompt reply.
I have seen that it has non impact on zigbee operations.
However I changed the timeout to 20 secondes and still have the problem.
Could you tell me the port you use to check the upgrade version ?
I will check on my side if the port is activated and not already used by another application
Many thanks for your reply
BR

pipiche38, 2025-05-11:
The plugin is doing a simple DNS query on a TXT record zigate_plugin.pipiche.net