zigbeefordomoticz/Domoticz-Zigbee #1854: Error with Lixee Zlinky - If "HP" in color:

Krakinou, 2025-09-05:
**Describe the bug**
Raspberry pi with domoticz 2025.1 on a debian 12, plugin version  7.1.022 [6.467].
Coordinator Sonoff Zigbee 3.0 USB
Device is a Lixee Zlinky, but I don"t know the firmware version.
Every minute, this error is displayed in the log:

````
{"0":{"0":{"Time":"05/09/2025 21:37:15","PermitToJoin":{"Duration":0,"Starttime":1756053521},"PluginHealth":"Ready","Thread":"ZigpyForwarder_3 1755","nwkid":"None","Module":"TransportFrwder","message":"forwarder_thread - Error while receiving a Coordinator command","context":{"StackTrace":"  File \"/usr/lib/python3.11/threading.py\", line 995, in _bootstrap\n    self._bootstrap_inner()\n  File \"/usr/lib/python3.11/threading.py\", line 1038, in _bootstrap_inner\n    self.run()\n  File \"/usr/lib/python3.11/threading.py\", line 975, in run\n    self._target(*self._args, **self._kwargs)\n  File \"/var/www/domoticz/plugins/Domoticz-Zigbee/Classes/ZigpyTransport/forwarderThread.py\", line 53, in forwarder_thread\n    self.log.logging(\"TransportFrwder\", \"Error\", \"forwarder_thread - Error while receiving a Coordinator command\")\n"}},"1":{"Time":"05/09/2025 21:37:15","PermitToJoin":{"Duration":0,"Starttime":1756053521},"PluginHealth":"Ready","Thread":"ZigpyForwarder_3 1755","nwkid":"None","Module":"TransportWrter","message":"Issue in request 0180020032ff0001040702010102fc38020000182f0a02012523a302000000c003, dumping stack: Traceback (most recent call last):\n  File \"/var/www/domoticz/plugins/Domoticz-Zigbee/Classes/ZigpyTransport/forwarderThread.py\", line 48, in forwarder_thread\n    forward_message(self, message)\n  File \"/var/www/domoticz/plugins/Domoticz-Zigbee/Classes/ZigpyTransport/instrumentation.py\", line 38, in f_out\n    result = f_in(self, message)\n             ^^^^^^^^^^^^^^^^^^^\n  File \"/var/www/domoticz/plugins/Domoticz-Zigbee/Classes/ZigpyTransport/forwarderThread.py\", line 64, in forward_message\n    self.F_out(message)\n  File \"/var/www/domoticz/plugins/Domoticz-Zigbee/plugin.py\", line 773, in processFrame\n    zigbee_receive_message(self, Devices, Data)\n  File \"/var/www/domoticz/plugins/Domoticz-Zigbee/Modules/input.py\", line 165, in zigbee_receive_message\n    _decode_message(self, MsgType, Devices, Data, MsgData, MsgLQI)\n  File \"/var/www/domoticz/plugins/Domoticz-Zigbee/Modules/input.py\", line 172, in _decode_message\n    decoding_method(self, Devices, MsgData, MsgLQI)\n  File \"/var/www/domoticz/plugins/Domoticz-Zigbee/Z4D_decoders/z4d_decoder_Read_Report_Attribute_Rsp.py\", line 52, in Decode8102\n    scan_attribute_reponse(self, Devices, MsgSQN, i_sqn, MsgSrcAddr, MsgSrcEp, MsgClusterId, MsgData, '8102')\n  File \"/var/www/domoticz/plugins/Domoticz-Zigbee/Z4D_decoders/z4d_decoder_Read_Report_Attribute_Rsp.py\", line 88, in scan_attribute_reponse\n    read_report_attributes( self, Devices, msgtype, MsgSQN, MsgSrcAddr, MsgSrcEp, MsgClusterId, MsgAttrID, MsgAttStatus, MsgAttType, MsgAttSize, MsgClusterData, )\n  File \"/var/www/domoticz/plugins/Domoticz-Zigbee/Z4D_decoders/z4d_decoder_Read_Report_Attribute_Rsp.py\", line 121, in read_report_attributes\n    ReadCluster( self, Devices, MsgType, MsgSQN, MsgSrcAddr, MsgSrcEp, MsgClusterId, MsgAttrID, MsgAttStatus, MsgAttType, MsgAttSize, MsgClusterData, Source=MsgType, )\n  File \"/var/www/domoticz/plugins/Domoticz-Zigbee/Modules/readClusters.py\", line 174, in ReadCluster\n    process_cluster_attribute_response( self, Devices, MsgSQN, MsgSrcAddr, MsgSrcEp, MsgClusterId, MsgAttrID, MsgAttType, MsgAttSize, MsgClusterData, Source, )\n  File \"/var/www/domoticz/plugins/Domoticz-Zigbee/Modules/readZclClusters.py\", line 81, in process_cluster_attribute_response\n    func( self, Devices, MsgSrcAddr, MsgSrcEp, MsgClusterId, MsgAttrID, value )\n  File \"/var/www/domoticz/plugins/Domoticz-Zigbee/DevicesModules/custom_zlinky.py\", line 43, in zlinky_clusters\n    zlinky_cluster_metering(self, domoticz_devices, nwkid, ep, cluster, attribut, value)\n  File \"/var/www/domoticz/plugins/Domoticz-Zigbee/DevicesModules/custom_zlinky.py\", line 333, in zlinky_cluster_metering\n    attribute_handlers[attribut]()\n  File \"/var/www/domoticz/plugins/Domoticz-Zigbee/DevicesModules/custom_zlinky.py\", line 318, in <lambda>\n    \"0102\": lambda: _handle_attribut_value(\"0102\", [\"EASF02\", \"HCHP\", \"EJPHPM\", \"BBRHCJW\"], update_color=True, totalize=True),\n                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^\n  File \"/var/www/domoticz/plugins/Domoticz-Zigbee/DevicesModules/custom_zlinky.py\", line 295, in _handle_attribut_value\n    zlinky_set_color_based_on_counter(self, domoticz_devices, nwkid, ep, cluster, attribut, value)\n  File \"/var/www/domoticz/plugins/Domoticz-Zigbee/DevicesModules/custom_zlinky.py\", line 122, in zlinky_set_color_based_on_counter\n    _zlinky_update_color(self, nwkid, op_tarifaire, previous_color, new_color)\n  File \"/var/www/domoticz/plugins/Domoticz-Zigbee/DevicesModules/custom_zlinky.py\", line 169, in _zlinky_update_color\n    ptect_value = _normalize_tempo_color(self, ptect_value)\n                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^\n  File \"/var/www/domoticz/plugins/Domoticz-Zigbee/DevicesModules/custom_zlinky.py\", line 227, in _normalize_tempo_color\n    if \"HP\" in color:\n       ^^^^^^^^^^^^^\nTypeError: argument of type 'NoneType' is not iterable\n","context":{"Message code:":"argument of type 'NoneType' is not iterable","Stack Trace":"Traceback (most recent call last):\n  File \"/var/www/domoticz/plugins/Domoticz-Zigbee/Classes/ZigpyTransport/forwarderThread.py\", line 48, in forwarder_thread\n    forward_message(self, message)\n  File \"/var/www/domoticz/plugins/Domoticz-Zigbee/Classes/ZigpyTransport/instrumentation.py\", line 38, in f_out\n    result = f_in(self, message)\n             ^^^^^^^^^^^^^^^^^^^\n  File \"/var/www/domoticz/plugins/Domoticz-Zigbee/Classes/ZigpyTransport/forwarderThread.py\", line 64, in forward_message\n    self.F_out(message)\n  File \"/var/www/domoticz/plugins/Domoticz-Zigbee/plugin.py\", line 773, in processFrame\n    zigbee_receive_message(self, Devices, Data)\n  File \"/var/www/domoticz/plugins/Domoticz-Zigbee/Modules/input.py\", line 165, in zigbee_receive_message\n    _decode_message(self, MsgType, Devices, Data, MsgData, MsgLQI)\n  File \"/var/www/domoticz/plugins/Domoticz-Zigbee/Modules/input.py\", line 172, in _decode_message\n    decoding_method(self, Devices, MsgData, MsgLQI)\n  File \"/var/www/domoticz/plugins/Domoticz-Zigbee/Z4D_decoders/z4d_decoder_Read_Report_Attribute_Rsp.py\", line 52, in Decode8102\n    scan_attribute_reponse(self, Devices, MsgSQN, i_sqn, MsgSrcAddr, MsgSrcEp, MsgClusterId, MsgData, '8102')\n  File \"/var/www/domoticz/plugins/Domoticz-Zigbee/Z4D_decoders/z4d_decoder_Read_Report_Attribute_Rsp.py\", line 88, in scan_attribute_reponse\n    read_report_attributes( self, Devices, msgtype, MsgSQN, MsgSrcAddr, MsgSrcEp, MsgClusterId, MsgAttrID, MsgAttStatus, MsgAttType, MsgAttSize, MsgClusterData, )\n  File \"/var/www/domoticz/plugins/Domoticz-Zigbee/Z4D_decoders/z4d_decoder_Read_Report_Attribute_Rsp.py\", line 121, in read_report_attributes\n    ReadCluster( self, Devices, MsgType, MsgSQN, MsgSrcAddr, MsgSrcEp, MsgClusterId, MsgAttrID, MsgAttStatus, MsgAttType, MsgAttSize, MsgClusterData, Source=MsgType, )\n  File \"/var/www/domoticz/plugins/Domoticz-Zigbee/Modules/readClusters.py\", line 174, in ReadCluster\n    process_cluster_attribute_response( self, Devices, MsgSQN, MsgSrcAddr, MsgSrcEp, MsgClusterId, MsgAttrID, MsgAttType, MsgAttSize, MsgClusterData, Source, )\n  File \"/var/www/domoticz/plugins/Domoticz-Zigbee/Modules/readZclClusters.py\", line 81, in process_cluster_attribute_response\n    func( self, Devices, MsgSrcAddr, MsgSrcEp, MsgClusterId, MsgAttrID, value )\n  File \"/var/www/domoticz/plugins/Domoticz-Zigbee/DevicesModules/custom_zlinky.py\", line 43, in zlinky_clusters\n    zlinky_cluster_metering(self, domoticz_devices, nwkid, ep, cluster, attribut, value)\n  File \"/var/www/domoticz/plugins/Domoticz-Zigbee/DevicesModules/custom_zlinky.py\", line 333, in zlinky_cluster_metering\n    attribute_handlers[attribut]()\n  File \"/var/www/domoticz/plugins/Domoticz-Zigbee/DevicesModules/custom_zlinky.py\", line 318, in <lambda>\n    \"0102\": lambda: _handle_attribut_value(\"0102\", [\"EASF02\", \"HCHP\", \"EJPHPM\", \"BBRHCJW\"], update_color=True, totalize=True),\n                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ [... cut ...]  forwarder_thread\n    handle_thread_error(self, e, message)\n  File \"/var/www/domoticz/plugins/Domoticz-Zigbee/Classes/ZigpyTransport/tools.py\", line 28, in handle_thread_error\n    self.log.logging(\"TransportWrter\", \"Error\", \"Issue in request %s, dumping stack: %s\" %( data, (traceback.format_exc() )), context=context)\n"}},"LastLog":1,"StartTime":"24/08/2025 18:38:17","FirmwareVersion":"6.10.3.0 build 297","FirmwareMajorVersion":"98","PluginVersion":"stable7-7.1.022"},"LastLog":0}
````

SylvainPer, 2025-09-09:
Is everything is working now ? are the widgets for each colors are refreshed ?
For now, it's blue since april so it's the default widget used.

I'm wondering if the pairing process was fully completed and if the specific frame is well received. Could you check in tools, tools, plugin raw devices, find the zlinky and look at the read attributes. You should find:

[image]

pipiche38, 2025-09-22:
Merged in stable7 within Release 7.1.023

closed 2025-09-22 by pipiche38